johanmeijer/grott #510: BMS error 304

PedroKTFC, 2024-01-18:
I know this isn't a Grott issue but I'm hoping someone can help. Last night my system came up with this error (and Grott shows System Fault Word 3 with 33024, all other fault words are 0). My battery no longer seems to report any updates - it's stuck at 33%. Has anyone else seen this? Any idea how to resolve? (I'm in touch with my installer but so far no suggestions from them.)

johanmeijer, 2024-01-18:
This is from the Growatt protocol (only in older documentation described) description: 

[image]

33024 decimal,  is 8100 hex, is 1000 0001 0000 0000 binary. 

Your bit 0 and 7 seems to be set. 7 is reserved. Bit 0 says: BatTerminalReversed. Does this helps? Have you switched the connections?

PedroKTFC, 2024-01-18:
Thanks for that. I'll try and look in the documentation I have so I can work it out next time. However, just as it came without warning, it's now cleared up without warning too! I have no idea what caused it. It happened at around 2:30am. I wonder if something got too cold (it was very very cold last night)? I've done nothing (I'm away just now) other than played with a version of   grottserver I'm working on that reads all the registers in one (repeated!) go and displays them on the web page.

PedroKTFC, 2024-01-18:
Looks like it was the cold! Here's what my installer sent me a little while ago:

> Thankyou for the information. I have had the manufacturer technician investigate this and they have concluded that the only other explanation is that due to the batteries being of lithium make, they have a temperature safety net whereby if the battery temperature drops below 0 degrees, it will automatically stop charging and discharging. It will automatically self-resolve once the temperature rises again.

closed 2024-01-18 by PedroKTFC