johanmeijer/grott #512: Add license information

vazhnov, 2024-01-23:
Hello!

Could you please add information about the license of this repository — is it allowed to copy, change and use the code in any purpose?
It would be great to see any popular open source license.

johanmeijer, 2024-01-23:
Hi, why do you think a license model is needed ? At this moment officially the standard no license rule applies _(without a license, the default copyright laws apply, meaning that you retain all rights to your source code and no one may reproduce, distribute, or create derivative works from your work)_ 

I am not sure if I want to make this project a totally open source project. It is my own "hobby" project and I like to be in control and able to do with it what I want. Others are free to suggest improvements and help to debug, but I still want to be in the lead. 

An other thing that I noticed that some people are using (parts of) my code in commercial applications. I do not say I am totally against it but I like to see some kind of appreciation (e.g. donation) and at least a request for permission. 

I think thousands of people are using the software and I am really happy to see that I build something that is useful. But it also cost a lot of free time ( i have to work to...) and effort to maintain and follow and answer the discussions. 

Maybe you can convince me to make it a real open source project.

FezVrasta, 2026-08-29:
It would much appreciated if you added a license to the project.