johanmeijer/grott #514: Wrong device ID : Xxxxxxxxx

Supered1967, 2024-01-25:
Hey,

maybe I am a noob, but for some reason my device id in the logs changed from CS363827 to XXXXXXX, and thus throwing a new device into HA.

Can I fix this to change the ID on the Grott.ini (id is now marked out with #)?

Its probably not a bug ……:)

regards
Ed

Supered1967, 2024-01-28:
Following up, here are the logs.

The first log is when the inverter started this morning

===================================================
2024-01-28T07:07:04.256688939Z 	 - ('192.168.1.30', 32140) has connected
2024-01-28T07:07:04.569909645Z 
2024-01-28T07:07:04.569967146Z 	 - Growatt packet received:
2024-01-28T07:07:04.569997642Z 		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.6', 37854), raddr=('47.91.67.66', 5279)>
2024-01-28T07:07:04.570100161Z 	 - Data less then minimum record length, data not processed
2024-01-28T07:07:04.570131167Z 
2024-01-28T07:07:04.570148511Z 	 - Growatt packet received:
2024-01-28T07:07:04.570166737Z 		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.6', 5279), raddr=('192.168.1.30', 32140)>
2024-01-28T07:07:04.570192259Z 	 - Data less then minimum record length, data not processed
2024-01-28T07:07:04.570212589Z 
2024-01-28T07:07:04.570238212Z 	 - Growatt packet received:
2024-01-28T07:07:04.570256214Z 		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.6', 37854), raddr=('47.91.67.66', 5279)>
2024-01-28T07:07:04.570279309Z 	 - Data less then minimum record length, data not processed
2024-01-28T07:07:04.570298653Z 
2024-01-28T07:07:04.570314275Z 	 - Growatt packet received:
2024-01-28T07:07:04.570331293Z 		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.6', 37854), raddr=('47.91.67.66', 5279)>
2024-01-28T07:07:04.570353549Z 	 - Grott - grottproxy - Invalid data record received, processing stopped for this record
2024-01-28T07:07:04.570377445Z 
2024-01-28T07:07:04.570393120Z 	 - Growatt packet received:
2024-01-28T07:07:04.570410244Z 		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.6', 37854), raddr=('47.91.67.66', 5279)>
2024-01-28T07:07:04.570432625Z 	 - Data less then minimum record length, data not processed
2024-01-28T07:07:04.570454022Z 
2024-01-28T07:07:04.570469500Z 	 - Growatt packet received:
2024-01-28T07:07:04.570486350Z 		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.6', 37854), raddr=('47.91.67.66', 5279)>
2024-01-28T07:07:04.570507878Z 	 - Grott - grottproxy - Invalid data record received, processing stopped for this record
2024-01-28T07:07:04.570527933Z 
2024-01-28T07:07:04.570545333Z 	 - Growatt packet received:
2024-01-28T07:07:04.570562829Z 		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.6', 37854), raddr=('47.91.67.66', 5279)>
2024-01-28T07:07:04.570584648Z 	 - Grott - grottproxy - Invalid data record received, processing stopped for this record
2024-01-28T07:07:04.570658077Z 
2024-01-28T07:07:04.570676705Z 	 - Growatt packet received:
2024-01-28T07:07:04.570694077Z 		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.6', 5279), raddr=('192.168.1.30', 32140)>
2024-01-28T07:07:04.570718785Z 	 - Data less then minimum record length, data not processed
2024-01-28T07:07:12.971667272Z 
2024-01-28T07:07:12.971920481Z 	 - Growatt packet received:
2024-01-28T07:07:12.971969321Z 		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.6', 37854), raddr=('47.91.67.66', 5279)>
2024-01-28T07:07:12.972084500Z 	 - Growatt original Data:
2024-01-28T07:07:12.972580299Z 		 \x00\x01\x00\x02\x00\xd9\x01\x03\x41\x48\x34\x35\x31\x36\x30\x36\x33\x36\x58
2024-01-28T07:07:12.972723860Z 		 \x58\x58\x58\x58\x58\x58\x58\x58\x58\x00\x00\x00\x00\x00\x00\x02\x00\x00\x00
2024-01-28T07:07:12.972752902Z 		 \x2c\x01\x01\x00\x01\x00\x00\x00\x64\x00\x00\x4e\x20\x00\x00\xa4\xd8\x0e\x42
2024-01-28T07:07:12.972799751Z 		 \x53\x2e\x31\x2e\x38\x00\x1e\x1e\x1e\x1e\x1e\x1e\x00\x01\x00\x05\x05\xdc\x00
2024-01-28T07:07:12.972828358Z 		 \x1e\x07\x3a\x0a\x50\x12\x91\x13\x99\x58\x58\x58\x58\x58\x58\x58\x58\x58\x58
2024-01-28T07:07:12.972871871Z 		 \x00\x00\x90\x41\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x07\x35\x09\xdd\x12
2024-01-28T07:07:12.972916066Z 		 \x91\x13\x99\x07\x80\x0a\x28\x12\x69\x13\xc1\x00\x64\x02\x01\x00\x2d\x00\x59
2024-01-28T07:07:12.972951531Z 		 \x00\x00\x00\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x96\x00\x96\x00\x05\x00
2024-01-28T07:07:12.972975346Z 		 \x05\x00\x96\x00\x96\x00\x03\x00\x03\x20\x20\x20\x50\x56\x20\x49\x6e\x76\x65
2024-01-28T07:07:12.973000626Z 		 \x72\x74\x65\x72\x20\x20\x53\x42\x41\x41\x33\x34\x30\x30\x00\x00\x00\x00\x01
2024-01-28T07:07:12.973022092Z 		 \x2c\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
2024-01-28T07:07:12.973043342Z 		 \x00\x00\x00\x00\x00\x00\x00\x53\x00\x55\x00\x57\x00\x59
2024-01-28T07:07:12.973073342Z 	 - Grott automatic protocol detection
2024-01-28T07:07:12.973102006Z 	 - Grott data record length 223
2024-01-28T07:07:12.973125192Z 	 - layout   :  T020103
2024-01-28T07:07:12.973144006Z 	 - no matching record layout found, try generic
2024-01-28T07:07:12.973165975Z 	 - Record layout used :  T020103
2024-01-28T07:07:12.973447729Z 	 - Growatt data decrypted V2
2024-01-28T07:07:12.973482296Z 	 - Grott Growatt data decrypted
2024-01-28T07:07:12.973503657Z 	 - Growatt plain data:
[... cut ...]
2024-01-28T08:01:48.903892523Z 			 0, "epv1today": 0, "epv1total": 177797, "epv2today": 0, "epv2total": 151485,
2024-01-28T08:01:48.903913275Z 			 "epvtotal": 329282}}
2024-01-28T08:01:48.903932424Z 	 - Grott MQTT topic used : energy/growatt
2024-01-28T08:01:48.904027648Z 	 - MQTT message message sent
2024-01-28T08:01:48.904073980Z 	 - Grott Send data to PVOutput disabled 
2024-01-28T08:01:48.904103057Z 	 - Grott Send data to Influx disabled 
2024-01-28T08:01:48.904126239Z 	 - Grott extension processing disabled

johanmeijer, 2024-10-08:
Ok. I seem to have missed this one. 

I do not understand the problem  totally. Is the inverterid wrong in Grott or in HA?

Supered1967, 2024-10-08:
Don’t worry Johan, I did solved the issue earlier by forcing the Device ID.
Problem was that Grott decoded somehow a very strange device ID, but somewhere in the code translated/converted in device XXXXXXXX. That message also was send to HA using MQTT. 
So in HA I had 2 devices, one with the right ID and one with ID XXXXXXXX

The only thing I remembered is that device XXXXXXX only appeared at the start of the inverter, and that ID was kept unless Grott was restarted when the inverter was running.

So, maybe this issue has to do something with the issue I am now facing

regards 
Ed

johanmeijer, 2024-10-08:
That would be strange while Grott does not have any memory. It retrieves the inverterid from the data record (every time again). 

But maybe it has to do with the double stacked records as well.