johanmeijer/grott #516: Is there a reason, that 2.7.8 ist still the lasted release?

pimisen, 2024-01-27:
I see 2.8.2 is out, but docker image:latest is still 2.7.8.
?

johanmeijer, 2024-01-28:
Nope there is no real reason for that. Beta release 2.8.3 (and docker: grott:2.8.2) is master "worthy". I only have not done it yet.

rany2, 2024-01-28:
@johanmeijer I know you don't take contributions but in my fork I have the process of generating docker images automated, you could have a look at:

- https://github.com/rany2/grott/blob/master/.github/workflows/docker-publish.yml
- https://github.com/rany2/grott/blob/master/Dockerfile

PianSom, 2024-02-01:
@johanmeijer Please can you release an "official" Docker 2.8.3 soon? I'm keen to lose the SPACE bug for entities in HA!

johanmeijer, 2024-02-02:
I created a docker 2.8.3 container (for all architectures). I am still testing it so not yet an official  (master or beta) image.
You are welcome to test it: ledidobe/grott:2.8.3 image on docker.

PianSom, 2024-02-02:
Thanks, pulled. Will see how it goes when the inverter kicks off tomorrow.

johanmeijer, 2024-02-04:
I am trying to cleanup the mesh I made with version (docker and github). For a small period in time 2.7.8 will be the master (latest) and 2.8.3 the beta for both Github and docker. 

Soon I will make the 2.8.3 the master and there will be a 3.0.0 Beta.

PianSom, 2024-02-04:
All working well. Thanks again.

johanmeijer, 2024-02-04:
@PianSom what is the reason you add this link to this issue? Still no PV2 values?

PianSom, 2024-02-04:
Not sure why GitHub put the link here, sorry. 

I left feedback for you (above) then put a link to this discussion on the Home Assistant Grott GitHub project discussion to help others.

pimisen, 2024-03-14:
@johanmeijer do you have any influence or know whom to contact to get the Docker ledidobe/grott updated to 2.8.2 or 2.8.3?

PianSom, 2024-03-14:
@pimisen See https://hub.docker.com/r/ledidobe/grott/tags for 2.8.3

johanmeijer, 2024-03-14:
Use ledidobe/grottt:beta (now 2.8.3)

pimisen, 2024-10-14:
@johanmeijer Hi Johann, may I ask to you to bring grott 2.8.3 to ledidobe/grott:latest ? I want to update without setting up the complete docker again :-)

johanmeijer, 2024-10-15:
Hi @pimisen I will plan todo this week (weekend)

pimisen, 2024-10-16:
don't hurry, i managed to switch to the beta channel of docker meanwhile (and are so happy with 2.8.3 latest version). No IP hassle anymore. Thanks so much @johanmeijer