johanmeijer/grott #521: Growatt MOD 10KTL3-X & Eastron SDM630 meter - no smart meter values

hnktnk, 2024-02-13:
Hi!

I recently got my smart meter installed along my inverter, Growatt MOD 10KTL3-X. I get the power values from the meter correct into the inverter and Grott successfully sends it upstream to Shine-platform.

I cannot get the values into HA though. My inverter values gets sent via MQTT correctly and that has worked fine since may. Looking at the grott log, I think I have found the meter-packets:
Feb 13 18:09:59 backend grott[803940]:          - Growatt original Data:
Feb 13 18:09:59 backend grott[803940]:                  \x00\xae\x00\x06\x00\xe6\x50\x20\x09\x33\x2c\x36\x22\x38\x36\x77\x31\x57\x77
Feb 13 18:09:59 backend grott[803940]:                  \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Feb 13 18:09:59 backend grott[803940]:                  \x2c\x2d\x2c\x44\x30\x06\x44\x5f\x43\x52\x74\x74\x47\x72\x6f\x77\x61\x74\x74
Feb 13 18:09:59 backend grott[803940]:                  \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\xc1
Feb 13 18:09:59 backend grott[803940]:                  \x74\x74\x47\xf4\x6f\x77\x69\x93\x74\x47\x7a\x96\x77\x61\x7d\x74\x47\x72\x6f
Feb 13 18:09:59 backend grott[803940]:                  \x63\x61\x74\x74\x4e\x72\x6f\x77\x64\x74\x74\x48\x3b\x6f\x77\x64\xf3\x74\x47
Feb 13 18:09:59 backend grott[803940]:                  \x71\xff\x77\x61\x65\x8c\x47\x72\x67\xe2\x61\x74\x70\xf5\x8d\x90\x81\xb3\x8b
Feb 13 18:09:59 backend grott[803940]:                  \x8b\xbe\x07\x90\x88\x9d\x82\x74\x47\x71\x34\x77\x61\x76\xf1\x47\x72\x6d\x8f
Feb 13 18:09:59 backend grott[803940]:                  \x61\x74\x6c\x27\x72\x6f\x53\x2d\x8b\x8b\xaa\x57\x6f\x77\x62\x6c\x74\x47\x73
Feb 13 18:09:59 backend grott[803940]:                  \x9c\x77\x61\x7b\x0e\x47\x72\x60\xe6\x61\x74\x7b\xc7\x72\x6e\x47\x42\x74\x74
Feb 13 18:09:59 backend grott[803940]:                  \x49\x29\x6f\x77\x68\xbc\x74\x47\x26\x03\x77\x60\x38\x56\x47\x73\x51\x09\x61
Feb 13 18:09:59 backend grott[803940]:                  \x74\x2a\x72\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
Feb 13 18:09:59 backend grott[803940]:                  \x77\x61\x74\x74\x47\x72\x6f\x77\xc6\xdb
Feb 13 18:09:59 backend grott[803940]:          - Grott automatic protocol detection
Feb 13 18:09:59 backend grott[803940]:          - Grott data record length 238
Feb 13 18:09:59 backend grott[803940]:          - layout   :  T065020
Feb 13 18:09:59 backend grott[803940]:          - Record layout used :  T065020
Feb 13 18:09:59 backend grott[803940]:          - Growatt data decrypted V2
Feb 13 18:09:59 backend grott[803940]:          - Grott Growatt data decrypted
Feb 13 18:09:59 backend grott[803940]:          - Growatt plain data:
Feb 13 18:09:59 backend grott[803940]:                  00ae000600e650204e414341434c4230433800000000000000000000000000000000000000004
Feb 13 18:09:59 backend grott[803940]:                  35a4d30444136303433000000000000000000000000000000000000000000000000000000a000
Feb 13 18:09:59 backend grott[803940]:                  000086000008e7000008f90000090000000014000000090000000500000f49000005870000039
Feb 13 18:09:59 backend grott[803940]:                  0000011f800000895000004b2fffff6d2fffff975fffffcf60000035b00000285000002f80000
Feb 13 18:09:59 backend grott[803940]:                  18600000244cffffed2500000318000001f300000f7a00000f9100000f800001302300000e5b0
Feb 13 18:09:59 backend grott[803940]:                  00009c80000546c00014c2200013e7e00005e3500000000000000000000000000000000000000
Feb 13 18:09:59 backend grott[803940]:                  0000000000a7af
Feb 13 18:09:59 backend grott[803940]:          - Growatt new layout processing
Feb 13 18:09:59 backend grott[803940]:                  - decrypt       :  True
Feb 13 18:09:59 backend grott[803940]:                  - offset        :  6
Feb 13 18:09:59 backend grott[803940]:                  - record layout :  T065020
Feb 13 18:09:59 backend grott[803940]:          - grottdata - error in keyword processing :  active_energy ,data processing stopped

I have tried set:
`#        self.recorddict11 = {"T06501b": {`
to
`        self.recorddict11 = {"T065020": {` to "capture" the meter values but nah, that wont work.

I also get this packet:
Feb 13 18:10:21 backend grott[803940]:          - Growatt original Data:
Feb 13 18:10:21 backend grott[803940]:                  \x00\xb1\x00\x06\x02\x3d\x50\x38\x09\x33\x2c\x36\x22\x38\x36\x77\x31\x57\x77
Feb 13 18:10:21 backend grott[803940]:                  \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Feb 13 18:10:21 backend grott[803940]:                  \x2c\x2d\x2c\x44\x30\x06\x44\x5f\x43\x52\x74\x74\x47\x72\x6f\x77\x61\x74\x74
Feb 13 18:10:21 backend grott[803940]:                  \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x71\x72
Feb 13 18:10:21 backend grott[803940]:                  \xac\x67\xb8\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
Feb 13 18:10:21 backend grott[803940]:                  \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
Feb 13 18:10:21 backend grott[803940]:                  \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
Feb 13 18:10:21 backend grott[803940]:                  \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
Feb 13 18:10:21 backend grott[803940]:                  \x61\x74\x74\x47\x72\x6f\x77\x75\x74\x60\x60\x72\x6f\x77\x61\x74\x74\x47\x72
Feb 13 18:10:21 backend grott[803940]:                  \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
Feb 13 18:10:21 backend grott[803940]:                  \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
Feb 13 18:10:21 backend grott[803940]:                  \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
Feb 13 18:10:21 backend grott[803940]:                  \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x75\x5c\x60\x08
Feb 13 18:10:21 backend grott[803940]:                  \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
Feb 13 18:10:21 backend grott[803940]:                  \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
Feb 13 18:10:21 backend grott[803940]:                  \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Feb 13 17:58:48 backend grott[803805]:          - Grott extension processing disabled
Feb 13 18:11:20 backend grott[803940]:                  \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
Feb 13 18:11:20 backend grott[803940]:                  \x47\x72\x6f\x77\x75\x24\x60\x30\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
Feb 13 18:11:20 backend grott[803940]:                  \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
Feb 13 18:11:20 backend grott[803940]:                  \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
Feb 13 18:11:20 backend grott[803940]:                  \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
Feb 13 18:11:20 backend grott[803940]:                  \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x75\x0c\x60\xd8\x72\x6f\x77
Feb 13 18:11:20 backend grott[803940]:                  \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
[... 13 lines cut ...]
Feb 13 18:11:20 backend grott[803940]:          - Growatt data decrypted V2
Feb 13 18:11:20 backend grott[803940]:          - Grott Growatt data decrypted
Feb 13 18:11:20 backend grott[803940]:          - Growatt plain data:
Feb 13 18:11:20 backend grott[803940]:                  00b60006023d50384e414341434c4230433800000000000000000000000000000000000000004
Feb 13 18:11:20 backend grott[803940]:                  35a4d3044413630343300000000000000000000000000000000000000000000000000000613d8
Feb 13 18:11:20 backend grott[803940]:                  13ff0000000000000000000000000000000000000000000000000000000000000000000000000
Feb 13 18:11:20 backend grott[803940]:                  00000000000000000000000000000000000000000000000000000000000000000000000000000
Feb 13 18:11:20 backend grott[803940]:                  00000000001400142700000000000000000000000000000000000000000000000000000000000
Feb 13 18:11:20 backend grott[803940]:                  00000000000000000000000000000000000000000000000000000000000000000000000000000
[... cut ...]
Feb 13 18:11:20 backend grott[803940]:                  00000000000000000000000000000000000000000000000000000000000000000000000000000
Feb 13 18:11:20 backend grott[803940]:                  00083c9
Feb 13 18:11:20 backend grott[803940]:          - Grott data ack record or data record not defined no processing done

I'm running Grott 2.8.3 and I've tried the master branch aswell with the same results.
I run invertertype as "min" in grott.ini, I've tried tl3 but that just messed up some readings in Home Assistant so I switched back to "min".

Am I doing something wrong here?

stefanionica, 2024-09-22:
Hi everybody, Please "light me" how to solve this issues for MOD 10KTL3-X : (300) AC V Outrange

https://github.com/stefanionica/html2024/blob/main/Screenshot%202024-09-22%20212248.png

johanmeijer, 2024-09-23:
For problem solving with the HA integration maybe @egguy can help. For MOD inverter you can use invtype=MOD now to get better information. 

AC voltage outrange seem to me a grid problem. Do you have a lot of neighbour solar panels in your environment? Overvoltage normale happens when the grid is overloaded. This can only be solved by the grid operator I am afraid.

hnktnk, 2024-09-27:
> For problem solving with the HA integration maybe @egguy can help. For MOD inverter you can use invtype=MOD now to get better information.
> 

As someone who doesn't speak Grott-Matrix, can you from my log see that the meter-data is at least present? I totally forgot that there is an integration between grott and HA.

RockyVod, 2024-10-11:
Hi, how did you manage to read data from smart meter? I use growattServer packager
I see the smart meter device
 {'deviceType': 'other', 'invType': '0', 'address': 1, 'value2': '0.0', 'value1': '280.3', 'iconimg': 'https://growatt-10051877.cos.ap-shanghai.myqcloud.com/publish/deviceicon_dianbiao.png', 'type': '134', 'deviceSn': 'XGD5BGL2F9_1', 'param1': '有功功率', 'param2': '累计用电量', 'datalogSn': 'XGD5BGL2F9', 'deviceAilas': 'XGD5BGL2F9_1', 'deviceStatusText': ' 在线'},

But am unable to read any data from them. How can I do it?

johanmeijer, 2024-10-13:
@RockyVod grott does process smart meter data and send it to MQTT (and HA?). 

@hnktnk 

In the grott log you would see your smartmeter data (for your record above): 

[image]