johanmeijer/grott #524: Incorrect data being output - SPH3000 and SPH specified in config but logs show no matching layout, instead using T060136X?

jerodequin, 2024-02-22:
Hello,

Trying to get this working but am getting odd mappings of the data from the inverter.

I have invtype set to sph in the config but there is no sign of an sph layout being used when looking at the logs as posted below.

What should I be doing/changing to get this to work please? (if any more info or logs are needed please let me know).

Log below:

<html><body>

2024/02/22 15:54:26 | stdout | - Data less then minimum record length, data not processed
-- | -- | --
2024/02/22 15:54:26 | stdout | <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.244', 5279), raddr=('192.168.1.191', 63680)>
2024/02/22 15:54:26 | stdout | - Growatt packet received:
2024/02/22 15:54:26 | stdout |  
2024/02/22 15:54:26 | stdout | - Grott data ack record or data record not defined no processing done
2024/02/22 15:54:26 | stdout | 0000000000071ca
2024/02/22 15:54:26 | stdout | 00000000000000000000000000000000000000000000000000000000000000000000000000000
2024/02/22 15:54:26 | stdout | 0071000009eb3000107770211ffffffcf004e0064000046f0000037c8006f0075000000000000
2024/02/22 15:54:26 | stdout | 000000000000000000000000000000000000000000fffff25400000041000007b500000018000
2024/02/22 15:54:26 | stdout | 00000000000000000000000000000000000000000000000000000000000000000000000000000
2024/02/22 15:54:26 | stdout | 00000000000000000000000000000000000000000000000000000000000000000000000000000
2024/02/22 15:54:26 | stdout | 00000000000000000000000000000000000000000000000000000000000000000000000000000
2024/02/22 15:54:26 | stdout | bb000000000000000000000000000000000000000079957a11000000000000000000000000000
2024/02/22 15:54:26 | stdout | 000000000000000000000000f5e0000040d000013990960000000000013000000000000008100
2024/02/22 15:54:26 | stdout | 00000000000000000000000000000000000000000000000000000000000000000000000000000
2024/02/22 15:54:26 | stdout | 00000000000640000000000000000000000000000000000000000000000000000000000000000
2024/02/22 15:54:26 | stdout | 00000000000000000000000000000000000000000000000000000000000000000000000000000
2024/02/22 15:54:26 | stdout | 00000000000000000000000000000000000000000000000000000000000000000000000000000
2024/02/22 15:54:26 | stdout | 79940005000300000000000000000000000000000000077a000008e7000000000000000000000
2024/02/22 15:54:26 | stdout | f554354434a4d30323200000000000000000000000000000000000000001802160f3619027918
2024/02/22 15:54:26 | stdout | 00c30006024101364a504338434c5330544a00000000000000000000000000000000000000004
2024/02/22 15:54:26 | stdout | - Growatt plain data:
2024/02/22 15:54:26 | stdout | - Grott Growatt data decrypted
2024/02/22 15:54:26 | stdout | - Growatt data decrypted V2
2024/02/22 15:54:26 | stdout | - Record layout used :  T060136X
2024/02/22 15:54:26 | stdout | - no matching record layout found, try generic
2024/02/22 15:54:26 | stdout | - layout   :  T060136X
2024/02/22 15:54:26 | stdout | - Grott data record length 585
2024/02/22 15:54:26 | stdout | - Grott automatic protocol detection
2024/02/22 15:54:26 | stdout | \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x03\xa5
2024/02/22 15:54:26 | stdout | \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
2024/02/22 15:54:26 | stdout | \x01\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
2024/02/22 15:54:26 | stdout | \x8d\x90\x88\xae\x74\x3a\x47\x16\x6f\x77\x27\x84\x74\x47\x45\xa7\x77\x0e\x74
2024/02/22 15:54:26 | stdout | \xc2\x61\x74\x74\x5f\x72\x6f\x70\x71\x74\x74\xd9\xc1\x6f\x76\x66\x03\x76\x56
2024/02/22 15:54:26 | stdout | \x74\x74\x47\x72\x6f\x77\x61\x74\x8b\xb8\x80\x3b\x77\x61\x74\x35\x47\x72\x68
2024/02/22 15:54:26 | stdout | \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
2024/02/22 15:54:26 | stdout | \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
2024/02/22 15:54:26 | stdout | \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
2024/02/22 15:54:26 | stdout | \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
2024/02/22 15:54:26 | stdout | \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
2024/02/22 15:54:26 | stdout | \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
2024/02/22 15:54:26 | stdout | \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
2024/02/22 15:54:26 | stdout | \x47\x72\x6f\x77\x61\x74\x0d\xd2\x08\x7e\x77\x61\x74\x74\x47\x72\x6f\x77\x61
2024/02/22 15:54:26 | stdout | \x6f\x77\xe0\x74\xcf\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
2024/02/22 15:54:26 | stdout | \x65\x79\x74\x47\x61\xf6\x7e\x01\x74\x74\x47\x72\x6f\x64\x61\x74\x74\x47\x72
2024/02/22 15:54:26 | stdout | \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x48\x2c\x6f\x77
2024/02/22 15:54:26 | stdout | \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
2024/02/22 15:54:26 | stdout | \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
2024/02/22 15:54:26 | stdout | \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
2024/02/22 15:54:26 | stdout | \x47\x72\x6f\x77\x61\x74\x74\x47\x16\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
2024/02/22 15:54:26 | stdout | \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
2024/02/22 15:54:26 | stdout | \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
2024/02/22 15:54:26 | stdout | \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
2024/02/22 15:54:26 | stdout | \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
2024/02/22 15:54:26 | stdout | \x77\x61\x74\x74\x40\x08\x6f\x77\x69\x93\x74\x47\x72\x6f\x77\x61\x74\x74\x47
2024/02/22 15:54:26 | stdout | \x6c\x0d\xd3\x72\x6a\x77\x62\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
2024/02/22 15:54:26 | stdout | \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x79\x76\x62\x48\x44\x76\x75\x18
[... 68 lines cut ...]
2024/02/22 15:54:05 | stdout | 000000000000000000000000000000000000000000000000000000000dd30
2024/02/22 15:54:05 | stdout | 0000c300c00000000000000000000000000000000000000000000000000000000000000000000
2024/02/22 15:54:05 | stdout | 000000000010cee0ceb00030030001200010076006c000400050000004f004e00000000000000
2024/02/22 15:54:05 | stdout | 00000000000000000000000000000000000000000000000000000000000000000000000000000
2024/02/22 15:54:05 | stdout | 00000000000000000000000000000000000000000000000000000000000000000000000000000
2024/02/22 15:54:05 | stdout | 000b060000001c0000090f00000e1000000e100000000800020000021f00020000027b0000000
2024/02/22 15:54:05 | stdout | 00000000000046504e10042000006ec0000000004270000000000000000000000000000001c00
2024/02/22 15:54:05 | stdout | 000000004150415000000cee0ceb00030e420cee0cee0ced0cec0ced0ced0ceb0ceb000000000
2024/02/22 15:54:05 | stdout | 300000000004f14aafe2a000b1e0037fa46f00000000200140064163000000000000000000000
2024/02/22 15:54:05 | stdout | 00000000000000000000000000000000000000000000000000000000000000000003e80000016
2024/02/22 15:54:05 | stdout | 049000008860000000100000209000000180000071000000041000007b50000002700000b4300
2024/02/22 15:54:05 | stdout | 0000000000000000000000000e74000000000000000000000e740000000b00050ba431ce00000
2024/02/22 15:54:05 | stdout | 00000000000000aea000000000211004f000000a30000000000000000000000a3000000000000
2024/02/22 15:54:05 | stdout | 0006ec00000000000000010000000000000000000003e80464000500000000000000000000000
2024/02/22 15:54:05 | stdout | 0b90000021310b910e60000000000000000000000000000000000000000000000000000004200
2024/02/22 15:54:05 | stdout | 0000000000000000000000000000000000000000000000000000000000000000056b00c700ab0
2024/02/22 15:54:05 | stdout | 8004f163f000000040000025a0000000300000307000000000000000000000000000000000000
2024/02/22 15:54:05 | stdout | 96095a001300000fa6000000000000000000000000000000000000000000000000001c000009d
2024/02/22 15:54:05 | stdout | 000000000000000000000000000000000000000000000000000000000000000000000000fa513
2024/02/22 15:54:05 | stdout | 007c0005000003e8066b0002000001c707ee00020000022f000150a009b907df08a100fa00000
2024/02/22 15:54:05 | stdout | f554354434a4d30323200000000000000000000000000000000000000001802160f3030030000
2024/02/22 15:54:05 | stdout | 00050006033f01504a504338434c5330544a00000000000000000000000000000000000000004

</body>
</html>

PedroKTFC, 2024-02-22:
Are you using SPH or sph? Think I remember it has to be SPH.

jerodequin, 2024-02-22:
Thank you, have just tried stopping the server and changing to SPH in config but on restart of the server, I still get assigned a layout of T060103X rather than SPH after no matching record layout found.

johanmeijer, 2024-02-28:
Do you have the messages after startup that shows the configuration being used?

jerodequin, 2024-03-03:
> Do you have the messages after startup that shows the configuration being used?

Hello Johan, thanks for replying - 

This is everything in the log immediately after starting grott:

2024/03/03 15:03:16 | stdout | - Data less then minimum record length, data not processed
-- | -- | --
2024/03/03 15:03:16 | stdout | <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.244', 5279), raddr=('192.168.1.191', 44480)>
2024/03/03 15:03:16 | stdout | - Growatt packet received:
2024/03/03 15:03:16 | stdout |  
2024/03/03 15:03:16 | stdout | - Grott extension processing disabled
2024/03/03 15:03:16 | stdout | - Grott Send data to Influx disabled
2024/03/03 15:03:16 | stdout | - Grott Send data to PVOutput disabled
2024/03/03 15:03:16 | stdout | - MQTT message message sent
2024/03/03 15:03:16 | stdout | - Grott MQTT topic used : energy/growatt
2024/03/03 15:03:16 | stdout | "pvtemperature": 334, "pvipmtemperature": 533}}
2024/03/03 15:03:16 | stdout | "epv1today": 3508, "epv1total": 6902925, "epv2today": 58, "epv2total": 889,
2024/03/03 15:03:16 | stdout | "totworktime": 0, "pvenergytoday": 0, "pvenergytotal": 0, "epvtotal": 116,
2024/03/03 15:03:16 | stdout | 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3": 2186089342,
2024/03/03 15:03:16 | stdout | "pvgridpower": 0, "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2":
2024/03/03 15:03:16 | stdout | "pvpowerout": 0, "pvfrequentie": 0, "pvgridvoltage": 0, "pvgridcurrent": 0,
2024/03/03 15:03:16 | stdout | "pv1watt": 16450, "pv2voltage": 2110, "pv2current": 95, "pv2watt": 20134,
2024/03/03 15:03:16 | stdout | "pvstatus": 5, "pvpowerin": 36500, "pv1voltage": 1775, "pv1current": 92,
2024/03/03 15:03:16 | stdout | "values": {"datalogserial": "JPC8CLS0TJ", "pvserial": "OUCTCJM022",
2024/03/03 15:03:16 | stdout | {"device": "OUCTCJM022", "time": "2024-03-03T15:02:09", "buffered": "yes",
2024/03/03 15:03:16 | stdout | - MQTT jsonmsg:
2024/03/03 15:03:16 | stdout | -  pvipmtemperature     :  53.3
2024/03/03 15:03:16 | stdout | -  pvtemperature        :  33.4
2024/03/03 15:03:16 | stdout | -  epv2total            :  88.9
2024/03/03 15:03:16 | stdout | -  epv2today            :  5.8
2024/03/03 15:03:16 | stdout | -  epv1total            :  690292.5
2024/03/03 15:03:16 | stdout | -  epv1today            :  350.8
2024/03/03 15:03:16 | stdout | -  epvtotal             :  11.6
2024/03/03 15:03:16 | stdout | -  pvenergytotal        :  0.0
2024/03/03 15:03:16 | stdout | -  pvenergytoday        :  0.0
2024/03/03 15:03:16 | stdout | -  totworktime          :  0.0
2024/03/03 15:03:16 | stdout | -  pvgridpower3         :  218608934.2
2024/03/03 15:03:16 | stdout | -  pvgridcurrent3       :  0.0
2024/03/03 15:03:16 | stdout | -  pvgridvoltage3       :  0.0
2024/03/03 15:03:16 | stdout | -  pvgridpower2         :  0.0
2024/03/03 15:03:16 | stdout | -  pvgridcurrent2       :  0.0
2024/03/03 15:03:16 | stdout | -  pvgridvoltage2       :  0.0
2024/03/03 15:03:16 | stdout | -  pvgridpower          :  0.0
2024/03/03 15:03:16 | stdout | -  pvgridcurrent        :  0.0
2024/03/03 15:03:16 | stdout | -  pvgridvoltage        :  0.0
2024/03/03 15:03:16 | stdout | -  pvfrequentie         :  0.0
2024/03/03 15:03:16 | stdout | -  pvpowerout           :  0.0
2024/03/03 15:03:16 | stdout | -  pv2watt              :  2013.4
2024/03/03 15:03:16 | stdout | -  pv2current           :  9.5
2024/03/03 15:03:16 | stdout | -  pv2voltage           :  211.0
2024/03/03 15:03:16 | stdout | -  pv1watt              :  1645.0
2024/03/03 15:03:16 | stdout | -  pv1current           :  9.2
2024/03/03 15:03:16 | stdout | -  pv1voltage           :  177.5
2024/03/03 15:03:16 | stdout | -  pvpowerin            :  3650.0
2024/03/03 15:03:16 | stdout | -  pvstatus             :  5
2024/03/03 15:03:16 | stdout | -  pvserial             :  OUCTCJM022
2024/03/03 15:03:16 | stdout | -  datalogserial        :  JPC8CLS0TJ
2024/03/03 15:03:16 | stdout | - Grott values retrieved:
2024/03/03 15:03:16 | stdout | - date-time:  2024-03-03T15:02:09
2024/03/03 15:03:16 | stdout | - Grott data record date/time processing started
2024/03/03 15:03:16 | stdout |  
2024/03/03 15:03:16 | stdout | - record layout :  T06NNNNX
2024/03/03 15:03:16 | stdout | - offset        :  6
2024/03/03 15:03:16 | stdout | - decrypt       :  True
2024/03/03 15:03:16 | stdout | - Growatt new layout processing
2024/03/03 15:03:16 | stdout | - Inverter serial not recognised - using inverter type default
2024/03/03 15:03:16 | stdout | - Possible Inverter serial OUCTCJM022
2024/03/03 15:03:16 | stdout | 000000000000000000000000000000000000000000000000000000000e13f
2024/03/03 15:03:16 | stdout | 0000c300400000000000000000000000000000000000000000000000000000000000000000000
[... 830 lines cut ...]
2024/03/03 15:02:13 | stdout | {'decrypt': {'value': 'True'}, 'datalogserial': {'value': 16, 'length': 10, 'type': 'text', 'incl': 'yes'}, 'pvserial': {'value': 36, 'length': 10, 'type': 'text'}, 'date': {'value': 56, 'divide': 10}, 'recortype1': {'value': 70, 'length': 2, 'type': 'num', 'incl': 'no'}, 'recortype2': {'value': 74, 'length': 2, 'type': 'num', 'incl': 'no'}, 'pvstatus': {'value': 78, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 82, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 90, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 94, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1watt': {'value': 98, 'length': 4, 'type': 'num', 'divide': 10},  [... cut ...]
2024/03/03 15:02:13 | stdout | pvoutput:    	 False
2024/03/03 15:02:13 | stdout | inverterid:  	 automatic
2024/03/03 15:02:13 | stdout | nomqtt:      	 False
2024/03/03 15:02:13 | stdout | output file: 	 <_io.TextIOWrapper name='<stdout>' mode='w' encoding='utf-8'>
2024/03/03 15:02:13 | stdout | config file: 	 grott.ini
2024/03/03 15:02:13 | stdout | verbose:     	 True
2024/03/03 15:02:13 | stdout | Grott Command line parameters processed:
2024/03/03 15:02:13 | stdout |  
2024/03/03 15:02:13 | stdout | Grott Growatt logging monitor : 2.8.3

johanmeijer, 2024-03-11:
Somehow your the configuration is not ok. I expected **invtype:               default** to be: invtype: sph. 

Are you using docker or "stand-alone" Grott. Howw did you specified the invtype parameter?

jerodequin, 2024-03-12:
I'm running everything using docker (container manager, running on a Synology NAS). My grott.ini file has the following lines for specifying inverter type:

```
# Specify inverter id (not necessary in version >2.1.0 if compat = false!)
#inverterid = OUCTCJM022
# Specify the type of the inverter (default/sph/spf/max)
invtype = sph
```
It does seem like the ini file is somehow not being read correctly or ignored. Not sure what to try next

jerodequin, 2024-03-12:
@johanmeijer I've just deleted and rebuilt the containers and everything is now working as expected! Not sure what changed, but it is reading the config file correctly now and mapping to SPH. Thank you!