johanmeijer/grott #525: grott 2.8.3 docker lot of log: "Data less then minimum record length, data not processed"

joris-weijters, 2024-02-25:
Hi,
 I just installed a new Growatt 5kTL3-XH invertor wih a ShineWiFi-X datalogger.
I installed grott version 2.8.3 docker image. 
in the grott.ini file I configured the intype to be : MIN
All seems to look ok, however in my docker log I get a lot of messages like these:
```
Growatt packet received:
		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.80.2', 5279), raddr=('192.168.80.1', 51128)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.80.2', 50690), raddr=('47.91.67.66', 5279)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.80.2', 5279), raddr=('192.168.80.1', 51128)>
	 - Data less then minimum record length, data not processed
```

The network where this docker container is running is: 
{"Subnet": "192.168.80.0/20", "Gateway": "192.168.80.1" }

The IP adres of my docker container is 192.168.80.2

Where do these invalid packets originate from?

sjoukebakker, 2024-02-26:
I have the same issue. When I restart the docker container it receives a lot of these 'data less then minimum record length' packets. After that it receives one packet that it can read (T060104X). This data is sent to the growatt server. And then it stops receiving anything until I restart the container. My inverter is a Growatt MIC 1000X.

johanmeijer, 2024-02-28:
I do understand the confusion but message : Data less then minimum record length, data not processed 
is not an error message. It only say that this is not a datarecord (probably is a keep alive ping) and will not be processed. 
This is normal behavior. 

If you have to restart Grott to restart the communication that is strange behaviour. Normally after Grott starts ter will some announcement records, (buffered datarecords) and after that keep alive messages will be sent till a new data record is available (i na normal configuration after 5 minutes).

sjoukebakker, 2024-02-28:
This is my log.. Maybe I should reset my wifi stick to 5 minutes...?
```2024-02-28T16:28:28.686734589Z Grott Growatt logging monitor : 2.6.1g
2024-02-28T16:28:28.691755335Z 
2024-02-28T16:28:28.691838889Z Grott Command line parameters processed:
2024-02-28T16:28:28.691876315Z 	verbose:     	 True
2024-02-28T16:28:28.691903870Z 	config file: 	 grott.ini
2024-02-28T16:28:28.692169905Z 	output file: 	 <_io.TextIOWrapper name='<stdout>' mode='w' encoding='UTF-8'>
2024-02-28T16:28:28.692403587Z 	nomqtt:      	 False
2024-02-28T16:28:28.692547567Z 	inverterid:  	 automatic
2024-02-28T16:28:28.692699362Z 	pvoutput:    	 False
2024-02-28T16:28:28.692926564Z 	blockcmd:    	 False
2024-02-28T16:28:28.692980396Z 	noipf:       	 False
2024-02-28T16:28:28.693008081Z 
2024-02-28T16:28:28.693033192Z Grott process configuration file
2024-02-28T16:28:28.694950340Z 
2024-02-28T16:28:28.695017191Z Grott process environmental variables
2024-02-28T16:28:28.695357002Z 
2024-02-28T16:28:28.695397076Z 	Pulled 'gmqttip=192.168.178.30' from the environment
2024-02-28T16:28:28.695693184Z 
2024-02-28T16:28:28.695733813Z Grott override settings if set in commandline
2024-02-28T16:28:28.696902005Z 
2024-02-28T16:28:28.696967283Z Grott process json layout files
2024-02-28T16:28:28.697248224Z t0060104sph.json
2024-02-28T16:28:28.698943485Z 
2024-02-28T16:28:28.699003688Z Grott layout records loaded
2024-02-28T16:28:28.699026336Z T02NNNN  : 
\00\00\00\00\00\0D\EF\BF\BD\32\30\32\34\2D\30\32\2D\32\38\54\31\36\3A\32\38\3A\32\38\2E\36\39\39\34\38\38\36\38\33\5A\20\7B\27\64\65\63\72\79\70\74\27\3A\20\7B\27\76\61\6C\75\65\27\3A\20\27\46\61\6C\73\65\27\7D\2C\20\27\70\76\73\65\72\69\61\6C
\27\3A\20\7B\27\76\61\6C\75\65\27\3A\20\33\36\2C\20\27\6C\65\6E\67\74\68\27\3A\20\31\30\2C\20\27\74\79\70\65\27\3A\20\27\74\65\78\74\27\2C\20\27\64\69\76\69\64\65\27\3A\20\31\30\7D\2C\20\27\64\61\74\65\27\3A\20\7B\27\76\61\6C\75\65\27\3A\20
\35\36\2C\20\27\64\69\76\69\64\65\27\3A\20\31\30\7D\2C\20\27\72\65\63\6F\72\74\79\70\65\31\27\3A\20\7B\27\76\61\6C\75\65\27\3A\20\37\30\2C\20\27\6C\65\6E\67\74\68\27\3A\20\32\2C\20\27\74\79\70\65\27\3A\20\27\6E\75\6D\27\2C\20\27\69\6E\63\6C
\27\3A\20\27\6E\6F\27\7D\2C\20\27\72\65\63\6F\72\74\79\70\65\32\27\3A\20\7B\27\76\61\6C\75\65\27\3A\20\37\34\2C\20\27\6C\65\6E\67\74\68\27\3A\20\32\2C\20\27\74\79\70\65\27\3A\20\27\6E\75\6D\27\2C\20\27\69\6E\63\6C\27\3A\20\27\6E\6F\27\7D\2C
\20\27\70\76\73\74\61\74\75\73\27\3A\20\7B\27\76\61\6C\75\65\27\3A\20\37\38\2C\20\27\6C\65\6E\67\74\68\27\3A\20\32\2C\20\27\74\79\70\65\27\3A\20\27\6E\75\6D\27\7D\2C\20\27\70\76\70\6F\77\65\72\69\6E\27\3A\20\7B\27\76\61\6C\75\65\27\3A\20\38
\32\2C\20\27\6C\65\6E\67\74\68\27\3A\20\34\2C\20\27\74\79\70\65\27\3A\20\27\6E\75\6D\27\2C\20\27\64\69\76\69\64\65\27\3A\20\31\30\7D\2C\20\27\70\76\31\76\6F\6C\74\61\67\65\27\3A\20\7B\27\76\61\6C\75\65\27\3A\20\39\30\2C\20\27\6C\65\6E\67\74
\68\27\3A\20\32\2C\20\27\74\79\70\65\27\3A\20\27\6E\75\6D\27\2C\20\27\64\69\76\69\64\65\27\3A\20\31\30\7D\2C\20\27\70\76\31\63\75\72\72\65\6E\74\27\3A\20\7B\27\76\61\6C\75\65\27\3A\20\39\34\2C\20\27\6C\65\6E\67\74\68\27\3A\20\32\2C\20\27\74
\79\70\65\27\3A\20\27\6E\75\6D\27\2C\20\27\64\69\76\69\64\65\27\3A\20\31\30\7D\2C\20\27\70\76\31\77\61\74\74\27\3A\20\7B\27\76\61\6C\75\65\27\3A\20\39\38\2C\20\27\6C\65\6E\67\74\68\27\3A\20\34\2C\20\27\74\79\70\65\27\3A\20\27\6E\75\6D\27\2C
\20\27\64\69\76\69\64\65\27\3A\20\31\30\7D\2C\20\27\70\76\32\76\6F\6C\74\61\67\65\27\3A\20\7B\27\76\61\6C\75\65\27\3A\20\31\30\36\2C\20\27\6C\65\6E\67\74\68\27\3A\20\32\2C\20\27\74\79\70\65\27\3A\20\27\6E\75\6D\27\2C\20\27\64\69\76\69\64\65
\27\3A\20\31\30\7D\2C\20\27\70\76\32\63\75\72\72\65\6E\74\27\3A\20\7B\27\76\61\6C\75\65\27\3A\20\31\31\30\2C\20\27\6C\65\6E\67\74\68\27\3A\20\32\2C\20\27\74\79\70\65\27\3A\20\27\6E\75\6D\27\2C\20\27\64\69\76\69\64\65\27\3A\20\31\30\7D\2C\20
\27\70\76\32\77\61\74\74\27\3A\20\7B\27\76\61\6C\75\65\27\3A\20\31\31\34\2C\20\27\6C\65\6E\67\74\68\27\3A\20\34\2C\20\27\74\79\70\65\27\3A\20\27\6E\75\6D\27\2C\20\27\64\69\76\69\64\65\27\3A\20\31\30\7D\2C\20\27\70\76\70\6F\77\65\72\6F\75\74
\27\3A\20\7B\27\76\61\6C\75\65\27\3A\20\31\32\32\2C\20\27\6C\65\6E\67\74\68\27\3A\20\34\2C\20\27\74\79\70\65\27\3A\20\27\6E\75\6D\27\2C\20\27\64\69\76\69\64\65\27\3A\20\31\30\7D\2C\20\27\70\76\66\72\65\71\75\65\6E\74\69\65\27\3A\20\7B\27\76
\61\6C\75\65\27\3A\20\31\33\30\2C\20\27\6C\65\6E\67\74\68\27\3A\20\32\2C\20\27\74\79\70\65\27\3A\20\27\6E\75\6D\27\2C\20\27\64\69\76\69\64\65\27\3A\20\31\30\30\7D\2C\20\27\70\76\67\72\69\64\76\6F\6C\74\61\67\65\27\3A\20\7B\27\76\61\6C\75\65
\27\3A\20\31\33\34\2C\20\27\6C\65\6E\67\74\68\27\3A\20\32\2C\20\27\74\79\70\65\27\3A\20\27\6E\75\6D\27\2C\20\27\64\69\76\69\64\65\27\3A\20\31\30\7D\2C\20\27\70\76\67\72\69\64\63\75\72\72\65\6E\74\27\3A\20\7B\27\76\61\6C\75\65\27\3A\20\31\33
\38\2C\20\27\6C\65\6E\67\74\68\27\3A\20\32\2C\20\27\74\79\70\65\27\3A\20\27\6E\75\6D\27\2C\20\27\64\69\76\69\64\65\27\3A\20\31\30\7D\2C\20\27\70\76\67\72\69\64\70\6F\77\65\72\27\3A\20\7B\27\76\61\6C\75\65\27\3A\20\31\34\32\2C\20\27\6C\65\6E
\67\74\68\27\3A\20\34\2C\20\27\74\79\70\65\27\3A\20\27\6E\75\6D\27\2C\20\27\64\69\76\69\64\65\27\3A\20\31\30\7D\2C\20\27\70\76\67\72\69\64\76\6F\6C\74\61\67\65\32\27\3A\20\7B\27\76\61\6C\75\65\27\3A\20\31\35\30\2C\20\27\6C\65\6E\67\74\68\27
\3A\20\32\2C\20\27\74\79\70\65\27\3A\20\27\6E\75\6D\27\2C\20\27\64\69\76\69\64\65\27\3A\20\31\30\7D\2C\20\27\70\76\67\72\69\64\63\75\72\72\65\6E\74\32\27\3A\20\7B\27\76\61\6C\75\65\27\3A\20\31\35\34\2C\20\27 [... cut ...]
2024-02-28T16:28:56.234722772Z 			 "#pactogrids": 0, "#pactogridt": 0, "pactogridtot": 0, "plocaloadr": 0,
2024-02-28T16:28:56.234745976Z 			 "#plocaloads": 0, "#plocaloadt": 0, "plocaloadtot": 0, "#ipm": 0,
2024-02-28T16:28:56.234768383Z 			 "#battemp": 0, "spdspstatus": 0, "spbusvolt": 0, "etouser_tod": 0,
2024-02-28T16:28:56.234790864Z 			 "etouser_tot": 0, "etogrid_tod": 0, "etogrid_tot": 0, "edischarge1_tod": 0,
2024-02-28T16:28:56.234813457Z 			 "edischarge1_tot": 0, "eharge1_tod": 0, "eharge1_tot": 0, "elocalload_tod":
2024-02-28T16:28:56.234836105Z 			 0, "elocalload_tot": 0}}
```

johanmeijer, 2024-02-28:
@sjoukebakker can you use the beta container (2.8.3) and specify the environmental ginvtype="min". That will fits your inverter better. I am not saying that will solve the the "stop sending" problem but it looks like you need this configuration. 

I also need the log after what so sent here (with the messages it sent the mqtt message etc) because probably the error is after processing and sending the data.

joris-weijters, 2024-02-29:
Johan, Thanks for your explanation.

anoppe, 2024-02-29:
@johanmeijer  I have a MIN 2500TL-X, and if I specify `invtype=min` in the `grott.ini` I get this in the logs: 
```
 - Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T060104XMIN
	 - no matching record layout found, try generic
	 - no matching record layout found, standard processing performed
	 - Record layout used :  none
```
Any idea why it doesn't recognize the inverter type I specified?

johanmeijer, 2024-02-29:
Which version of Grott are you using. You have to use the beta version, 2.8.3.

anoppe, 2024-02-29:
my bad, I pulled v2.8.3 branch, but didn't update my container image. I'll see what happens tomorrow when my inverter comes online...

sjoukebakker, 2024-02-29:
I changed ledidobe/grottrpi:latest to ledidobe/grott:2.8.3 and deployed the container. Now i have this in the log.. 

```

Actions

      

Traceback (most recent call last):

  File "/app/grott.py", line 19, in <module>

    from grottconf import Conf

  File "/app/grottconf.py", line 9, in <module>

    from grottdata import format_multi_line, str2bool

  File "/app/grottdata.py", line 19, in <module>

    import paho.mqtt.publish as publish

  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/publish.py", line 31, in <module>

    from . import client as paho

  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 18, in <module>

    import logging

  File "/usr/local/lib/python3.10/logging/__init__.py", line 57, in <module>

    _startTime = time.time()

PermissionError: [Errno 1] Operation not permitted

Traceback (most recent call last):

  File "/app/grott.py", line 19, in <module>

    from grottconf import Conf

  File "/app/grottconf.py", line 9, in <module>

    from grottdata import format_multi_line, str2bool

  File "/app/grottdata.py", line 19, in <module>

    import paho.mqtt.publish as publish

  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/publish.py", line 31, in <module>

    from . import client as paho

  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 18, in <module>

    import logging

  File "/usr/local/lib/python3.10/logging/__init__.py", line 57, in <module>

    _startTime = time.time()

PermissionError: [Errno 1] Operation not permitted

Traceback (most recent call last):

  File "/app/grott.py", line 19, in <module>

    from grottconf import Conf

  File "/app/grottconf.py", line 9, in <module>

    from grottdata import format_multi_line, str2bool

  File "/app/grottdata.py", line 19, in <module>

    import paho.mqtt.publish as publish

  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/publish.py", line 31, in <module>

    from . import client as paho

  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 18, in <module>

    import logging

  File "/usr/local/lib/python3.10/logging/__init__.py", line 57, in <module>

    _startTime = time.time()

PermissionError: [Errno 1] Operation not permitted
```

also i changed invtype = "min" to ginvtype = "min"

what do you mean with the logs after? This is all i get after a restart of the container.

anoppe, 2024-03-01:
@johanmeijer my issue was fixed by rebuilding the docker image with the right code base:
```
- layout   :  T060103XMIN
	 - no matching record layout found, try generic
	 - Record layout used :  T060103XMIN
```
I suppose the above log snippet is what is expected?
Secondly: I still face a lot of `	 - Data less then minimum record length, data not processed` messages in the logs.

molekuul, 2024-07-11:
as there is no update I will close this issue