johanmeijer/grott #526: MOD 5kTL3-XH with ShineWifi-X time calculation not always correct

joris-weijters, 2024-02-25:
It looks like the time is not calculated correctly from the Growatt data after restart of the grott container.

today at ~12:30 I restarted my grott because of a config change for pvoutput, I set pvtemp = True and pvdisv1 to True.
After the restart in the Log of my container I saw that the date, calculated from the growatt data was 11:39 

This is the data, I replaced my Serial and other secrets 

```
- Growatt original Data:
		 \x00\xd7\x00\x06\x02\x41\x01\x04\x1f\x35\x2b\x41\x22\x3f\x26\x7e\x28\x5d\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x3f\x2e\x2d\x44\x37\x0f\x4a\x5f\x4f\x2c\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x79\x76\x6d\x4c\x55\x79\x75\x6a
		 \xcc\x78\x73\x72\x6e\x77\x61\x2a\x4e\x4f\x8d\x6f\x44\x61\x74\x59\xa6\x7b\x14
		 \x77\x52\x74\x74\x77\x2b\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x2a\x7d\x72\x6f\x76\x94\x74\x74\x11\x88\x7c\xf1\x68\x55
		 \x74\x58\x72\x6f\x6b\x2d\x7d\x5e\x47\x6d\x6f\x77\x7d\x1c\x7c\xb1\x72\x4f\x77
		 \x61\x68\xd8\x48\x96\x60\xea\x6e\xfe\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x31\x4d\x61\x74\xf2\xa2\x72\x6f\x77\x41\x74\x74\x47\x52\x6f\x77\x61\x55\x74
		 \x47\x72\x7f\x77\x61\x74\x64\x47\x72\x6f\x66\x61\x74\x74\x56\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x66\x72\x6f\x77\x40\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x4e\x77\x61\x74\x74\x5e\x2a\x6c\x7c\x60\x2e\x75\x45\x72\x6f\x6f\x0d\x75
		 \xe8\x46\xa9\x6e\xf9\x61\x74\x75\xc5\x7e\x07\x7b\x64\x3a\x54\x47\x58\x6f\x77
		 \xa2\x24\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x4e\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x19\x48\x6f\x77\x61\x56\x78\x72\x7e\xde\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x4d
		 \x77\x61\x74\x56\x47\x72\x6f\x55\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x53\x64\x47\x72\x77\x1e\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x46\x86\x3a\x5f\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\xd9\x90
	 - Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T060104XMIN
	 - no matching record layout found, try generic
	 - Record layout used :  T06NNNNXMIN
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00d700060241010458474436434b52395a3200000000000000000000000000000000000000005
		 0594c3043483830384d00000000000000000000000000000000000000001802190b2716020bb8
		 0c34000100005e3a08ff003300002de1097b00330000305900000000000000000000000000000
		 00000005e3a000001f5000056fa13860921001f00001c4c092a001f00001c6808f6002000001c
		 ac0fe40f9d0f8a000000000000000000005e3a000086e50000002000000020000000210000001
		 00000001000000011000000110000000000000000000000000000000000000000000000000000
		 002100000021000000000000000000000021000000001958030b015a01020000186c019c01db0
		 18e000001820c680c054e20002a0000c350000000000000000000000000000000000000000000
		 00003c0000000000000000000000005e3a000000220c350cb1000000000000000000000000000
		 00000000000000000000000000022000000220000002200000000000000000000000000000000
		 00000000000000000000000000000000000000002710000018690000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 0000000000000000000000000000000000000000000001f455280000000000000000000000000
		 00000000000abff
	 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
		 - record layout :  T06NNNNXMIN

	 - Grott data record date/time processing started
[... 180 lines cut ...]
			 "pf": 10000, "dcv": 0, "bdc1_sysstatemode": 0, "bdc1_faultcode": 0,
			 "bdc1_warncode": 0, "bdc1_vbat": 0, "bdc1_ibat": 0, "bdc1_soc": 0,
			 "bdc1_vbus1": 0, "bdc1_vbus2": 0, "bdc1_ibb": 0, "bdc1_illc": 0,
			 "bdc1_tempa": 0, "bdc1_tempb": 0, "bdc1_pdischr": 0, "bdc1_pchr": 0,
			 "bdc1_edischrtotal": 0, "bdc1_echrtotal": 0, "bdc1_flag": 0,
			 "bdc2_sysstatemode": 0, "bdc2_faultcode": 0, "bdc2_warncode": 0,
			 "bdc2_vbat": 0, "bdc2_ibat": 0, "bdc2_soc": 0, "bdc2_vbus1": 0,
			 "bdc2_vbus2": 0, "bdc2_ibb": 0, "bdc2_illc": 0, "bdc2_tempa": 0,
			 "bdc2_tempb": 0, "bdc2_pdischr": 0, "bdc2_pchr": 0, "bdc2_edischrtotal": 0,
			 "bdc2_echrtotal": 0, "bdc2_flag": 0, "bms_status": 0, "bms_error": 0,
			 "bms_warninfo": 0, "bms_soc": 0, "bms_batteryvolt": 0, "bms_batterycurr": 0,
			 "bms_batterytemp": 0, "bms_maxcurr": 0, "bms_deltavolt": 0, "bms_cyclecnt":
			 0, "bms_soh": 0, "bms_constantvolt": 0, "bms_bms_info": 0, "bms_packinfo":
			 0, "bms_usingcap": 0, "bms_fw": 0, "bms_mcuversion": 0, "bms_commtype": 0}}
	 - Grott MQTT topic used : energy/growatt
	 - MQTT message message sent
	 - Grott send data to PVOutput systemid:  MYSYSTEMID for inverter:  MYSERIAL
	 - Grott PVOutput send V1 disabled
		 -  {'X-Pvoutput-Apikey': 'MYAPIKEY', 'X-Pvoutput-SystemId': 'MYSYSTEMID'}
		 -  {'d': '20240225', 't': '11:39', 'v2': 2226.6, 'v6': 233.7, 'v5': 41.2}
	 - Grott PVOutput response:
		 -  OK 200: Added Status
	 - Grott Send data to Influx disabled
	 - Grott extension processing disabled
```

joris-weijters, 2024-02-29:
I worked around this by using the server time for now.

anoppe, 2024-02-29:
As I pointed out [in this issue](https://github.com/johanmeijer/grott/issues/442), the datalogger date time should be ignored and not be used to identify the metrics' date/time.

johanmeijer, 2024-02-29:
There are 2 ways to influence time behaviour.

Normally grott is using the time in the data record. This time is being set by the growatt server the moment the inverter start communicating (you have to set this in the growatt settings).

For influxdb grott will change the time to gmt time (monitoring tooling like grafana will display the data in right time zone). 
The gmt/utc offset is taken from the server offset. You can override this by setting a timezone in grott.ini. (e.g. tmzone= Europe/Amsterdam).

As specified above you can set grott to ignore the date/time from the data record by specifying time=server. Grott will use the time of the server it is running on.

joris-weijters, 2024-03-01:
This issue can be closed

closed 2024-03-01 by joris-weijters