johanmeijer/grott #528: Growatt SPF 5000 ES not working

tsf0x13, 2024-03-02:
Hi!
Growatt SPF 5000 ES not working correctly with Grott.

Here is data from api/cloud

KAM2H4A0QK
 
AC input frequency 49,94 Hz
AC input voltage 220 V
Ac output frequency 49,95 Hz
Battery percentage 100%
Battery voltage 55,67 V
Charge today 0,6 kWh
Grid charge current 0 A
Grid charged today 0,6 kWh
Grid discharged today 10,3 kWh
Grid out current 4,4 A
Import from grid 957 W
Import from grid today 10,9 kWh
Import from grid total 1 224,2 kWh
Lifetime load consumption 1 154,5 kWh
Lifetime Storage production 25 kWh
Lifetime storaged charged 89,3 kWh
Load consumption 978 W
Load consumption today 10,2 kWh
Load consumption(Solar + Storage) 1 059 VA
Load percentage 19,5%
Output voltage 220 V
PV charging voltage 0 V
Solar charge current 0 A
Solar current to storage 0 A
Solar power production 0 W
Storage charging/ discharging(-ve) -38 W
Storage production today 0 kWh

and in the attached file docker log
[grott.docker.log](https://github.com/johanmeijer/grott/files/14468706/grott.docker.log)

tabascoz, 2024-03-05:
Hi, i have this inverter, but i don't run grott or grottserver in docker. But from the log file you posted, it seems that grott is receiving the data. What is it not working?

johanmeijer, 2024-03-11:
I think you have to specify invtype=spf (for docker set the environmental at the docker run command with -e ginvtype="spf".

Hopefully this helps.

tsf0x13, 2024-03-31:
> I think you have to specify invtype=spf (for docker set the environmental at the docker run command with -e ginvtype="spf".
> 
> Hopefully this helps.

Thank you so much! Now is all ok!

johanmeijer, 2024-04-04:
Can we close this one?

tsf0x13, 2024-04-05:
sure! thanks you again. all works like a charm

closed 2024-04-05 by tsf0x13