johanmeijer/grott #530: documentation: simple docker command line does not work due to "."  at the end

typxxi, 2024-03-08:
I'm talking about this wiki page
https://github.com/johanmeijer/grott/wiki/Docker-support#simple-docker-startup-with-mqtt

As it is my first docker and mqtt adventure I have copied and pasted this line

`**docker run -d -p 5279:5279 --restart on-failure -e gmqttip="192.168.0.206" ledidobe/grott.**`

and then I got this
root@PiOS64bit:~# 

```
_**docker run -d -p 5279:5279 --restart on-failure -e gmqttip="192.168.178.26" ledidobe/grott.
docker: invalid reference format.**_
```

If I remove the . at the end I get this 

```
**_docker run -d -p 5279:5279 --restart on-failure -e gmqttip="192.168.178.26" ledidobe/grott
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
e3d974e2351012c8ae6e37d7b12a2f5fe9a6fece9dc52245df77fd73a7055924_**

I am running grott on a Pi 3B with the most recent Pi OS 64bit from december 2023 installed yesterday with all the updates available.
I wanted to use grottrpi but found out that it has moved from 2.8.2 on to a general version
https://hub.docker.com/r/ledidobe/grott

Therefore I had used this on the commandline: docker pull ledidobe/grott

Maybe this does not download the 2.8 general version but there was nothing mentioned in the wiki how to proceed and get the system installed right.

I feel a bit lost.

johanmeijer, 2024-03-11:
Just to be clear (I realize it is not). Since grott 2.8.2 (Beta) and higher I started creating a multiple architecture docker containers. 
For now these are: 

- ledidobe/grott:beta (or ledidobe/grott:2.8.3)
- ledidobe/grottserver:beta (or ledidobe/grott:2.8.3)

Ledidobe/grottrpi and ledidobe/grottserverrpi are still there while these are stilling being downloaded but will not be updated with the new releases.

pve, 2024-07-22:
so to be clear: ledidobe/grott is not yet multi architecture. or is it?

johanmeijer, 2024-07-22:
@pve The beta is multi architecture. I still have to promote it to latest. Will plan it for to do sometime the coming weeks.