johanmeijer/grott #537: TARGET REFUSED ERROR - from time to time for no reason

typxxi, 2024-03-20:
Good morning,
this morning I had no solar power results and when I checked grott which I have installed on a pi 3b runningn pios 64 latest stable from dec. 2023 downloaded last week I found a message like target refused error and had no data at all for all the 7 inverters active.

I had the same issue days before only once and so far it felt like those were caused by the grott.ini line about the mapping even though I had not changed anything and it had worked till noon yesterday.

So far i have opened grott.ini and saved it, then restarted grott script and then the sensors came back - I had not changed anything in that mapping line nor the mqtt password nor had I rebooted HA at that point. 

Not sure why, but it seems to happen from time to time that I come across target refused error when mostly only data of 1 inverter are missing, but each time a different one. 

Next time I will try to solve it just with a stop of the grott script and new start. But who knows when it would happen again or not. 
thanks for maintaining this piece of art - as it looks to me.

typxxi, 2024-03-21:
UPDATE: 

I faced the same issue and it took me about 3 or 4 hours to get behind cause in the last 2 days I tried to fix my setup on the Pi 3 which means : 

I wanted to start the terminal and the script at the end of the boot process in a visible terminal that I could check easily what is going on. I tried user, system Autostart, rc.local  and what not but nothing achieved. So I tried a lot.

And therefore I had to stop grott quite often, changed directories and what not.

When i went on I had to restart grott and I started my grott with
`sudo python3 /home/pi/grott/grott.py -v ` 

And if you start grott from OUTSIDE ITS OWN directory you will get an error, the **"target refused"** error
I do not know what is causing the issue, I simply cross checked it cause it did not disappear for hours regardless what I tried like editing grott.ini which seemed to have worked before but most likely I also had changed the directory. 

Therefore I guess it is at the end a bug or risk that has to be mentioned.

PeterKawa, 2024-03-22:
AAAAAAHHHH YES! You made my day, thanks a mil! 

While I had to reinstall my debian box, I couldn't get grott to send MQTT messages anymore, while it worked for months on the 'old' debian box
I start grott now with:
`cd ~/grott && python3 grott.py -v`
(grott folder resides in the root of my home folder)

Yeah, this is what I want to see:

> [image]

johanmeijer, 2024-04-04:
Can we close this one?