johanmeijer/grott #538: Grottserver not setting time registers

PedroKTFC, 2024-03-24:
I'm running grottserver and I'm trying to set the registers in my inverter dynamically using NodeRed so that I can optimise my export of power and to top up my battery if it's needed. However, I am getting some peculiar behaviour:
- If I use the Growatt app/website to set the times for when I want to force discharge, they often seem to get randomly reset to past values or values in other areas (eg from Battery first times)
- If I try to set a time register using a PUT, the register is never set although I get a success code back from the GET (eg I try to set register 1081 to 4667, 18:59, I get a 200 status code back yet when I use a GET to read the register, it has the value before the PUT)
- However, If I try to set a non-time register, it works (eg setting register 1082 to 1)
I'm doing this a register at a time and I suspect that, at least for time registers, it might be that you have to set both start and stop times in the same "transaction"? It's interesting the Growatt website seems to read all the registers in one go as there's very little delay in tapping the setting button and getting the data displayed. I've produced a version of grottserver that reads all the registers but it does it one-by one and takes several minutes.

PedroKTFC, 2024-04-04:
In case anyone other than me is following this, I've made some progress understanding this and even with an approach. It seems that to set a time register, you have to set all 3 registers for a time slot ie the start time, end time and enable/disable flag in a single write (PUT). I can only be sure for my inverter, type SPH.

I've updated my [grottserver ](https://github.com/PedroKTFC/grott/tree/Beta-(2.8.x)) in github so it's possible to update a time slot in a single go using the following syntax (the capitalised words are replaced with the appropriate values which must be decimal):

`//server:port/inverter?command=timeslot&startregister=STARTREG&inverter=INVERTER&starttime=STARTTIME&endtime=ENDTIME&enable=ENABLE
`

It does various checks to try to ensure that, for example, the start register is a genuine start time register, a time value is a time between 00:00 & 23:59 and the enable flag is 0 or 1.

I've also updated my read all registers to read the inverter registers in one go (I didn't know this was possible!!). The syntax for this is (and it's a GET):

`//server:port/registers
`
Remember, if you use this you do so at your won risk. I've no idea how other setups might react.

Feedback welcome.

PedroKTFC, 2024-04-05:
I've now improved the timeslot processing so it takes a human readable time value of hh:mm rather than an decimal integer variant of the hex value so the command should now be:

`//server:port/inverter?command=timeslot&startregister=STARTREG&inverter=INVERTER&starttime=HH:MM&endtime=HH:MM&enable=ENABLE 
`

molekuul, 2024-06-27:
The 1000 - 1124 registers only work for MIX, SPA and SPH inverters.
The  TL3-X(MIN, MOD, MAC Type) and the MOD TL3-XH inverters use a different set of registers for the timeslots
These only use 2 registers per timeslot 1 until 4, 3038/3039 until 3044/3045 and timeslot 5 -9 from 3050/3051 until 3058/3059

see: 
Growatt Inverter Modbus RTU Protocol  V1.24

PedroKTFC, 2024-07-03:
Are you trying my version? I'm afraid I don't have access to those inverters so can't develop and test something to work with them.