johanmeijer/grott #542: Shinewifi-x problem - URL resetting to default

J4CE2, 2024-03-28:
This is possibly not a GROTT issue. But I need help and there's no other place really.

As of yesterday it appears my shinewifi-x dongle keeps resetting back to the default growatt server URL. And this bypassing the GROTT local server I have setup. 

I've tried re-updating the dongle via the website, it works once and then stops. I've tried doing it via AP hot spot, I can see it has reset to default value, again it works once and then stops. 

Any one had anything similar??

J4CE2, 2024-03-29:
Actually I noticed something new in the GROTT Logs after the message is received. It tells me my dongle IP has disconnected. This never used to happen from what I can tell in the logs.

ar 29 08:10:35 rpi-monitor grott[1162]:          - Grott MQTT topic used : energy/growatt
Mar 29 08:10:36 rpi-monitor grott[1162]:          - MQTT message message sent
Mar 29 08:10:36 rpi-monitor grott[1162]:          - Grott Send data to PVOutput disabled
Mar 29 08:10:36 rpi-monitor grott[1162]:          - Grott Send data to Influx disabled
Mar 29 08:10:36 rpi-monitor grott[1162]:          - Grott extension processing disabled
Mar 29 08:10:36 rpi-monitor grott[1162]:          - Growatt packet received:
Mar 29 08:10:36 rpi-monitor grott[1162]:                   <socket.socket fd=7, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, la
Mar 29 08:10:36 rpi-monitor grott[1162]:          - Data less then minimum record length, data not processed
Mar 29 08:10:36 rpi-monitor grott[1162]:          - Growatt packet received:
Mar 29 08:10:36 rpi-monitor grott[1162]:                   <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, la
Mar 29 08:10:36 rpi-monitor grott[1162]:          - Data less then minimum record length, data not processed

Mar 29 08:11:05 rpi-monitor grott[1162]:          - ('192.168.1.30', 31980) has disconnected

J4CE2, 2024-03-29:
Theory so far:

- GROTT receives message, crashes
- Shinewifi-x resets to default URL if custom IP is not responding

The line
" ('192.168.1.30', 31980) has disconnected" seems to be consistent with terminating the service.

Will try run GROTT on a different device to test.

J4CE2, 2024-03-29:
I updated the IP field in the grott.ini file to be 47.91.92.81 (server-au.growatt.com) as opposed to the default 47.91.67.66.

It now appears to be running, I also changed update frequency to 2mins - so far so good.

Not really clear why this has helped yet.

lmdc45, 2024-03-29:
Thanks a lot for posting this, it seems to solve the same issue I had as well since yesterday

pimisen, 2024-03-29:
Same here. I drove me crazy this morning on Good Friday. Thanks for saving my day!

bryceIT, 2024-03-30:
absolute lifesaver! Thanks for sharing.
For anyone else using docker, just add the environment variable ip = 47.91.92.81 then go back into https://server.growatt.com/index and readd in your domain/port in the datalogger section to get information back into grott

J4CE2, 2024-03-30:
> edit: actually - even when setting the ip to 47.91.92.81 dongle is still resetting itself to the default url

At this stage not really sure what they are doing. Only for me and a few others since changing to the aussie IP it hasn't reset it self.

I thought this morning when the inverter powered on I may have issues, but so far all so good

I also am not seeing any of these in the Grott logs (when verbose is true) anymore
`Mar 29 08:11:05 rpi-monitor grott[1162]: - ('192.168.1.30', 31980) has disconnected`

FreemanJD, 2024-03-30:
> POSSIBLE SOLUTION
> 
> I updated the IP field in the grott.ini file to be 47.91.92.81 (server-au.growatt.com) as opposed to the default 47.91.67.66.
> 
> It now appears to be running, I also changed update frequency to 2mins - so far so good.
> 
> Not really clear why this has helped yet.
> 
> **I suspect that when they receive my message, if it went to server.growatt.com (i.e. the default grott IP address 47.91.67.66) , they update my dongle to server-au.growatt.com everytime. Very sneaky IF SO! It's possible server-au.growatt.com is new ... I had not seen this before from memory.**

Sorry. I tried to go into my grott.ini but the files is blank. 
what have I done wrong ? I’ve got the same issue where the dongle keeps getting reset to the Australian Server.

J4CE2, 2024-03-30:
> Sorry. I tried to go into my grott.ini but the files is blank. what have I done wrong ? I’ve got the same issue where the dongle keeps getting reset to the Australian Server.

How are you running GROTT? Docker? HA add on?

FreemanJD, 2024-03-30:
> How are you running GROTT? Docker? HA add on?

Sorry. I’m running Grott as a HA add on.

Neflardio, 2024-03-30:
> Sorry. I’m running Grott as a HA add on.
If your add-on is the same as mine, you should be able to expose config entries by enabling  "Show unused optional configuration options". Then it's a matter of adding the ip in the ggrowattip field.

[image]
[image]

KasperEdw, 2024-03-31:
solution at least running grott on Windows, I dont know of the same applies to the HA version. 

I changed the IP back to the grott server and set **blockcmd = True** in the grott.ini configuration file
It has been running fine since I did it a couple of hours ago. 

Best wishes, 
Kasper

J4CE2, 2024-03-31:
Good to know that's an option too. So far mines still going strong after changing the IP field only

Neflardio, 2024-03-31:
Also have no issues after changing ggrowattip to growatt au. I also left refresh rate at 1 min.

danielschwab, 2024-03-31:
Same issue here. Setting `blockcmd = True` in the `grott.ini` file seems to prevent the datalogger to switch back to the growatt server and disconnect from grott.  
This raises the question why the growatt backend wants the datalogger to use a different api server and if ignoring this request will have a negative effect.

faaarmer, 2024-04-01:
Same issue here - did what J4CE2 did and it seems to be working so far.

DonStone2188, 2024-04-01:
J4CE2’s solution worked for me so far.

closed 2024-04-02 by J4CE2

johanmeijer, 2024-04-04:
Nice you find a solution. I know there is also a request for allowing an URL in stead of a IP address for the growatt server. Maybe that will solve the problem also. I am working on a new version of Grott/Grottserver and that is at the wish list. But it a bigger project then I expected and finishing it might take a while. 

Be aware the blockcmd = True will block all commands (expect set time) from the growatt server. This is save if you do not want Growatt to control your inverter/datalogger.  But if you want to use the growatt website or phone app to make changes to your config (and update firmware) it will not work anymore.

tabascoz, 2024-04-04:
> Nice you find a solution. I know there is also a request for allowing an URL in stead of a IP address for the growatt server. Maybe that will solve the problem also. I am working on a new version of [...]

I confirm the blockcmd=True is able to prevent this change pushed by Growatt, as i have had a "good" Good friday :)

J4CE2, 2024-04-05:
> Nice you find a solution. I know there is also a request for allowing an URL in stead of a IP address for the growatt server. Maybe that will solve the problem also. I am working on a new version of [...]

Good point about blockcmd

For me the simple ip change was sufficient and still going strong.

RichardRamshaw, 2024-04-10:
Thanks so much for this, been running round trying to find this answer for days.

J4CE2, 2024-04-14:
> Just to comment, this issue is not only true for Australia. It happend to me in Germany as well.

Good to know.

Did you use a german IP / growatt server as a solution?

pimisen, 2024-06-22:
The Issue is back.
I (from Germany) had to change to this IP now:
 Server-au.growatt.com, with hostname set to Ec2-3-24-103-55.ap-southeast-2.compute.amazonaws.com, is assigned a AU server IP 3.24.103.55.
 But those aws host change randomly.
 I think it would be really cool, if hostnames in the grott.ini were allowed @johanmeijer :
Is there by the way anything know, when the docker image:latest will be the 2.8.3 Version?
(For smooth inplace upgrade, i don't want to set up the docker from scratch 😁)

pimisen, 2024-06-22:
That are the possible IPs (currently):
[3.24.103.55](https://www.ip-tracker.org/lookup.php?ip=3.24.103.55)  >>  ec2-3-24-103-55.ap-southeast-2.compute.amazonaws.com

[54.253.11.203](https://www.ip-tracker.org/lookup.php?ip=54.253.11.203)  >>  ec2-54-253-11-203.ap-southeast-2.compute.amazonaws.com

[13.239.127.14](https://www.ip-tracker.org/lookup.php?ip=13.239.127.14)  >>  ec2-13-239-127-14.ap-southeast-2.compute.amazonaws.com

J4CE2, 2024-06-22:
> The Issue is back. I (from Germany) had to change to this IP now: Server-au.growatt.com, with hostname set to Ec2-3-24-103-55.ap-southeast-2.compute.amazonaws.com, is assigned a AU server IP [...]

They have changed all of their IP Addresses. 

I changed grott.ini to use server-au.growatt.com rather than IP and I am back online

lmdc45, 2024-06-22:
Thanks again.

pimisen, 2024-06-24:
> > The Issue is back. I (from Germany) had to change to this IP now: Server-au.growatt.com, with hostname set to Ec2-3-24-103-55.ap-southeast-2.compute.amazonaws.com, is assigned a AU server IP 3.24.103.55. But those aws host change randomly. I think it would be really cool, if hostnames in the grott.ini were allowed @johanmeijer : Is there by the way anything know, when the docker image:latest will be the 2.8.3 Version? (For smooth inplace upgrade, i don't want to set up the docker from scratch 😁)
> 
> They have changed all of their IP Addresses.
> 
> I changed grott.ini to use server-au.growatt.com rather than IP and I am back online. Host names do work for me.

Not working here. If i use the DNS Name, it is not working.

J4CE2, 2024-06-24:
> Not working here. If i use the DNS Name, it is not working.

What DNS name are you using?

pimisen, 2024-06-24:
server-au.growatt.com

Neflardio, 2024-06-24:
The domain name wasn't resolving for me either (using HA addon veresion)

Using one of the new ip addresses worked fine though.

J4CE2, 2024-06-24:
ok, I am on version 2.7.8 it looks and run standalone version on a raspberry pi. 
it resolves server-au.growatt.com no problem

PedroKTFC, 2024-06-25:
I use grottserver and haven't connected to the Growatt servers for months!

r4tt4, 2024-06-26:
Well also got this problem again. First didn't recognize cause blockcmd=true 😉
But also after using server-au.growatt.com the re-configure tries are not gone.
_Grott: Record blocked:  0118_
Is there another DNS entry needed in germany?

pimisen, 2024-07-03:
I still have the issue that my docker (Docker Desktop on Debian 12) can't resolve the DNS names. Now they changed to dynamic IPs for server.growatt.com also use a cloud provider here also. currently it's a pain in the shoulder using it. No idea, why DNS resolving is not working for my docker setup