johanmeijer/grott #548: #enhancement add Fan power to SPF inverters

tabascoz, 2024-05-20:
Hi there! 

Would it be possible to include the Fan1 value readings for SPF inverters? 

this inclusion in grottconf.py after line 942 did the trick

`"BatWatt"           : {"value" :394, "length" : 4, "type" : "numx", "divide" : 10} `

Thanks, Fennando

johanmeijer, 2024-05-21:
Hi Fennando. I like to at this but if I like in grottconf (2.7.8 and 2.8.3) this is already in there. Do I miss something?

tabascoz, 2024-05-21:
sorry Lol, posted the wrong line by mistake although i mentioned the right value. 

The right line: 

`"Fan1"              : {"value" :494, "length" : 2, "type" : "num", "divide" : 1}`

I checked and recloned the repo and it is not in v 2.8.3

Cheers

johanmeijer, 2024-06-18:
Dan. I only called it  invfanspeed (to match the documentation).

tabascoz, 2024-08-02:
Thanks Johan, all working fine !

closed 2024-08-02 by tabascoz