johanmeijer/grott #551: MOD 6000TL3-XH recognized but no MQTT to HA

InvertedMinor, 2024-06-23:
Great application, deep respect for all the work.

The data from MOD 6000TL3-XH is flowing into the Grott Beta branch (2.8) Current version: 95acf9c (see below) but no MQTT data is received. I have tried all of the inverter types, min is giving a meaning full data set in the log but not in HA MQTT.

Any suggestions?

Logs below:
 
Grott external record whitelist 'recwl.txt' not found

Grott records whitelisted :  {'5250', '501b', '5216', '5005', '5105', '5116', '0119', '5229', '0120', '5004', '5016', '0150', '5050', '5119', '5129', '0105', '5003', '5219', '5104', '0103', '0104', '5103', '0116', '5019', '5150'}

Grott settings:

_Generic:
	version:             	 2.8.3
	verbose:             	 True
	trace:               	 False
	config file:         	 /config/grott/grott.ini
	minrecl:             	 100
	decrypt:             	 True
	compat:              	 False
	invtype:             	 min
	invtypemap:          	 {}
	include_all:         	 False
	blockcmd:            	 False
	noipf:               	 False
	time:                	 auto
	sendbuf:             	 True
	timezone:            	 local
	valueoffset:         	 6
	offset:              	 6
	inverterid:          	 automatic
	mode:                	 proxy
	grottip              	 default
	grottport            	 5279
_MQTT:
	nomqtt               	 True
	mqttip:              	 localhost
	mqttport:            	 1883
	mqtttopic:           	 energy/growatt
	mqttmtopic:          	 False
	mqttmtopicname:      	 energy/meter
	mqttinverterintopic: 	 False
	mqtttretain:         	 True
	mqtttauth:           	 True
	mqttuser:            	 mosquito
	mqttpsw:             	 **secret**
_Growatt server:
	growattip:           	 47.91.67.66
	growattport:         	 5279
_PVOutput:
	pvoutput:            	 False
	pvdisv1:             	 False
	pvtemp:              	 False
	pvurl:               	 https://pvoutput.org/service/r2/addstatus.jsp
	pvapikey:            	 yourapikey
	pvinverters:         	 1
	pvsystemid:          	 systemid1
_Influxdb:
	influx:             	 False
	influx2:            	 False
	database:           	 grottdb
	ip:                 	 localhost
	port:               	 8086
	user:               	 grott
	password:           	 **secret**
	organization:       	 grottorg
	bucket:             	 grottdb
	token:              	 **secret**
[... 129 lines cut ...]
	 - Record layout used :  T060138XMIN
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 006a0006023d01385847443643473732464500000000000000000000000000000000000000004
		 a434d30444136303039000000000000000000000000000000000000000018061713173b0613d8
		 13ff0000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000001400142700000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 0000000000000000000000001428144f000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000145014770000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000001478149f00000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000014a014c7000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00078a5
	 - Grott data ack record or data record not defined no processing done

	 - Growatt packet received:
		  <socket.socket fd=5, family=2, type=1, proto=0, laddr=('172.30.33.13', 5279), raddr=('192.168.2.29', 11909)>
	 - Data less then minimum record length, data not processed

ZonPV, 2024-06-23:
You better try in your config

nomqtt = False

As in your log there is the sentence MQTT is disabled.

InvertedMinor, 2024-06-23:
already done in my grott.ini

`[MQTT]
# Mqtt parameters definitions
# Be aware nomqtt = True means no MQTT processing will be done!!!!!!

nomqtt = False
ip = localhost
port = 1883
topic= energy/growatt
auth = True
user = mosquito
password = QQQQQ
retain = True`

So why is it then switching to True?

InvertedMinor, 2024-06-29:
Issue resolved. Inverter now 100% visible in HA. Confusion on my side with all the variants and options of the Grott server & HA.

closed 2024-06-29 by InvertedMinor