johanmeijer/grott #552: Growatt NEO 800M-X Support / Protocol

robertzaage, 2024-07-02:
I bought a Growatt NEO 800M-X and tried to connect it to a grottserver. Unfortunately grott proxy couldnt decode the data and therefore dropped it.

The factory endpoint is 'mqtt.growatt.com:7006' which looks like an TLS+MQTT endpoint. I analyzed the traffic with this server and saw a tls handshake and sometime a routine which downloaded (root) certificates from digicert. I'm currently tcpdumping the traffic into a file, just in case it helps to find patterns. Worst part, if even possible, would be breaking up the TLS. One attack vector may be the mentioned certificate update routine. The whole fw can't be too complex, as it's an Espressif  based system...

Has anyone here experience with the NEO-class devices?

robertzaage, 2024-07-02:
Offtopic: I set `trace = True` in `grott.ini` but the trace mode didn't turn on. What is this mode about and is this a bug?

robertzaage, 2024-07-27:
@fdreckmann no, i currently think of buying another 800M-X just for tearing it down and dump the firmware.

CalvinSchwartz, 2024-08-28:
Before reading this issue I set my Neo 800M-X to the Grott server. However I can't connect it to the Growatt servers anymore as there's currently the (non-working) Grott Server set.

I already tried resetting via the Shine app (setting to mqqt.growatt.com:7006) however the connection to the server fails. Any ideas how one can reset the Wifi Module to connect back to the Growatt servers?

robertzaage, 2024-08-28:
> Before reading this issue I set my Neo 800M-X to the Grott server. However I can't connect it to the Growatt servers anymore as there's currently the (non-working) Grott Server set.
>
> I already tried [...]

Maybe a typo: mqtt.growatt.com:7006

CalvinSchwartz, 2024-08-28:
Typo was only on GitHub 😕
I did set the server of the Datalogger in the settings section on server.growatt.com

Using the shine app to connect the Datalogger back to Wifi/Growatt Server there was already mqtt.growatt.com filled in. Changed it to server.growatt and then back to mqtt.growatt.com however there's still no connection to the growatt servers possible. I'll contact the growatt support to see if there's any option to completely reset the datalogger to factory settings.

robertzaage, 2024-08-29:
> @fdreckmann no, i currently think of buying another 800M-X just for tearing it down and dump the firmware.

I tried to dissemble a unit. It has two solid metal halves glued together - impossible to separate and the case is completely filled with grey gum. So no luck here...

fdreckmann, 2024-09-20:
> > @fdreckmann no, i currently think of buying another 800M-X just for tearing it down and dump the firmware.
> 
> I tried to dissemble a unit. It has two solid metal halves glued together - impossible to separate and the case is completely filled with grey gum. So no luck here...

Do you have see this addons 
https://github.com/mtrossbach/noah-mqtt

Maybe you can Work together?

robertzaage, 2024-09-20:
> Do you have see this addons https://github.com/mtrossbach/noah-mqtt
>
> Maybe you can Work together?

Afaik this addon just polls the growatt API and is no offline solution or proxy.

fdreckmann, 2024-09-20:
> Afaik this addon just polls the growatt API and is no offline solution or proxy.

That's how I understand it too. Then only the addons description would be wrong: 

"Integration Growatt Noah 2000 Home Batteries with Home Assistant via MQTT, Enabling Real-Time Monitoring and Control of Your Energy Storage System."

Here it is called "real-time". 
But I guess you're right.

johanmeijer, 2024-09-23:
Yes that solution works with the Growatt API and pulls information from the growatt server. Nothing wrong with that but it is a total different solution then Grott that act as a proxy and intercepts the data the moment is being send (and cannot block/filter the commands comming from growatt). 

At this moment I am not able to decode the Noah information. Maybe I will try i again later and get a brilliant idea. If somebody can give a hint it wil be very welcome.

FranzSchi, 2024-10-13:
Hi,
I guess i just faded the same thing, just a a small question to the other affected users.
Do you also see the regular dialogue in the https://server.growatt.com/index Dialogue?
i changed here something but it seems it had no effect at all.
>> Is it the same in your setup?

Then I’ve tried to reconfigure using the app and faced the mqtt hostname

[image]

deiberger211, 2024-10-13:
Hello,
I also have an Neo 800M-X inverter and it is not working:. Anyone an idea how to fix this?

	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.5', 51102), raddr=('47.254.130.145', 5279)>
	 - Grott - grottproxy - Invalid data record received, processing stopped for this record

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.30.33.5', 60672), raddr=('47.254.130.145', 5279)>
	 - Grott - grottproxy - Invalid data record received, processing stopped for this record

	 - Growatt packet received:
		  <socket.socket fd=8, family=2, type=1, proto=0, laddr=('172.30.33.5', 53730), raddr=('8.209.71.240', 5279)>
	 - Grott - grottproxy - Invalid data record received, processing stopped for this record

deiberger211, 2024-10-13:
> Hello, I also have an Neo 800M-X inverter and it is not working:. Anyone an idea how to fix this?
>
> ```
>  - Growatt packet received:
> 	  <socket.socket fd=4, family=2, type=1, proto=0, [...]

Ok in the mean time I have understood that this kind of inverter is not supported yet.

FranzSchi, 2024-10-23:
As this Microinverter allows no other thing to be plugged in.
The only possibility so far is using https://github.com/PLCHome/growatt libs, which read from cloud.
I’ve found this docker implementation which reads from the cloud and posts into MQTT.
https://hub.docker.com/r/bbme/growattnpm

Anyone knows a better solution so far for this microinverter

robertzaage, 2025-02-15:
> [image]
> 
> I bit the bullet and forcefully opened a unit. It has a ESP C3 13U and traces for a UART header. I will dump the FW later this week.

I dumped the Firmware over the JTAG-USB port (GPIO18/19) as UART was still dead after populating a header. They use MQTT+TLS and verify certificates as I suspected earlier. Looks like the only way to get this thing off the cloud is to flash an custom firmware. If anyone wants to look into the firmware and help me reversing it and/or to develop a replacement please contact me.

robertzaage, 2025-02-15:
TLDR:

It's unlikely the NEO 800M-X will ever get support by this project. Growatt secured the device against MITM attacks and is encrypting the MQTT traffic validated by officially trusted certs.

closed 2025-02-15 by robertzaage

johanmeijer, 2025-02-16:
Yep this will be hard (unpossible) for Grott to support I think. Thank you for investigating.

mtrossbach, 2025-02-19:
Hi Robert,

That sounds really interesting! I’m particularly curious about the firmware, as I suspect it’s using the same mechanism as the NOAH 2000 devices. I’m not very experienced in firmware reversing, but I’d love to take a look and try to understand how it works if I find the time.

Would you be able to share your dump somehow? That would be much appreciated!

Thanks,
Markus

hacst, 2025-02-28:
> > Growatt [...] is encrypting the MQTT traffic validated by officially trusted certs.
> 
> @robertzaage What about the trusted certs? Have you extracted the trust store too? If there is a "public" CA (like Lets Encrypt, ...) in there it could be possible to:
> 
> 1. Register a domain like "mqtt.grott.com"
> 2. Get a TLS certificate for it
> 3. Share it with the community (the whole keypair, everyone can impersonate "mqtt.grott.com")
> 4. Locally override the DNS name "mqtt.grott.com" with the local grott server IP
> 5. Configure NOAH2000 / NEO 800M-X to use "api.grott.com" (as i understand the whole previous conversation, this should be possible)

There is a trust store in there with the let's encrypt root certificate. Cannot say for sure it is used for that specific validation yet. @robertzaage mentioned he tried with a LE cert before, but was not sure at what part of the testing. So retrying with a LE cert and a mqtt server behind sounds like it still has a good chance of actually working. Personally currently do not want to risk my device connectivity outside of a test setup, so that is the direction I'm currently investigating. I am more interested in custom firmware anyways.

fdreckmann, 2025-02-28:
A very good idea.. I have an untouched WR here. I could test it if I wanted. I still don't understand on wich Point in can Set api.growatt at the WR ?

CalvinSchwartz, 2025-02-28:
@fdreckmann I set the regular Grott IP adress on the Web Panel and basically bricked my device. Once there is a new api server the NEO tries to connect to it. I found no way to reset it to the regular server as it obviously tries to connect to my (not working due to the limitations mentioned here) Grott server. So before you try it on the web panel be sure you know how to reset it.

BTW if anyone knows how to reset the NEO to work with the regular Growatt server I would appreciate it. See: [Comment on Auf 28, 2024](https://github.com/johanmeijer/grott/issues/552#issuecomment-2315678126)

hacst, 2025-02-28:
Definitely do not set the IP. That will make you fail domain validation for the cert. There is a separate "domain" setting which has a chance to work assuming it changes what domain is validated against. As I mentioned I am not taking that risk for my device right now. The device is heavily potted and withour heroic destructive effort like from @robertzaage you won't be reflashing it through some electrical interface. Ideally such a sacrifical device can be used to investigate ways to intercept & reflash in safe ways so we minimize (or ideally get rid of) the danger of bricking devices we cannot realistically get open and back together as before again.

hacst, 2025-02-28:
@CalvinSchwartz So it is connecting to the IP you set? What you could try, is to low-level forward the IP connecting to the original growatt IP. Basically playing NAT between you and the growatt mqtt server. That way you might restore connectivity to their service and get the ability to unset/reset the IP.

PSMGoossens, 2025-03-09:
> If anyone wants to look into the firmware and help me reversing it and/or to develop a replacement please contact me.
@robertzaage Please if you have the possiiblity! I expect that the firmware is also encrypted but we always could try it!

Trackhe, 2025-03-21:
@robertzaage i would be very interested in that fw dump.

sphr2k, 2025-03-23:
> There is a trust store in there with the let's encrypt root certificate. Cannot say for sure it is used for that specific validation yet.

I don't think Let's Encrypt is used. `mqtt.growatt.com:7006` uses the same DigiCert-issued Wildcard cert `*.growatt.com` that's also use for the other endpoints (APIs, website):

```
$openssl s_client -connect mqtt.growatt.com:7006 

Connecting to 47.254.130.145
CONNECTED(00000008)
depth=2 C=US, O=DigiCert Inc, OU=www.digicert.com, CN=DigiCert Global Root G2
verify return:1
depth=1 C=US, O=DigiCert Inc, OU=www.digicert.com, CN=Encryption Everywhere DV TLS CA - G2
verify return:1
depth=0 CN=*.growatt.com
verify return:1
```

hacst, 2025-03-25:
@sphr2k by now have a working test setup and a let's encrypt certificate is accepted by the inverter. If you setup a mosquitto with TLS+MQTT (not websocket) it will connect to it and start sending packets to a topic on there. Contents look a lot like they are at least derived from what grott can decode. Same masking but the grott decoder does not recognize the layout inside. Might just be a matter of establishing the correct mapping for this device. Note that there is also a different style of packet it sends from time to time and I have not run this outside of the test setup and even there only for short periods of time. No idea whether the inverter will get unhappy after a while if not connected to a real server.

joni347, 2025-03-26:
@hacst Were you able to reset the mqtt server to the growatt server? Which topics are displayed in the mqtt broker? I have a Noah 2000 and would love to connect it to my mosquito. I can happily do without the growatt cloud.

hacst, 2025-03-26:
@joni347 my test setup never talked to real growatt servers or the growatt app. Also it only reflects NEO 800M-X. I have not tried with my actual "production" inverter.

On the broker it created a single topic "c/33/<deviceid>". Device ID is the QMNXXXXXXXXXXXXX serial of the device/datalogger.  No idea what the 33 means. Packets mostly go in there in what looks like to be the configured datalogger interval.

You ought to be able to change the server domain at any time using the bluetooth mode from the app. No guarantees though.

joni347, 2025-03-26:
Ok, understood.
Would you share the data? 
Do you think it is possible that they only package their modbus in mqtt?

hacst, 2025-03-26:
I'll try to do a proper collection run so I have something others can usefully look at soon. The modbus protocol is documented and it is what neo 800M-X iot processor uses internally to talk to the actual inverter portion of the device. You can imagine something similar to a shine wifi-s stick but built directly into the same enclosure. I would be surprised if they have not mostly repurposed their existing protocol towards the server while adding mqtt around it. I am not sure if that format is derived from the modbus registers. Is that the case for what grott supports already? Got sidetracked before I was able to take a proper look.

joni347, 2025-03-26:
That's what I thought. I program low-level drivers for microcontrollers for a living. So I'm familiar with binary code. If you have some data sets, I can help you. 
For example, the Noah 2000 has a connector to connect it to other Noahs. This connector consists of 2 big and 3 small pins. I could imagine that there is a Modbus via UART on the 3 small pins. I wanted to look at this with a logic analyzer, but it would be nicer to be able to receive and send the data directly via mqtt without having to build an extra device.

robertzaage, 2025-03-28:
Thanks to the hard work of @hacst, we were able to decode and bridge NEO-series MQTT messages to Home Assistant. See here: https://github.com/robertzaage/GroBro

joni347, 2025-03-28:
Do you think you can also implement the Noah 2000? Or what would you need for it?

robertzaage, 2025-03-28:
> Do you think you can also implement the Noah 2000? Or what would you need for it?

We need dumped MQTT messages. Set up an MQTT+TLS Mosquitto instance (or I can set one up for you), and try pointing the battery to it in the *Advanced > Server settings* on the ShinePhone app. If the certificate chain is valid, the battery will start publishing packets. If you dump them, we can decode and analyze the registers.

joni347, 2025-03-28:
@robertzaage I think the easiest way would be if you provide me with a mqtt server. But what worries me a bit is, were you also able to restore the neo to the growatt mqtt server?

fdreckmann, 2025-03-29:
Integration to grott Server possible?

robertzaage, 2025-03-29:
> [@robertzaage](https://github.com/robertzaage) I think the easiest way would be if you provide me with a mqtt server. But what worries me a bit is, were you also able to restore the neo to the growatt mqtt server?

The ShinePhone app uses Bluetooth to configure the settings on the inverter. So there is no problem to restore the original server later.

Can you send me a mail and open a feature request on the GroBro repo?

Fexiven, 2025-04-15:
> I dumped the Firmware over the JTAG-USB port (GPIO18/19) as UART was still dead after populating a header. They use MQTT+TLS and verify certificates as I suspected earlier. Looks like the only way [...]

I would like to take a look at the firmware too. Could you please share it with me?