johanmeijer/grott #553: Add solar storage Noah 2000 an option ?

BArweiler, 2024-07-08:
Hello,
thanks for providing this nice project.
It would fit in my smart home landscape when I would have a Growatt inverter.
I bought the new storage Noah2000 ... 
It would be great when you could check whether this could be added too...

I'm currently rethinking also changing from Deye inverter to the neo from Growatt.. but this is an other story

Thanks for your thoughts and feedback 
BR

BArweiler, 2024-07-08:
By the way... Noah 2000 based on the same protocol as the Growatt NEO 800M-X  which was opened by robertzaage

johanmeijer, 2024-07-29:
See also discussion #560

joba-1, 2024-08-02:
@BArweiler what do you mean by "opened"? Physically opened or somehow understood the protocol? And who is robertzaage?

BArweiler, 2024-08-02:
robertzaage is a user how mentioned that the neo inverter is als based on the tls encrypted protocol as it seems to be the case at Noah 2000 as well. 
#560 seems to address it as well

joba-1, 2024-08-03:
Thank you! 
There is nothing wrong with encrypting my data before sending it over the internet, but if I cannot know the key (like downloading it from my account), that is not my data anymore. 
Looked good, but as long as I cannot decrypt my own data these products are not for me :(