johanmeijer/grott #557: Data-Logger resets back to growatt servers

splitbrain, 2024-07-16:
I have grott running successfully for months, but today my Shine-Lan-X reset back to the growatt servers twice already.

I suspect that Growatt is tinkering with the configuration by periodically resetting the network settings. Would it be feasible to filter what data Growatt is allowed to send back to the datalogger (and inverter)? I would like to be able to prevent them from willy-nilly updating firmwares or reconfiguring stuff without me knowing.

Pelmo23, 2024-07-16:
same issue today :(

Steamerzone, 2024-07-16:
Same issue today.

phil0002, 2024-07-16:
Same issue since 8:42 am :( 

Maybe something changed at Growatt cloud?

splitbrain, 2024-07-16:
Seems to be a duplicate of https://github.com/johanmeijer/grott/issues/542

The solution is to set the new IP server.growatt.com resolves to in the ggrowattip option. Ideally one should be able to set the DNS name there but people report that DNS resolution doesn't work in the Homeassistant Addon (haven't tried yet).

Alternatively (and that answers my question) remote commands can be disabled via the `gblockcmd` option.

phil0002, 2024-07-16:
Don't understand the solution. 
I have running it in docker. 
Any help where to set the new IP? In the datalogger I am already doing it.

johanmeijer, 2024-07-16:
The challenge is that Growatt resetting the shine-lan back from sending data via grott to sending data directly to growatt servers (by changing IP or URL in the shinelan). 

So you do have to a setup of the shine-lan (shinewifi or shinelink) again. Or change the IP via the website/shinephone(?)  configuration dialogue. 

The only way to prevent this to enable command blocking in Grott so that Growatt can not change your configuration. 
That can be done by blockcmd = True in the .ini or -e gblockcmd = "True" in docker run command or set this with docker compose.

johanmeijer, 2024-07-16:
Reading true the other issue I do understand it more. Grott uses by default a old or wrong IP address for your region. 

There seems to be another way to prevent growatt to prevent reset the IP address and that is the right IP growatt ip address in the grott.ini by  specifying grottip = <new ip address> or -e ggrottip = in docker. 

Lets see if we can prevent this in the future y not using an ip but an URL in grott. I have to change grott a bit for that.

egguy, 2024-07-16:
Hi @johanmeijer, 

Regarding the use of the DNS, it seems it is already possible to use it in your code, throught the `.ini` file [as pointed by dante2 ](https://github.com/johanmeijer/grott/pull/216#issuecomment-1303646696).

For the env variable, the call to `ip_address` seems to be preventing using a DNS in the env variable: https://github.com/johanmeijer/grott/blob/Master/grottconf.py#L454C17-L454C37

Thanks for the information about the `blockcmd` and your software

blauwblaatje, 2024-07-17:
> So you do have to a setup of the shine-lan (shinewifi or shinelink) again. Or change the IP via the website/shinephone(?) configuration dialogue.

Curious, where can I find it on the website. I set up my shine-lan a year ago and can remember the IP was easy to set somewhere, but I can't find it anymore... So I'm wondering if they removed the setting.

Jokrok, 2024-07-17:
Best solution would be that grott intercepts this command and sets the server ip accordingly.

johanmeijer, 2024-07-17:
@egguy 

A thanks for pointing me to the env ip coding part. I will change that. 

And I will see if I can find an URL that we specify as default so no additional parameter settings are needed.

egguy, 2024-07-18:
It seems growatt is updating its infrastructure. 

Depending on where you are, the `server.growatt.com` seems to resolve to a different place: https://github.com/egguy/addon-grott-beta/issues/51#issuecomment-2233616991

My current resolution seems to point in the direction of Growatt going to use MQTT in Australia:
```bash
$ dig server,growatt,com
...
;server.growatt.com.		IN	A

;; ANSWER SECTION:
server.growatt.com.	10	IN	CNAME	mqtt-au-e2224fa6621725d9.elb.ap-southeast-2.amazonaws.com.
mqtt-au-e2224fa6621725d9.elb.ap-southeast-2.amazonaws.com. 10 IN A 54.253.11.203
mqtt-au-e2224fa6621725d9.elb.ap-southeast-2.amazonaws.com. 10 IN A 3.24.103.55
mqtt-au-e2224fa6621725d9.elb.ap-southeast-2.amazonaws.com. 10 IN A 13.239.127.14
```

Jokrok, 2024-07-18:
Please keep up the good work, The integration of grottserver and grott will make all this a lot easier.
Thanks for your quick reply.

mmx01, 2024-07-21:
The same is happening to me and I was pulling my hair out. I am using Grott on ubuntu VM on hyper-V with shine wifi-f and it was rock solid for months. Now every few moments my Home Assistant stops seeing updates. I started suspecting DNS, vswitch, firewall etc. and indeed after troubleshooting what pointed me here was that growatt website got updates and Grott/Home Assistant did not. Wifi stick defaults back to server.growatt.com after few moments. 

What worked for me (in EU) is that server.growatt.com resolves to few addresses some from 8 A class and some from 47 A class 
C:\Users\User>ping server.growatt.com
Pinging nlb-nbtf7vlspietfgd3ro.eu-central-1.nlb.aliyuncs.com [47.254.130.145] with 32 bytes of data:
Reply from 47.254.130.145: bytes=32 time=57ms TTL=90

In grott.ini I set fixed IP instead of this dns load balancing and for now it works for about an hour without issues.

[Growatt]
ip = 47.254.130.145
port = 5279

What I suspect is that this is new target shine server IP for my country/region and when stick connects to another IP resolved by this DNS considered old server, it responds with configuration command to set server to DNS server.growatt.com. Perhaps Growatt is "broadcasting" this new address to sticks to ease migration. It may be possible some/older sticks came with hardcoded IP addresses or other DNS entries they are overriding this way.

Otherwise I don't see why some servers (8 A class) would reply with reconfiguration command while others (47 A class) do not. 
Let me know if you configure in EU server from class 47 and that fixes it for you too. Would support assumption that's part of some migration process.

M.

pve, 2024-07-22:
This works for me from NL.
ip = 47.254.130.145
though I am using docker compose
            - ggrowattip=47.254.130.145

It seems I had to set the grott server address to manual instead of auto on the dongle itself (in AP mode).

egguy, 2024-07-22:
Couldn't try as don't have a running inverter with me, but using the `.ini` file. You can put:
`ip = server.growatt.com`

The [ `.connect()`](https://docs.python.org/3/library/socket.html#socket.socket.connect) python call should resolve:

---
A pair (host, port) is used for the [AF_INET](https://docs.python.org/3/library/socket.html#socket.AF_INET) address family, where host is a string representing either a **hostname in internet domain notation like 'daring.cwi.nl'** or an IPv4 address like '100.50.200.5', and port is an integer.

--

You can't use it in an environnement variable (configuration of the addon) as there is a check for a valid IP, not DNS

mmx01, 2024-07-22:
@egguy the issue with using DNS name at the moment is that server.growatt.com resolves to various different IPs/ growatt servers. Some servers behind server.growatt.com issue a stick server reconfigure command back to the stick. This then breaks proxy setup as stick no longer points to the grott proxy but directly to growatt so we lose all stats/updates handled by grott. 

This seems to be the case regardless of whether grott runs on HA, as docker or standalone since issue comes from growatt side with remote reconfigure. I did not want to block processing all commands in grott.ini since I sometimes do use them and this "hack" with manual IP setup in grott.ini appears to work. Good news is that 47.254.130.145 still works for a day and stick has not been reconfigured remotely - previously it happened every few minutes.

M.

peters59, 2024-07-22:
> Das funktioniert für mich von NL aus. IP = 47.254.130.145, obwohl ich Docker Compose verwende – ggrowattip=47.254.130.145
> 
> Anscheinend musste ich die Grott-Serveradresse am Dongle selbst (im AP-Modus) auf manuell statt automatisch einstellen.

> This works for me from NL. ip = 47.254.130.145 though I am using docker compose - ggrowattip=47.254.130.145
> 
> It seems I had to set the grott server address to manual instead of auto on the dongle itself (in AP mode).

Does the web server for manual input still work for you in AP mode? For me, port 80 no longer exists in AP mode.

gsnhf, 2024-07-22:
I fixed it with blockcmd last week. It works fine. 

But now the connection ist broken again.

PedroKTFC, 2024-07-22:
> Please keep up the good work, The integration of grottserver and grott will make all this a lot easier. Thanks for your quick reply.

I already use grottserver all the time (with some personal tweaks) and haven't sent any data to China for some months now.

phil0002, 2024-07-22:
@johanmeijer would it also work when I add an DNS entry in my pihole for server.growatt.com to the right IP?
It centralized and can be changed without reboots. IF grott is using the url to connect and not any hardcoded IPs.

johanmeijer, 2024-07-22:
@phil0002 

I am going to test this. But probably it will work.

johanmeijer, 2024-07-22:
@phil0002,  I changed the default growatt IP address to url: growatt.server.com. 

Hopefully this will prevent growatt from changing IP address and the need for changing IP address in grott.ini

Be aware if you do not want Growatt to change your configuration you have to specify: 

blockcmd = True 

In the grott.ini 
(for docker -e gblockcmd = "True"

johanmeijer, 2024-07-22:
BTW: I think 2 IP addresses will work fine (from the Netherlands)

- 47.254.130.145
- 8.209.71.240

pve, 2024-07-22:
> @pve Does the web server for manual input still work for you in AP mode? For me, port 80 no longer exists in AP mode.

don't know. after applying the setting it flips out of AP mode. have not have time to test your scenario.

mmx01, 2024-07-23:
I am running wifi shine-f with fw 3.0.0.2 and web server in AP mode works okay on port 80. There are still options to configure stick with either IP or DNS via web UI.

Also if you don't use blockcmd you can reconfigure stick via growatt's official website from your chair.  When in Dashboard scroll down to My Photovoltaic Devices, do not click on the device (configuring inverter settings) but click on All Devices instead. There you will have a list of all dataloggers with option to configure datalogger settings. This gives you what web UI in AP mode does and not inverter settings. No need to reset to AP mode and run around the setup.

peters59, 2024-07-23:
> I am running wifi shine-f with fw 3.0.0.2 and web server in AP mode works okay on port 80. There are still options to configure stick with either IP or DNS via web UI.
> 
With firmware version 3.1.0.5 this unfortunately no longer works, so you can no longer enter a fixed IP and the update interval cannot be set either.

PeterKawa, 2024-07-23:
> BTW: I think 2 IP addresses will work fine (from the Netherlands)
> 
> - 47.254.130.145
> - 8.209.71.240

Only 47.xxx works (NL), Johan.
I also enabled blockcmd. Let's see for how long it keeps on running (Growatt not touching it) 😉

Thx 4 this great tool and your work!

johanmeijer, 2024-07-24:
> With firmware version 3.1.0.5 this unfortunately no longer works, so you can no longer enter a fixed IP and the update interval cannot be set either.

You can also do not do this via the growatt website?

peters59, 2024-07-27:
> You can also do not do this via the growatt website?
No, on the website I can only change the IP address, domain and port, nothing else.

btw, the IP address is unfortunately still reset every few hours. despite

blockcmd = True

I tried it with both IPs:

- 47.254.130.145
- 8.209.71.240

pve, 2024-07-27:
I run Grott with docker-compose. 
These environment variables work for me. No reset in the past days.

        environment:^M
            - gblockcmd=True^M
            - gnomqtt=False^M
            - gmqttip=127.0.0.1^M
            - gmqttauth=False^M
            - gmqttuser=growatt^M
            - gmqttpassword= growatt2020^M
            - gpvoutput=False^M
            - ggrowattip=47.254.130.145
            - gpvapikey=yourapikey ^M
            - gpvinverterid1^M
            - gpvsystemid1=12345^M

johanmeijer, 2024-07-28:
> > You can also do not do this via the growatt website?
> > No, on the website I can only change the IP address, domain and port, nothing else.
> 
> btw, the IP address is unfortunately still reset every few hours. despite
> 
> blockcmd = True
> 
> I tried it with both IPs:
> 
> * 47.254.130.145
> * 8.209.71.240

Are you sure the configuration has been read by Grott. With blockcmd = True Growatt should not be able to reset your datalogger.

peters59, 2024-07-28:
This is my grott.ini:

```
[Growatt]
blockcmd = True
#IP = 47.254.130.145
IP = 8.209.71.240
Port = 5279
```
Yesterday the setting worked for about 14 hours, after which the logger no longer sent to the set IP

typxxi, 2024-07-28:
blockcmd will prevent you from editing or changing the inverter settings. 
And I bet you will forget about that feature and in 10 months you will open a complain.

For me the growattip setting alone is doing the trick right now, no blockcmd needed with that ip.

[Here](https://github.com/muppet3000/homeassistant-grott/issues/91#issuecomment-2254581671) is the description of my HA grott addon configuration for those running the HA grott addon - but from the configuration or UI side without touching the grott.ini file.

johanmeijer, 2024-07-29:
> This is my grott.ini:
> 
> ```
> [Growatt]
> blockcmd = True
> #IP = 47.254.130.145
> IP = 8.209.71.240
> Port = 5279
> ```
> 
> Yesterday the setting worked for about 14 hours, after which the logger no longer sent to the set IP

The blockcmd=True parameter sould be in the [Generic] part of the .ini. I agree with @typxxi it might be better to configure:

     ip = server.growatt.com

Then growatt will not have to need to reset ip address in the datalogger.

valdenny, 2024-07-29:
I resolved it with "NAT traversal".
Let's see how it goes.
The Chinese need to understand that if we buy an object, we are the owners of that object.
And as owners, we can do whatever we want with it, even break it.
The Chinese will never have me! :)

peters59, 2024-07-29:
```
[Generic]
blockcmd = True
noipf = true
[Growatt]
#IP = 47.254.130.145
#IP = 8.209.71.240
IP = server.growatt.com
Port = 5279
```
worked with these settings for exactly 30 minutes, yesterday with the same settings but the fixed IP (8.209.71.240) it worked for about 20 hours before it was reset again.
it's probably just a coincidence how long it works.

typxxi, 2024-09-04:
**UPDATE:**

The JULY nightmare is over but there is still a new behaviour which I could observer among my fleet of 8 Growatt inverters.

SHUT DOWN GROTT SERVER seems to trigger the GROWATT CLOUD and resets the datalogger back to growatt servers

Has happened 4 times during maintenance when Home Assistant & HA Grott Addon were down. But it does not happen every time and only 1 or 2 were resetted out of 8. 

Could be a kind of "parenting" and systematic fix if a growatt inverter got lost for too long that when the next contact will be established the IP will be deleted and replaced by Growatt Cloud.

typxxi, 2024-09-05:
> ,

> yesterday with the same settings but the fixed IP (8.209.71.240) it worked for about 20 hours before it was reset again.
> it's probably just a coincidence how long it works.

As said above: the reset to the growatt cloud can be caused during a shutdown of grott or grott homeassistant cause I have 8 inverters and had to do a lot of maintenance in the past 10 days or so and therefore lost a 5 times 1 inverter or 5 out of 64 possibilities.

FOr me it feels that it has always been caused during shutdown / maintenance of the grott server.

johanmeijer, 2024-09-07:
@typxxi do you mean shutdown/maintance of the growatt server or really grott server? 

I am a  little bit puzzled here. I thought that using the server.growatt.com url and blocking the commands would prevent the resetting of the growatt IP addresses in the datalogger. 

Do you have a shinelink (with all inverters connected) or shinewiffi dataloggers per inverter?

pameruoso, 2024-11-21:
Here I am with my case. I have grott running in a vm
[image]
I changed the IP in the shinelan box
[image]
Still the box talks with growatt servers ignoring my setup
[image]
I think that probably this is a new update to the shinelan box (7.0.0.9).

Even if try to change the DNS server from 8.8.8.8 to my prviate dns server (to manually resolve the server.growatt.com domain to my private grott server), as soon as I click save it is changed back to 8.8.8.8

Someone with the same?

pameruoso, 2024-11-21:
Ok so it looks like that after creating a firewall rule to block communication from my shinelan to growatt servers after some minutes the shinelan gives up and starts sending data to my private grott server as shown up.

[image]