johanmeijer/grott #558: Growatt reconfiguring the Shine WIFI-X so it doesn't send data to grott anymore

molekuul, 2024-07-16:
Hi,

It looks like this morning,  Growatt is reconfiguring the Shine wifi-X adapter to default values so the IP adres where GROTT lives is removed and the Shine doesn't send data tot GROTT anymore 

Is someone else also seeing this behaviour ?

kind regards,
Joris

blauwblaatje, 2024-07-16:
Ahh, that makes sense.  I'm also suddenly missing packets going to GROTT...

molekuul, 2024-07-16:
I restarted grott in verbose mode at  10:44 UTC
and didn't recieve any data from my inverter.
At 10:46 UTC,  I reconfigured my Shine WIFI-X to send the data to my GROTT

Then at 10:50:28 UTC, I get this log:

2024-07-16T10:50:28.157640657Z 	 - ('192.168.208.1', 34810) has disconnected

And after this no entries anymore in my GROTT however my data does get updated in the shine app and at the growatt website

Data Logger：XGD******. SignalGood(-68) Device Type: ShineWiFi-X
Firmware Version: 3.1.0.5
IP & Port /**.**.***.185:15567 
Update Time：2024-07-16 12:53:30

blauwblaatje, 2024-07-16:
I think this is the same issue:   https://github.com/johanmeijer/grott/issues/557

molekuul, 2024-07-16:
Yes you are right. I will close this issue

closed 2024-07-16 by molekuul