johanmeijer/grott #563: Data-Logger resets back to growatt servers

mvdh02, 2024-07-16:
This item is closed. But there is no option mentioned that worked. I changed grott.ini and have restarted this command in my raspberry. But no data.

What are the steps to get this working again.
Thanks all.

PSMGoossens, 2024-07-16:
@mvdh02 Change the configuration for grott

`[Growatt]
#ip = server.growatt.com
ip = 8.209.71.240
port = 5279
`

Server IP from Growatt is changed this morning and by sending a request to the old server, a command to update the ip address is issued from Growatts old server

VisionedWheat66, 2024-07-16:
@mvdh02 I think what has happened is when Growatt pushed the new IP it has replaced the config on the datalogger that had been set to point at grott internally. So that part of the setup has to be repeated in addition to updating grott.ini as shown in #565.

mvdh02, 2024-07-17:
The change of the IP and port number as mentioned in grott.ini and a restart was the solution. After that i changed the ip nummer in de growatt X back to the IP number of my raspberry. And everthing works as before. Thanks

closed 2024-07-17 by mvdh02