johanmeijer/grott #567: no response from grott since 2024-07-16 8:00

cbshopper, 2024-07-17:
since yesterday grott is not working any more - maybe something in the growat api has changed... - 
`     

Grott settings:

:
	version:             	 2.7.8
	verbose:             	 True
	trace:               	 False
	config file:         	 grott.ini
	minrecl:             	 100
	decrypt:             	 True
	compat:              	 False
	invtype:             	 sph
	invtypemap:          	 {}
	include_all:         	 False
	blockcmd:            	 False
	noipf:               	 False
	time:                	 auto
	sendbuf:             	 True
	timezone:            	 local
	valueoffset:         	 6
	offset:              	 6
	inverterid:          	 automatic
	mode:                	 proxy
	grottip              	 0.0.0.0
	grottport            	 5279
_MQTT:
	nomqtt               	 False
	mqttip:              	 192.168.0.100
	mqttport:            	 1883
	mqtttopic:           	 energy/growatt
	mqttmtopic:          	 False
	mqttmtopicname:      	 energy/meter
	mqttinverterintopic: 	 False
	mqtttretain:         	 False
	mqtttauth:           	 True
	mqttuser:            	 hugo
	mqttpsw:             	 **secret**
_Growatt server:
	growattip:           	 8.209.71.240
	growattport:         	 5279
_PVOutput:
	pvoutput:            	 False
	pvdisv1:             	 False
	pvtemp:              	 False
	pvurl:               	 https://pvoutput.org/service/r2/addstatus.jsp
	pvapikey:            	 
	pvinverters:         	 1
	pvsystemid:          	 12345
_Influxdb:
	influx:             	 False
	influx2:            	 False
	database:           	 grottdb
	ip:                 	 localhost
	port:               	 8086
	user:               	 grott
	password:           	 **secret**
	organization:       	 grottorg
	bucket:             	 grottdb
	token:              	 **secret**
_Extension:
	extension:          	 False
	extname:            	 grottext
	extvar:             	 {'none': 'none'}
Grott proxy mode started
Hostname : hc.fritz.box
IP :  192.168.0.100 , port :  5279 `

nothing more, no packages are received. 
The IP-Address of server.growatt.com has changed - i had to change the settings in the web-Interface or my inverter. Growatt Server Web Interface still working.

pseuss, 2024-07-17:
Same for me. Web Interface and Shine App still works and has actual data.

wbrocker, 2024-07-17:
Have a look at #563

cbshopper, 2024-07-17:
have a look on the configuration output - ip address is already updated. But still no communication ....

XethaX, 2024-07-17:
Restart datalogger, after that you get what i get.
Data less then minimum record length, data not processed

fbojor, 2024-07-17:
I also have this issue. The first time when I reconfigure data logger it sends the data then stops. It's like growatt updates the server on data logger or smth

wbrocker, 2024-07-17:
Have you tried to add blockcmd = True to the grott.ini

That solved it for me after updating my local IP again on the Growatt interface.

pseuss, 2024-07-17:
After updating the .ini and adding my IP on the logger, everything works as before. Thanks!

fbojor, 2024-07-17:
All good for me also, updated the IP and reconfigured the logger. TY!

lspaula, 2024-07-17:
Same thing here. Is there a way of reconfiguring the server IP on the dongle from a remote location, without having to connect to the dongle locally via a Wi-Fi hotspot?

cbshopper, 2024-07-17:
Yes - now it works again :-) 
Thank you all!

Gemies, 2024-07-18:
> Great,
> 
> So here is everything I have done (after changing the IP in the server.growatt.com site as above mentioned):
> 
> Uncommented the following line to the grott.ini file:
> 
> `blockcmd = True
> `
> 
> Change the growatt server in the grott.ini file:
> 
> ```
> [Growatt]
> #ip = server.growatt.com
> ip = 8.209.71.240
> port = 5279
> ```
> 
> Restart the grott service:
> 
> `sudo service grott restart
> `
> 
> Thank you very much for the fast response. Both inverters are sending data to my InFlux database again.
> 
> Regards,
> 
> Luciano

This worked for me, thanks a lot!

Although i did  fullrestart of home assistant and did the changes and the restart in the grott addon configuration.

alby6969, 2024-07-20:
Hi, I am a fossil, and this logging into dongles and ip is a bit beyond my levels of understanding. Will this be rectified in the next ShinePhone update?

I am happy to use the web to view my data for now. So long as the ShinePhone update they do next will fix it?

johanmeijer, 2024-07-22:
What do you expect to be fixed by a shinephone update? Growatt will not fixing the grott challenges I think.

jirris, 2024-07-24:
I can confirm that changing IP to server name server.growatt.com in grott.ini fixed the issue for me. No need for blocking the commands.

johanmeijer, 2024-07-24:
Probably it will work until growatt decide to change IP address again ;) 

Using the server.growatt.com url instead of IP address might prevent future outages.

jirris, 2024-07-25:
Well that's what I meant, changed to domain name so it started working.

bockelaar, 2024-08-05:
server.growatt.com looks like a round-robin for 8.209.71.240 and 47.254.130.145 and is an alias for nlb-nbtf7vlspietfgd3ro.eu-central-1.nlb.aliyuncs.com which looks like a loadbalancer. Stuff breaks this way :)