johanmeijer/grott #571: DATA LOGGER MESS and the unsufficient, limited blockcmd solution - OR - what is there a better fix / workaround that is fully working ?

typxxi, 2024-07-28:
In the past 10 days I had been quite busy replacing our old WLAN with new LAN cables across the house by adding mesh repeater each connected via cable to the router. Those LAN and WLAN issues have been fixed.
Then I had to reconfigure our AC side to a NEW FIRST AC DISTRIBUTION BOX (the old one has become the 2nd) which required to shut down all of our 8 growatt inverters (5 different models min 4600, mic 600 / mic 2000/ mic 3000 + SPH 4600) and replace most of their AC cables too . 

Once that was done GROTT was dead and still seems to be dead even though I have tried a lot first on my own without checking github first. So I followed my receipe book how I had set up Grott and the Growatt HA integration. 
I found out that my dataloggers seemed to have lost the IP adress of my HA server and then it seemed to work fine ... cause I did not realise that mqtt was only working for a minute or so and then it was broken again. 
That was the point when I had to rethink and went back to github where I found topics about the issue.

Right now I have shut down all inverters except the SPH / Hybrid one to fix the issue cause with 8 online there is too much traffic going on. 

So far I have read many assumptions and suggestions and none of them still work and it had gotten even worse.

Current state:
HA OS running on a pi 4 with 8 growatt inverters
Grott beta 2.8 Addon is running but with other issues described below

I have enabled the option "gblockcmd" which should solve the whole issue - till I read further that it might no longer work.

Anyhow I found that someone had posted to repower the datalogger stick where I was unsure what that would mean like "reboot or restart of the stick from the UI" or a real reboot of the inverter by turning of the circuit breaker, but it seems he meant to repower the stick by pulling it out and pushing it in afterwards. 

Finally it seemed to work fine until I found other issues that others might be aware off but I was not.

This is the new error when I want to change a SPH setting like discharging times and what not via growatt web UI.
[image]

It turns out that "block cmd" is also rejecting changes on the server via web interface and you will get the same error over and over again. Later I found out that if I deactivate the blockcmd button I can change the settings.

Now I ended up in a sequence of changes I need to make and hate to make.
#1 inside HA: deactivate the blockcmd option
#2 growatt web UI: make the SPH inverter changes
#3 then also open the datalogger and change the IP to the local grott server
#4 back to HA turn the clockcmd option "ON" again

And for every change such procedure has to be done to keep the system running fully.

johanmeijer, 2024-07-28:
Why not setting growattip = "server.growatt.com" ? Then growatt does not want to reset  the IP address anymore I think.

johanmeijer, 2024-07-28:
- till I read further that it might no longer work 

This one I does not understand. Blockcmd (gblockcmd) should work. Unless the HA addon is different then the original Grott build.

typxxi, 2024-07-28:
> Why not setting growattip = "server.growatt.com" ? Then growatt does not want to reset the IP address anymore I think.

Sorry, I had used the wrong screenshot so the error is a different one as you can see here.
[image]

Do you mean to set the "hidden" setting growattip = "server.growatt.com"?
[image]

I will try that now.

typxxi, 2024-07-28:
> * till I read further that it might no longer work
> 
> This one I does not understand. Blockcmd (gblockcmd) should work. Unless the HA addon is different then the original Grott build.

Someone had mentioned in a topic that gblockcmd or blockcmd in the ini did no longer work which has caused a confusion to me and made me hesitant to try it out first.

Then I started using blockcmd to find out that I was no longer able to change inverter settings.

Right now I am following your suggestion and try to use the growattip thing . I am still checking it and hope it will work this way that others can follow these instructions.

typxxi, 2024-07-28:
I have added this 
[image]

Grott is working, mqtt is receiving the messages and Grott HA integration is showing the changes every minute.
But I **can not change the settings** of the inverter via web UI.

I get the same error over and over again : Data logger timeout without reply
[image]

SIDE NOTE:

I have now checked my data logger settings again while GROTT is working fine but in the datalogger settings the HA Grott IP has gone again, but still working fine.
[image]

typxxi, 2024-07-28:
and if I also deactivate gblockcmd then growatt will remove the IP adresse of my HA Grott Server again.

typxxi, 2024-07-28:
I have tried a lot since yesterday but:

It only works with the gblockcmd turned on.
And then I can not change the SPH inverter settings if I do not turn the gblockcmd off before I change the inverter settings.

It works for now - but in the long run I will need to change the SPH settings more often and therefore will need a kind of automatism based on HomeAssistant.

Would be nice to have a solution that will not have such side effects as this one has right now.

typxxi, 2024-07-28:
thanks for your quick support and I will send you a paypal donation from a different paypal account name but referring in the comment to typxxi Wolf-Gang

typxxi, 2024-07-28:
thanks for your quick support and I have send you a paypal donation right from a different paypal account name called "Hermann"

closed 2024-07-28 by typxxi

PedroKTFC, 2024-07-28:
Have you tried grottserver? That avoids using the Growatt servers completely.

typxxi, 2024-07-28:
> Have you tried grottserver? That avoids using the Growatt servers completely.

not yet, cause we have family members that love the app and learned to understand it over the past few years which also are at an age where new things are not so welcomed anymore (above 80 years) even though they have fully moved ahead from diesel car to electric 5 years ago and from diesel central house heating to split AC heating and central heat pump too.

But what is the difference between grott 2.8.3 or so and grott server and the upcoming grott 3.0 ?
Thanks for sharing

typxxi, 2024-07-28:
Here final update which solved my issues with the growatt ip resets and also allows me to edit the inverter settings like before:

**For me the growattip setting alone is doing the trick and no blockcmd needed.** 
Therefore I can still edit the inverter settings like before. Full description for HA GROTT addon useres can be found [here](https://github.com/muppet3000/homeassistant-grott/issues/91#issuecomment-2254581671) 

The easiest way ist from the grott addon configuration side without touching the grott.ini file.