johanmeijer/grott #572: Can't establish connection with remote server

gsnhf, 2024-07-29:
Hello,
when I start my docker-compose installation of grottserver  on a raspi 4. Configuration only works with mounting the grott.ini file. But that doesn't matter now.

When I configure 

```
[Growatt] 
ip = server.growatt.com
``` 

I get the following errors. Endless.

```
[Errno -3] Try again
         - Can't establish connection with remote server.
         - Closing connection with client side ('192.168.178.83', 11585)
[Errno -3] Try again
         - Can't establish connection with remote server.
         - Closing connection with client side ('192.168.178.83', 41259)
``` 

What could it be? Could someone help me?

johanmeijer, 2024-07-29:
The docker environment is ggrowattip (that is what you have to use  in the docker compose).

gsnhf, 2024-07-29:
The ip works for me. Thank you. 

I tried it with the docker environment variables. But mounting the ini file works. The environment variables not.

closed 2024-07-29 by gsnhf