johanmeijer/grott #579: GroWatt pulled a new trick (i think)

horstefan, 2024-08-07:
So after postponing to deal with this new crap from growatt i finally took the time today.
i updated GROTT, changed back the server URL and enabled command blocking...

it now looks like my invertor refuses to report anything. it looks like it awaits a confirmation from growatt before it will start
My shinelink firmware version is 3.6.0.0

i also get no information published to mqtt but this might be because there is nothing to report...
maybe a feature request: can we have online/offline reporting into mqtt?

tabascoz, 2024-08-07:
Hi, 
I just reported in https://github.com/johanmeijer/grott/discussions/578 that despite my initial assumtion, in firmware 3.0.0.2 this behavior doesn't happen.

I see some options for you: Try to fully enable connection to outside (growatt) to have this initial check to pass, then block everything again. See if that works. 

If it works, good. Perhaps you could do something similar than I did. If you happen to have a wifi router with OpenWRT installed, you could sniff the traffic, post it back and may Johan or someone could code Grott server to mimic the command. That would be nice! 

If not possible, there's another option which i don't know how difficult it is ( or possible ) would it be t downgrade the firmware to 3.0.0.2. What is your stick model? if same as mine, i could try to backup my firmware...

horstefan, 2024-08-07:
Thanks @tabascoz unblocking and blocking appears to have worked (for now). 
im curious to see what happens after the sun goes down and comes back up in the morning...

i ordered some stuff to create my own connector with ESPHome to fix this on long term.

tabascoz, 2024-08-07:
Great, esphome is my plan 'b' but grott is working really awesome so don't plan to change it (lol) 

Cheers,

NosNL, 2024-08-08:
after 2 hours testing it works again (fw 3.6.0.2)

i get ESP home stuff .. it is waiting for the next trick