johanmeijer/grott #580: Grott - grottproxy - Invalid data record received, processing stopped for this record

Snowless3, 2024-08-12:
Hello,

can someone help me with the following issue?
`IP :  172.18.0.10 , port :  5279 
	 - ('192.168.1.46', 53716) has connected
	 - Growatt packet received:
		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.18.0.10', 43304), raddr=('47.254.130.145', 5279)>
	 - Grott - grottproxy - Invalid data record received, processing stopped for this record`

my config:
`	version:             	 2.8.3
	verbose:             	 True
	trace:               	 False
	config file:         	 grott.ini
	minrecl:             	 100
	decrypt:             	 True
	compat:              	 False
	invtype:             	 default
	invtypemap:          	 {}
	include_all:         	 False
	blockcmd:            	 True
	noipf:               	 False
	time:                	 auto
	sendbuf:             	 True
	timezone:            	 local
	valueoffset:         	 6
	offset:              	 6
	inverterid:          	 automatic
	mode:                	 proxy
	grottip              	 default
	grottport            	 5279
_Growatt server:
	growattip:           	 server.growatt.com
	growattport:         	 5279`

Thx

Snowless3, 2024-08-15:
No answer, is this project dead?

johanmeijer, 2024-08-16:
NEO is not supported (yet. It is a totally different protocol). See also #552.

I am not beeing paid for this hobby project and have sometimes other priorities.