johanmeijer/grott #587: MOD 7000TL3-XH - no Batterie Values and Strange Values

schlidda83, 2024-08-30:
Hi,
i´m using the latest stable version 2.8.3 running in HA with Grott custom components.

the best values do i get with 

[image]
[image]
[image]

But this is the Datalogger Serial! and the log is saying that the SN wasnt found and set to default.

If i use the Inverter serial, i get rubbish
[image]

What i´m doing wrong?

Can´t get battery stats :(

I get the other values using differnt Inverter types. The "MIN" value gave me battery stats - but the other stats wasnt correct.

I really appreaciate your support. This is such a great tool! thanks for all your efforts.

Stephan

johanmeijer, 2024-09-02:
Do you use the HA version of the original Grott docker image?

schlidda83, 2024-09-03:
Hi @johanmeijer, i´m using Grott from HACS.

johanmeijer, 2024-09-03:
Ok I really have to look how HA (HACS) and the Grott add-ons are working. I see there are (at least) two add-ons (from muppet3000 and egguy). 

I am not sure if they use they original Grott code or have there own implementation (fork) of Grott. 

I have to install HA I think but i do not like they idea of monopolising  an RPI for this. I have HA installed in a docker container but that comes with a lot of limitations. 

At this moment the layout definitions of the MOD inverter is in the examples directory (and standard included in the grott beta docker container). For now I will put it in the grott beta base code folder so it will be part of the basic installation. 
If you want to use the layout you have to specify invtype=MOD.

schlidda83, 2024-09-03:
SORRY  yes i´m using this one: https://github.com/egguy/addon-grott

what do i need to do now?

typxxi, 2024-09-04:
Grott under HA is basically egguys work cause the part in the screenshot is the translation . Grott by muppetguy is the way of how the sensors are established and he is using the decoded results grott HA is providing.

If there would be an issue than on egguys part who is doing the translation cause you can already read that decoded values in the protocol / log you showed in the screenshots. But I doubt that this is the cause. Muppet 3000 integration is adding or providing just the right parameters for grott which means serialnumber and model type.

I have 8 inverters or even now 9 (the last is still in the box) and I made the whole journey from grott native on a pi with HA and then the muppet 3000 integration over to egguys grott addon (not using his sensor creation).  And it worked till growatt kicked my HA ip out of the datalogger settings.

AGAIN: We are in the mess of the "grott home assistant naming ideas" where everyone gets confused.

GROTT is GROTT
but under HA things get complicated by the naming confusion which should be resolved cause it is hard  to understand for newbies who does what and what is based on what. 
I also miss the clear reference to grott which code of grott is used. What I found was 2.8 stable but not 2.8.3
But that can also be missing number cause 2.8 had not supported some of the inverters I am using . 

Time for a GROTT naming sheme to reduce the mess at least. Some started moving to the solax integration which has started to support growatt too.

typxxi, 2024-09-06:
> Ok I really have to look how HA (HACS) and the Grott add-ons are working. I see there are (at least) two add-ons (from muppet3000 and egguy).
> 
> I am not sure if they use they original Grott code or have there own implementation (fork) of Grott.

Afaik egguy is taking your code and adding the HA integration parts. 
Why ?
Back then in spring I had asked for implementation of certain inverter model like now and he had said that he would be waiting for your release which back then was 2.8.3 becoming the new stable. 
So he did not change anything and described how I could try something out meanwhile. Then it was may or so and he ported your grott into "HA grott Addon".

But maybe it would be better if  @schlidda83  tries to setup a genuine grott server on a pi or vm like I had done in the beginning cause then it would be far easier to adress the points / issues. This single MOD question is now running on all "instances" and I guess it might end up here i Grott again - or egguy will reply and reassure that his HA Grott addon is based on your 2.8.3 

I am quite happy that my instance is now running stable again - after I had found out that maintenance shutdowns of grott or homeassistant can awake the growatt cloud to clean at least 1 or even more inverter (if the shutdown might be longer) datalogger settings (IP adress only). So I have to check that twice if all inverters are devlivering (I have 13 strings connected to 8 inverters running here).

thanks

johanmeijer, 2024-09-07:
I think my Grott naming is very clear (I think). At this moment there is no difference between the Master and Beta. Both are 2.8.3 including MOD support by an addition MOD layout file (also available in the Beta Docker container). 

The docker master is still 2.7.8. Sorry for that but I have to reorganize my docker image generation pc / procedure before I can promote this one to 2.8.3.

typxxi, 2024-09-08:
> I think my Grott naming is very clear (I think). At this moment there is no difference between the Master and Beta. Both are 2.8.3 including MOD support by an addition MOD layout file (also available in the Beta Docker container).
> 
> The docker master is still 2.7.8. Sorry for that but I have to reorganize my docker image generation pc / procedure before I can promote this one to 2.8.3.
> 
> I am not using HA myself and are not able to create or test HA addons. I rely there on egguy and muppet3000. Maybe we can imbed the HA support in grott in the future.

My point is a more stringent name convention like Grott and then Grott HA Addon not vice versa which can be confusing for beginners that not always are familiar with the history and background.

Type into the github search "grott" and look what you get. 
I had - back then - not assumed that "HA grott addon" means GROTT like your grott 
while muppet3000 has developed the GROTT SENSOR INTEGRATION FOR HA which then can be used with both
GROTT and GROTT HA ADDON. 

So many are not aware of that therefore I am emphasizing a clear "naming" sheme across the grott solution family which it has become. I am using egguys add on as it is provided and muppets3000 integration and read along what you are doing in the genuine GROTT, which I had been using for  a month or so to make it working and then to migrate the whole fleet in 1 evening.

THANKS

PedroKTFC, 2024-09-10:
I do agree the naming convention is a little confusing. This is not a criticism, it's just that for those of us not looking at the different components regularly, the different namings don't come naturally and you have to some work to remember what is what.

egguy, 2024-09-10:
> I am not sure if they use they original Grott code or have there own implementation (fork) of Grott.

Hi, the add-on is just a docker container. I'm using your code unmodified as you can see here: https://github.com/egguy/addon-grott/blob/main/grott/Dockerfile#L18. At the time of writing I'm using your commit 09a99347b0ae79398071b26f198e2630aadd4eb3 as the base of the code: https://github.com/egguy/addon-grott/blob/main/grott/Dockerfile#L5

The addon run the built container with my [grott addon](https://pypi.org/project/grott-ha-plugin/) (which can be fully disabled). This is using the plugin infrastructure of grott (https://github.com/johanmeijer/grott/wiki/Extensions) to add the reformatting the JSON payload to [HA MQTT sensor](https://www.home-assistant.io/integrations/sensor.mqtt/) without changing the original code.

The addon provide a way to configure the environment variable through the HA interface: https://github.com/egguy/addon-grott/blob/main/grott/script.sh#L234

You can set nothing in the graphical HA configuration and access the `grott.ini` in the folder `/addon_configs/ef7b0eff_grott/` (including custom layout...) 

--

For the naming, I used the convention of HA, where they use the format "HA addon : software". I added the version at the time, when there was a beta and stable (2.7/2.8) and added the mention of **the included grott plugin** to make it compatible with HA automatically. 

I'm still searching for a better way to name it as there is often a confusion with difference between plugin (grott side)/integration (ha side)/addon (docker container run by HA).

schlidda83, 2024-09-10:
@egguy thanks for replying. But i´m just a simple HA user -  are you seeing the problem? is there anything i need to add in the ini?

egguy, 2024-09-10:
I'm not including currently the JSON layout, as I was thinking they were already implemented in python. I'm going to add it in the next build (probably during this week). 

@johanmeijer Do you have any guidance on which layout should be loaded from JSON and which should not?

From what I'm understanding, the support for MOD is only JSON at the moment?

johanmeijer, 2024-09-16:
At this moment only the MOD configuration is provided in a separated layout file (T06NNNNXMOD.json). 

Supported inverter types: 

**By default:** 
S/TL-S (TL-X normal type?)

**invtype = MIN**
TL-X/TL-XH/TL-XH US（MIN Type）

**invtype = XTL3**
TL3-X(MAX、MID、MAC Type)

**invtype = MAX**
MAX 1500V、MAX-X LV

**invtype = MOD** 
MOD TL3-XH 
(external layout file needed)

**invtype = SPH** 
Storage(MIX Type) (mapping might miss some values)
Storage(SPH Type)

**invtype = SPA** 
Storage(SPA Type)

johanmeijer, 2024-09-16:
Next to it the Growatt 1phase and 3 phase (and EATRON SDM630) energy meters are supported.

FlexxFR, 2024-12-08:
> this moment only the MOD configuration is provided in a separated layout file (T06NNNNXMOD.json).

I'm still struggling to get it to work for a  MOD 8000TL3-XH. 
Started with the HA- egguy version but I wasn't sure if the new MOD layouw was working or how to enable it. I tried many options but values were still rubbish.

I' switch to run grott directly in a container. Again, not sure if this includes the latest and greatest. It's working, receiving data, mentions something about layout T065104XMOD, so I believe it should be working, right?

```
         - Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T065104XMOD
	 - no matching record layout found, try generic
	 - no matching record layout found, standard processing performed
	 - Record layout used :  none
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
```

I do use the grott integration from muppet to "translate" the MQTT topics into HA entities. But it not really working when using 
`
```
# Specify inverter id (not necessary in version >2.1.0 if compat = false!)
#inverterid = ABC1234567
# Specify the type of the inverter (default/sph/spf/max)
**invtype = mod**
```
mod, MOD, tl3, I also read that it has to be ginvtype=mod ? What is correct here?

No type specified: (uses layout: T065104X / Record layout used :  T06NNNNX)
- PV Generation Today + Total seems to be okay
- Many other value seems to be off, not sure as sun isn't shining now
-  Diagnostic data partially correct,  last data update not working

invtype = mod
- No data update anymore, I see layout: T065104XMOD being used, but I do not see any values in grott log. Also no updates in HA anymore

invtype = tl3
- Saw some others with success using tl3, but it's not work either. Rubbish values.

Without anything specified:

```
	 - Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T065104X
	 - no matching record layout found, try generic
	 - Record layout used :  T06NNNNX
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:

	 - Possible Inverter serial Q............C
	 - Inverter serial not recognised - using inverter type default
	 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
		 - record layout :  T06NNNNX
	 - Grott data record date/time processing started
	 - no or no valid time/date found, grott server time will be used (buffer records not sent!)
```

Why is MOD not working for me?

JusNoGood, 2025-06-11:
Hi @johanmeijer 

I have a similar issue with my 2 x MIN 6000TL-X  inverters I hope you can help me with. I've tried invtype of min, mod, spf sph tl3 and I don't think any are mapping correctly. I either get zeros in most values or very all values. 

When this log file was created I had generated 106.6kwh today and my batteries were 98% full. 

[ledidobe-grottserver-2-8-3 (1).csv](https://github.com/user-attachments/files/20698639/ledidobe-grottserver-2-8-3.1.csv)

I'm running 2.8.3 in a container.

Will be very grateful for some help. 

Jus.