johanmeijer/grott #590: Incomplete data points for a day

Ebrink, 2024-09-05:
Hi,

I'm using grott (ledidobe/grott:beta) to read data from two MIN 3600 TL-X inverters and then put into influxdb. for some time now, I've had incomplete data: most of the days, I won't get any new data points starting at around 15:00 CEST (see attached image). Next data points start the following morning.

[influx]
# Influxdb parameters definitions
influx = True
influx2 = True
ip = 192.168.1.3
port = 8086
token  = redacted
org  = redacted
bucket = redacted 
timezone = europe/amsterdam

Anyone else experiencing this?
[image]

[image]
[image]

Ebrink, 2024-09-09:
Hi @johanmeijer,

Please find the logs attached. I have changed the configuration to reflect the time=server parameter, and that seems to work. 

In the logs below, inverter RAL2CJT03G does not have any information in influx after 14:12.
[detail.log](https://github.com/user-attachments/files/16926843/detail.log)

johanmeijer, 2024-09-16:
Hmm strange time values, interesting 

The RAL2CJT07T has the last "normal" time @14:15 after that he jumps to 07.10
The RAL2CJT03G never has a "normal" date/time  field in this log. 

I am not sure what cause the datalogger to insert the wrong time. It can be a firmware issue or a timedrift. Seeing the offset of 7 hours at once I think it is firmware issue. 

What type of dataloggers do you use? The shinewifi / shinelan or shinelink with RF dataloggers?

Ebrink, 2024-09-17:
Hi @johanmeijer, these are shinewifi-x modules.

johanmeijer, 2024-09-18:
Ok. There seems to be more dataloggers with the same problem. 

I think the only way to bypass this is to use the time=server parameter or stop/start grott at regular basis. 

If you use grottserver it is also possible to set the date/time via the API. 

Please also see #595.