johanmeijer/grott #592: Support for ShineWifi X2

jafar-atili, 2024-09-12:
I have just installed a Growatt Hybrid inverter.

I received the ShineWifi X2 and the server configuration doesn't seems to allow changing the port from 7006:

[image]

I tried changing the Grott Addons port to 7006 but it still comes up either tje default port (5279)

And I am not sure if the integration even expecting an MQTT packet as the default domain seems to be MQTT.

My inverter is MID 11-30KTL3-XH

johanmeijer, 2024-09-16:
This one does not ring a bell to me. Which configuration tool is this and what do you try to configure? 
Do you want to use Home assistent?

jafar-atili, 2024-09-16:
This is the shinephone app datalogger configuration

I am trying to configure grott in HA.

l-k-r, 2025-05-29:
Is there a way to configure grott to output the not decoded packets to support you in this activity?

T1gg1, 2025-07-02:
I tried to put in my mqtt server but it seems that the stick does not connect to it.