johanmeijer/grott #595: Time register has wrong time (but not always!)

PedroKTFC, 2024-09-16:
This isn't (I think!) an issue with Grott/grottserver but I'm hoping someone can enlighten me.

When I read register 31 of my data logger (which should contain the data logger's time), I get odd results. This morning I got the correct time (within seconds) but, I've just read it at 19:56, I get 05:47, wrong and illogically wrong!! The date is always correct (allowing for before/after midnight). Anyone any ideas what's going wrong? Also, what is the format of the command to set the time?

johanmeijer, 2024-09-16:
You can use the command=datetime in the grottserver api. That should update the datetime of the datalogger. 

post <your ip>:5782/datalogger?command=datetime&datalogger=<your datalogger>

Other possibility is to stop/start grott or grottserver. The time will be set during initialisation phase.  

It seems that more dataloggers have this date-time problem(eg #595).  I think this should be a firmware problem.

PedroKTFC, 2024-09-17:
Thanks Johann. I'm aware of this. I've decided to try and investigate a bit. So last night I restarted Grott and set off a simple NodeRed flow to read the time every hour. It didn't take long for the time to be wrong! If I work anything out I'll report back here.

PedroKTFC, 2024-09-19:
I've now got a NodeRed flow that sets the data logger time every 15 minutes.

anoppe, 2024-09-19:
@PedroKTFC out of curiosity: I'm running grott flawless using server date time, why don't you change your config like Johan suggested, instead of (in my opinion) try to fix it by applying a bandage?

PedroKTFC, 2024-09-19:
Grott has worked flawlessly for me too, for months. As I understand it, using server time means Grott uses the local HA server time when it records data. Meanwhile, the time in the data logger can drift as this thread suggests. If I set the inverter to, as I do, export power between 16:00 and 19:00 (top rate for me), then as the time drifts this will go wrong. I therefore have to apply the bandage don't I? Note I'm using my version of grottserver so the Shine app can never correct the time.

anoppe, 2024-09-19:
oh, I see. You are dependent on the inverter datetime to control it on a specific date time... too bad for you :(