johanmeijer/grott #596: Chint DTSU666 Smart meter data using MOD6000TL3-XH with battery and shinelan

martinj, 2024-09-17:
Grott version: 2.8.3
Currently using MIN layout to get data with battery stats

Would also like to get smart meter data. 

Are there any guides on how to process and create new layout from debug data?

Here is som samples when "invtype" is set to default that i think might contains smart meter data

```
- Growatt original Data:
	\x00\xdc\x00\x06\x00\xed\x50\x29\x09\x33\x2c\x36\x22\x38\x43\x77\x22\x5a\x77
	\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
	\x2e\x2d\x24\x40\x37\x0b\x45\x5f\x27\x54\x74\x74\x47\x72\x6f\x77\x61\x74\x74
	\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x2b\x37\x39\x76\x36\x2c\x40\x51
	\x46\x36\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
	\x77\x61\x74\x61\x5f\x72\x6b\x77\x60\x41\x74\x42\x72\x6e\x46\x61\x72\x74\x45
	\x41\x5d\x77\x69\x74\x6a\x06\x28\x2a\x43\x22\x38\x43\x77\x22\x5a\x77\x61\x74
	\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x7e
	\x61\x75\x44\x47\x78\x6f\x76\x51\x74\x7f\x47\x72\x6f\x7a\x61\x76\x45\x7f\x72
	\x7b\x77\x69\x33\x20\x14\x25\x5f\x47\x51\x44\x74\x52\x72\x68\x40\x4f\x40\x5a
	\x76\x5c\x5c\x77\x77\x74\x70\x11\x43\x41\x47\x61\x6f\x74\x46\x44\x6f\x6b\x61
	\x75\x44\x47\x6c\x6f\x72\x26\x39\x20\x6c\x4a\x6f\x68\x61\x67\x46\x77\x40\x5b
	\x5a\x51\x4d\x59\x76\x45\x4f\x46\x55\x4e\x41\x72\x48\x5f\x47\x3d\xd3
- Grott automatic protocol detection
- Grott data record length 245
- layout   :  T065029
- no matching record layout found, try generic
- Record layout used :  T065029
- Growatt data decrypted V2
- Grott Growatt data decrypted
- Growatt plain data:
	00dc000600ed50294e414341434c3730503500000000000000000000000000000000000000004
	15a4534434c3730503500000000000000000000000000000000000000004a434d314443373032
	42000000000000000000000000000000000000000015180004000135000500013100060002333
	20008001e415a4534434c37305035000000000000000000000000000000000000000000090001
	30000a000130000b0000000d0002313800140008475453573030303000150007372e342e312e3
	30016000456312e30001b000136001c000130001e0005474d542b38001f0013323032342d3039
	2d31372031343a35353a30305ca7

- Growatt original Data:
	\x00\xdd\x00\x06\x01\x0e\x50\x20\x09\x33\x2c\x36\x22\x38\x43\x77\x22\x5a\x77
	\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
	\x25\x34\x2c\x45\x30\x04\x45\x5f\x45\x23\x74\x74\x47\x72\x6f\x77\x61\x74\x74
	\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\xa9
	\x74\x74\x47\xfc\x6f\x77\x69\xab\x74\x47\x7a\xb5\x77\x61\x7c\x95\x47\x72\x6f
	\x61\x61\x74\x74\x55\x72\x6f\x77\x79\x8b\x8b\xab\x30\x90\x88\x96\xe7\x8b\xb8
	\x98\xa4\x77\x61\x67\xa8\x47\x72\x61\x7c\x61\x74\x61\x2b\x8d\x90\x8a\xb1\x8b
	\x8b\xb3\xb1\x90\x88\x9d\x80\x8b\xb8\x8e\x48\x88\x9e\x8a\x75\xb8\x8d\x93\x58
	\x9e\x8b\xba\xd9\x72\x6f\x40\x32\x8b\x8b\xa8\xf4\x90\x88\x9d\xdb\x74\x47\x73
	\x9c\x77\x61\x7b\x2d\x47\x72\x60\x2d\x61\x74\x7b\x18\x72\x6f\x61\xb6\x74\x74
	\x44\xe0\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
	\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
	\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
	\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
	\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x24\x39
- Grott automatic protocol detection
- Grott data record length 278
- layout   :  T065020
- no matching record layout found, try generic
- Record layout used :  T065020
- Growatt data decrypted V2
- Grott Growatt data decrypted
- Growatt plain data:
	00dd0006010e50204e414341434c3730503500000000000000000000000000000000000000004
	a434d31444337303242000000000000000000000000000000000000000000000000000000c800
	00008e000008df000008da000008e1000000160000001200000018ffffec42fffff793ffffeac
	b000013dc00000e0b0000156cfffffdd0fffff4c3fffffcf4fffffc27fffffe01fffffc2fffff
	ce9e00003753ffffef86fffffcaf000001f300000f5900000f5a00000f5f000016d7000003920
	00000000000000000000000000000000000000000000000000000000000000000000000000000
	00000000000000000000000000000000000000000000000000000000000000000000000000000
	00000000000004b4e

- Growatt original Data:
	\x00\xde\x00\x06\x00\x99\x50\x37\x09\x33\x2c\x36\x22\x38\x43\x77\x22\x5a\x77
	\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
	\x25\x34\x2c\x45\x30\x04\x45\x5f\x45\x23\x74\x74\x47\x72\x6f\x77\x61\x74\x74
	\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x76\x72
	\xfc\x67\xe8\x72\x6f\x77\x61\x44\x37\x1f\x3f\x2c\x47\x3b\x26\x45\x73\x4b\x3b
	\x47\x51\x30\x47\x47\x72\x35\x32\x02\x15\x53\x56\x24\x2b\x16\x00\x74\x61\x56
	\xec\x6f\x62\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x64\x47\x72\x6f\x77\x61\xe2
	\x74\x46\x73\x69\x77\x6b\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x52\x6e\x77
	\x05\x6c\x7d\x56\x7c\x59\x5b\x4d\x70
- Grott automatic protocol detection
- Grott data record length 161
- layout   :  T065037
- no matching record layout found, try generic
- Record layout used :  T065037
- Growatt data decrypted V2
- Grott Growatt data decrypted
- Growatt plain data:
	00de0006009950374e414341434c3730503500000000000000000000000000000000000000004
	a434d314443373032420000000000000000000000000000000000000000000000000000011388
	13af000000003043584d43305a52313439543030443300005a4563612711564461610015119e0
	0150000000000000000001000000000009600010106000a000000000000000000002001006418
	09110e362c2c04

	- Growatt original Data:
	\x00\xdf\x00\x06\x00\xed\x50\x37\x09\x33\x2c\x36\x22\x38\x43\x77\x22\x5a\x77
	\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
	\x25\x34\x2c\x45\x30\x04\x45\x5f\x45\x23\x74\x74\x47\x72\x6f\x77\x61\x74\x74
	\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x75\x74
	\x6c\x61\x78\x24\x35\x3b\x51\x44\x44\x77\x42\x5d\x44\x51\x4d\x44\x77\x35\x35
	\x20\x20\x35\x35\x47\x70\x3e\x36\x23\x35\x74\x45\x72\x6f\x56\x61\x74\x29\x47
	\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
	\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
	\x61\x74\x74\x47\x72\x6f\x77\x74\x34\x61\x20\x24\x35\x3b\x51\x44\x44\x77\x42
	\x5d\x44\x51\x4d\x44\x77\x4a\x35\x20\x20\x35\x35\x47\x70\x3e\x36\x23\x35\x74
	\x45\x72\x6f\x55\x61\x74\x29\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
	\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
	\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x8e\x3d
- Grott automatic protocol detection
- Grott data record length 245
- layout   :  T065037
- no matching record layout found, try generic
- Record layout used :  T065037
- Growatt data decrypted V2
- Grott Growatt data decrypted
- Growatt plain data:
	00df000600ed50374e414341434c3730503500000000000000000000000000000000000000004
	a434d314443373032420000000000000000000000000000000000000000000000000000021518
	153f565a4c3030303030323330393030475a57414141000251414241000200002100005d00000
	00000000000000000000000000000000000000000000000000000000000000000000000000000
	000000000015401567565a4c3030303030323330393030385a574141410002514142410002000
	02200005d00000000000000000000000000000000000000000000000000000000000000000000
	000000000000000000000000ef49
```

johanmeijer, 2024-09-18:
The T065020 record is the smartmeter record. 

I already decoded this record but only in combination with a shinewifi datalogger (hint T060120 record). 

I included the layout for the smartmeter in combination with the shinelink (I think that is what you use). 

You can use this one by coping the T065020.json file from the examples folder into your grott main folder and stop/start grott. 
After this the smartmeter record will be decoded (and hopefully looks like what you expected)

martinj, 2024-09-19:
Thank, that worked.

closed 2024-09-19 by martinj