johanmeijer/grott #600: Shine Wifi-S White - Have to reboot docker every morning

Supered1967, 2024-09-23:
Good day,

Since a few weeks (after having Grott running for months) I have to reboot Docker every morning to get Grott working (reboot of Grott Container only doesn't solve the issue).

What I see every morning that ever 5 seconds Grott is processing Growatt data but doesn't decode it. Seems like an endless loop.

Below are some logs of the first 5 minutes when Wifi-S dongle is connected to Grott.

I am using the latest Beta. Strange thing is I have to reboot the Docker ?
I also deployed a new docker (ProxMox), installed Grott but with same issue. After a disconnect of the Wifi-S dongle (sun under), and a connect (Sun up) the issue is there.

Some logs:

2024-09-23T05:30:23.518227000Z 	 - ('192.168.1.30', 7747) has connected
2024-09-23T05:30:23.519340000Z 
2024-09-23T05:30:23.519402000Z 	 - Growatt packet received:
2024-09-23T05:30:23.519445000Z 		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('192.168.1.247', 36770), raddr=('8.209.71.240', 5279)>
2024-09-23T05:30:23.519481000Z 	 - Data less then minimum record length, data not processed
2024-09-23T05:30:23.628261000Z 
2024-09-23T05:30:23.628369000Z 	 - Growatt packet received:
2024-09-23T05:30:23.628942000Z 		  <socket.socket fd=5, family=2, type=1, proto=0, laddr=('192.168.1.247', 5279), raddr=('192.168.1.30', 7747)>
2024-09-23T05:30:23.629041000Z 	 - Data less then minimum record length, data not processed
2024-09-23T05:30:23.641934000Z 
2024-09-23T05:30:23.642005000Z 	 - Growatt packet received:
2024-09-23T05:30:23.642049000Z 		  <socket.socket fd=5, family=2, type=1, proto=0, laddr=('192.168.1.247', 5279), raddr=('192.168.1.30', 7747)>
2024-09-23T05:30:23.642088000Z 	 - Data less then minimum record length, data not processed
2024-09-23T05:30:23.690655000Z 
2024-09-23T05:30:23.690750000Z 	 - Growatt packet received:
2024-09-23T05:30:23.690792000Z 		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('192.168.1.247', 36770), raddr=('8.209.71.240', 5279)>
2024-09-23T05:30:23.690828000Z 	 - Data less then minimum record length, data not processed
2024-09-23T05:30:23.692786000Z 
2024-09-23T05:30:23.692856000Z 	 - Growatt packet received:
2024-09-23T05:30:23.693433000Z 		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('192.168.1.247', 36770), raddr=('8.209.71.240', 5279)>
2024-09-23T05:30:23.693471000Z 	 - Grott - grottproxy - Invalid data record received, processing stopped for this record
2024-09-23T05:30:23.714742000Z 
2024-09-23T05:30:23.714844000Z 	 - Growatt packet received:
2024-09-23T05:30:23.714913000Z 		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('192.168.1.247', 36770), raddr=('8.209.71.240', 5279)>
2024-09-23T05:30:23.714950000Z 	 - Data less then minimum record length, data not processed
2024-09-23T05:30:23.730582000Z 
2024-09-23T05:30:23.730647000Z 	 - Growatt packet received:
2024-09-23T05:30:23.730699000Z 		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('192.168.1.247', 36770), raddr=('8.209.71.240', 5279)>
2024-09-23T05:30:23.730734000Z 	 - Grott - grottproxy - Invalid data record received, processing stopped for this record
2024-09-23T05:30:23.732332000Z 
2024-09-23T05:30:23.732455000Z 	 - Growatt packet received:
2024-09-23T05:30:23.732511000Z 		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('192.168.1.247', 36770), raddr=('8.209.71.240', 5279)>
2024-09-23T05:30:23.732558000Z 	 - Grott - grottproxy - Invalid data record received, processing stopped for this record
2024-09-23T05:30:59.208350000Z 	 - ('192.168.1.30', 28939) has connected
2024-09-23T05:30:59.208918000Z 
2024-09-23T05:30:59.208981000Z 	 - Growatt packet received:
2024-09-23T05:30:59.209025000Z 		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('192.168.1.247', 55356), raddr=('8.209.71.240', 5279)>
2024-09-23T05:30:59.209060000Z 	 - Data less then minimum record length, data not processed
2024-09-23T05:30:59.245839000Z 
2024-09-23T05:30:59.245928000Z 	 - Growatt packet received:
2024-09-23T05:30:59.246022000Z 		  <socket.socket fd=7, family=2, type=1, proto=0, laddr=('192.168.1.247', 5279), raddr=('192.168.1.30', 28939)>
2024-09-23T05:30:59.246067000Z 	 - Data less then minimum record length, data not processed
2024-09-23T05:30:59.355531000Z 
2024-09-23T05:30:59.355632000Z 	 - Growatt packet received:
2024-09-23T05:30:59.355717000Z 		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('192.168.1.247', 55356), raddr=('8.209.71.240', 5279)>
2024-09-23T05:30:59.355783000Z 	 - Data less then minimum record length, data not processed
2024-09-23T05:30:59.357083000Z 
2024-09-23T05:30:59.357210000Z 	 - Growatt packet received:
2024-09-23T05:30:59.357920000Z 		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('192.168.1.247', 55356), raddr=('8.209.71.240', 5279)>
2024-09-23T05:30:59.358166000Z 	 - Grott - grottproxy - Invalid data record received, processing stopped for this record
2024-09-23T05:30:59.373342000Z 
2024-09-23T05:30:59.373429000Z 	 - Growatt packet received:
2024-09-23T05:30:59.373464000Z 		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('192.168.1.247', 55356), raddr=('8.209.71.240', 5279)>
2024-09-23T05:30:59.373504000Z 	 - Data less then minimum record length, data not processed
2024-09-23T05:30:59.394287000Z 
2024-09-23T05:30:59.394361000Z 	 - Growatt packet received:
2024-09-23T05:30:59.394403000Z 		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('192.168.1.247', 55356), raddr=('8.209.71.240', 5279)>
[... 356 lines cut ...]
2024-09-23T05:31:54.077994000Z 		 \x2c\x01\x01\x00\x01\x00\x00\x00\x64\x00\x00\x4e\x20\x00\x00\xa4\xd8\x0e\x42
2024-09-23T05:31:54.078029000Z 		 \x53\x2e\x31\x2e\x38\x00\x1e\x1e\x1e\x1e\x1e\x1e\x00\x01\x00\x05\x05\xdc\x00
2024-09-23T05:31:54.078066000Z 		 \x1e\x07\x3a\x0a\x50\x12\x91\x13\x99\x58\x58\x58\x58\x58\x58\x58\x58\x58\x58
2024-09-23T05:31:54.078098000Z 		 \x00\x00\x90\x41\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x07\x35\x09\xdd\x12
2024-09-23T05:31:54.078133000Z 		 \x91\x13\x99\x07\x80\x0a\x28\x12\x69\x13\xc1\x00\x64\x02\x01\x00\x2d\x00\x59
2024-09-23T05:31:54.078163000Z 		 \x00\x00\x00\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x96\x00\x96\x00\x05\x00
2024-09-23T05:31:54.078207000Z 		 \x05\x00\x96\x00\x96\x00\x03\x00\x03\x20\x20\x20\x50\x56\x20\x49\x6e\x76\x65
2024-09-23T05:31:54.078244000Z 		 \x72\x74\x65\x72\x20\x20\x53\x42\x41\x41\x33\x34\x30\x30\x00\x00\x00\x00\x01
2024-09-23T05:31:54.078278000Z 		 \x2c\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
2024-09-23T05:31:54.078324000Z 		 \x00\x00\x00\x00\x00\x00\x00\x53\x00\x55\x00\x57\x00\x59
2024-09-23T05:31:54.078376000Z 	 - Grott automatic protocol detection
2024-09-23T05:31:54.078411000Z 	 - Grott data record length 223
2024-09-23T05:31:54.078457000Z 	 - layout   :  T020103
2024-09-23T05:31:54.078505000Z 	 - no matching record layout found, try generic
2024-09-23T05:31:54.078543000Z 	 - Record layout used :  T020103
2024-09-23T05:31:54.078587000Z 	 - Growatt data decrypted V2
2024-09-23T05:31:54.078642000Z 	 - Grott Growatt data decrypted
2024-09-23T05:31:54.078680000Z 	 - Growatt plain data:
2024-09-23T05:31:54.078730000Z 		 0001000200d90103063a5b425042447141592f392c2c1f2a372f392c7447726f7761767447724
2024-09-23T05:31:54.078770000Z 		 37660747547726f1361743a67726fd3b97a36145c5e5959746a596c71697f747547776aab616a
2024-09-23T05:31:54.078810000Z 		 737d783f65f067ed1f2a372f392c2c1f2a377761e43547736f7761747447726f70547da955e37
2024-09-23T05:31:54.078848000Z 		 cee66f47e6f600664a0741045736f5a612d7447726f7762747447726f7761e274d1726a776474
2024-09-23T05:31:54.078878000Z 		 e247e46f7461775467523f21413d1a31171d0304065467212d3620474077426f776174756b726
2024-09-23T05:31:54.078916000Z 		 e7761747447726f7761747447726f7761747447726f776127741272387738
2024-09-23T05:31:54.078956000Z 	 - Grott data ack record or data record not defined no processing done

johanmeijer, 2024-09-23:
Strange. It seems that the inverter/datalogger is trying to start the communication (by sending the T0120103 initialisation record after starting a ping conversation) but does not get a reaction from growatt (or grottserver) 

I also see an invalid record comming from the inverter /datalogger (strange). 

Do you use grottserver?  Did you changed the data update interval (register 4 of the datalogger?). 

Where are you located? If you are located in North America you might need to connect to another growatt server (server-us.growatt.com).

Maybe it also interesting if we can see the last part of the log when the data is beeing sent in the evening. Maybe there is a error in closing the communication?

Supered1967, 2024-09-25:
Johan,

Thanks for the reply.

I am using Grott (not Grottserver) and I am living in the Netherlands ;-).

I Will gather more logs this weekend but did some more testing about the reboots of docker.

If I reboot the docker at midnight (when the inverter is off) the issue is still there in the morning when the inverter starts.
I have to reboot docker when the inverter is on, it is working the whole day. 
I even can see the data on the Growatt web.

any particular logging or settings you need?

oh, and I have blokcmd =true since july 

regards
De

johanmeijer, 2024-09-25:
If the restart in the evening does not work I think there is nothing to see in the log.

I will see in my own logging if something has changed in the communication between growatt and the inverter.

Can you also try it without the blockcmd?

Is it possible to specify minrecl=1 (gminrecl = 1 for docker) and send the log from the startup fail. Maybe we get more information.

Supered1967, 2024-09-25:
Johan,

I thought I had gblockcmd=True but made a type mistake, so setting showed {} as blockcmd setting.
I don't know what Grott will do with this (defaulting to True?)

I changed the ENV setting as requested, and set the gblockcmd=False.
I will get the log tomorrow, and do the same again with gblockcmd=True (if application fails)

What I am wondering.

When system goes "wrong", every 5 seconds the seqence repeats.
Is this indeed a message that comes from Inverter/Growatt? Is the message waiting for an answer back to the inverter, so after an answer it will continue then with the following-up messages like power production etc?

I am just curious about how the inverter works with Grott and Growatt Cloud ;-)

=============================================================================
2024-09-23T05:32:02.723402000Z 	 - Growatt packet received:
2024-09-23T05:32:02.723442000Z 		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('192.168.1.247', 55356), raddr=('8.209.71.240', 5279)>
2024-09-23T05:32:02.723797000Z 	 - Growatt original Data:
2024-09-23T05:32:02.723860000Z 		 \x00\x01\x00\x02\x00\xd9\x01\x03\x41\x48\x34\x35\x31\x36\x30\x36\x33\x36\x58
2024-09-23T05:32:02.723913000Z 		 \x58\x58\x58\x58\x58\x58\x58\x58\x58\x00\x00\x00\x00\x00\x00\x02\x00\x00\x00
2024-09-23T05:32:02.723963000Z 		 \x2c\x01\x01\x00\x01\x00\x00\x00\x64\x00\x00\x4e\x20\x00\x00\xa4\xd8\x0e\x42
2024-09-23T05:32:02.724009000Z 		 \x53\x2e\x31\x2e\x38\x00\x1e\x1e\x1e\x1e\x1e\x1e\x00\x01\x00\x05\x05\xdc\x00
2024-09-23T05:32:02.724052000Z 		 \x1e\x07\x3a\x0a\x50\x12\x91\x13\x99\x58\x58\x58\x58\x58\x58\x58\x58\x58\x58
2024-09-23T05:32:02.724096000Z 		 \x00\x00\x90\x41\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x07\x35\x09\xdd\x12
2024-09-23T05:32:02.724140000Z 		 \x91\x13\x99\x07\x80\x0a\x28\x12\x69\x13\xc1\x00\x64\x02\x01\x00\x2d\x00\x59
2024-09-23T05:32:02.724173000Z 		 \x00\x00\x00\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x96\x00\x96\x00\x05\x00
2024-09-23T05:32:02.724211000Z 		 \x05\x00\x96\x00\x96\x00\x03\x00\x03\x20\x20\x20\x50\x56\x20\x49\x6e\x76\x65
2024-09-23T05:32:02.724248000Z 		 \x72\x74\x65\x72\x20\x20\x53\x42\x41\x41\x33\x34\x30\x30\x00\x00\x00\x00\x01
2024-09-23T05:32:02.724281000Z 		 \x2c\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
2024-09-23T05:32:02.724310000Z 		 \x00\x00\x00\x00\x00\x00\x00\x53\x00\x55\x00\x57\x00\x59
2024-09-23T05:32:02.724340000Z 	 - Grott automatic protocol detection
2024-09-23T05:32:02.724369000Z 	 - Grott data record length 223
2024-09-23T05:32:02.724399000Z 	 - layout   :  T020103
2024-09-23T05:32:02.724433000Z 	 - no matching record layout found, try generic
2024-09-23T05:32:02.724465000Z 	 - Record layout used :  T020103
2024-09-23T05:32:02.724508000Z 	 - Growatt data decrypted V2
2024-09-23T05:32:02.724543000Z 	 - Grott Growatt data decrypted
2024-09-23T05:32:02.724999000Z 	 - Growatt plain data:
2024-09-23T05:32:02.725035000Z 		 0001000200d90103063a5b425042447141592f392c2c1f2a372f392c7447726f7761767447724
2024-09-23T05:32:02.725071000Z 		 37660747547726f1361743a67726fd3b97a36145c5e5959746a596c71697f747547776aab616a
2024-09-23T05:32:02.725101000Z 		 737d783f65f067ed1f2a372f392c2c1f2a377761e43547736f7761747447726f70547da955e37
2024-09-23T05:32:02.725133000Z 		 cee66f47e6f600664a0741045736f5a612d7447726f7762747447726f7761e274d1726a776474
2024-09-23T05:32:02.725162000Z 		 e247e46f7461775467523f21413d1a31171d0304065467212d3620474077426f776174756b726
2024-09-23T05:32:02.725199000Z 		 e7761747447726f7761747447726f7761747447726f776127741272387738
2024-09-23T05:32:02.725229000Z 	 - Grott data ack record or data record not defined no processing done
 ======================================================================

Another thing I see, and maybe you can explain:

The record/message that goes wrong at startup is decoded as a T020103 layout.
When this record is processed correctly (after restart) the following records/messages are decoded as a T020104 layout.
Are these different type of messages?

And at last, for your info : My settings:

===========================================================================
2024-09-25T19:16:33.049145000Z Grott records whitelisted :  {'5119', '5104', '0105', '501b', '5016', '0103', '5250', '5116', '0119', '5105', '0116', '0120', '5050', '5150', '0150', '5103', '5129', '5219', '0104', '5005', '5003', '5004', '5019', '5229', '5216'}
2024-09-25T19:16:33.049180000Z 
2024-09-25T19:16:33.049211000Z Grott settings:
2024-09-25T19:16:33.049247000Z 
2024-09-25T19:16:33.049275000Z _Generic:
2024-09-25T19:16:33.049314000Z 	version:             	 2.8.3
2024-09-25T19:16:33.049344000Z 	verbose:             	 True
2024-09-25T19:16:33.050723000Z 	trace:               	 False
2024-09-25T19:16:33.050764000Z 	config file:         	 grott.ini
2024-09-25T19:16:33.050796000Z 	minrecl:             	 1
2024-09-25T19:16:33.050828000Z 	decrypt:             	 True
2024-09-25T19:16:33.050858000Z 	compat:              	 False
2024-09-25T19:16:33.050888000Z 	invtype:             	 default
[... 36 lines cut ...]
2024-09-25T19:16:33.052316000Z _Influxdb:
2024-09-25T19:16:33.052347000Z 	influx:             	 False
2024-09-25T19:16:33.052894000Z 	influx2:            	 False
2024-09-25T19:16:33.052954000Z 	database:           	 grottdb
2024-09-25T19:16:33.053001000Z 	ip:                 	 localhost
2024-09-25T19:16:33.053048000Z 	port:               	 8086
2024-09-25T19:16:33.053085000Z 	user:               	 grott
[... cut ...]
2024-09-25T19:16:33.053274000Z 	extension:          	 False
2024-09-25T19:16:33.053308000Z 	extname:            	 grottext
2024-09-25T19:16:33.053335000Z 	extvar:             	 {'none': 'none'}
2024-09-25T19:16:33.053363000Z 
2024-09-25T19:16:33.053395000Z 
2024-09-25T19:16:33.053427000Z Grott proxy mode started
2024-09-25T19:16:33.053457000Z Hostname : docker2
2024-09-25T19:16:33.053501000Z IP :  192.168.1.247 , port :  5279 
2024-09-25T19:16:33.053540000Z 
===============================================================================

Regards,
Ed

Supered1967, 2024-09-26:
Johan,

I tried to change the "minrecl" to 1, but the grott/script crashed this morning when the first data arrived.
It tried to compare an "int" (value received) with a "str" (value entered in docker ENV)

Regards
Ed

2024-09-26T06:23:57.750257000Z 
2024-09-26T06:23:57.750286000Z Grott settings:
2024-09-26T06:23:57.750321000Z 
2024-09-26T06:23:57.750351000Z _Generic:
2024-09-26T06:23:57.750385000Z 	version:             	 2.8.3
2024-09-26T06:23:57.750423000Z 	verbose:             	 True
2024-09-26T06:23:57.750570000Z 	trace:               	 False
2024-09-26T06:23:57.750618000Z 	config file:         	 grott.ini
2024-09-26T06:23:57.750648000Z 	minrecl:             	 1
2024-09-26T06:23:57.750683000Z 	decrypt:             	 True
2024-09-26T06:23:57.750712000Z 	compat:              	 False
2024-09-26T06:23:57.750744000Z 	invtype:             	 default
2024-09-26T06:23:57.750782000Z 	invtypemap:          	 {}
2024-09-26T06:23:57.750817000Z 	include_all:         	 False
2024-09-26T06:23:57.750847000Z 	blockcmd:            	 False
2024-09-26T06:23:57.750880000Z 	noipf:               	 False
2024-09-26T06:23:57.750910000Z 	time:                	 auto
2024-09-26T06:23:57.750943000Z 	sendbuf:             	 True
2024-09-26T06:23:57.750980000Z 	timezone:            	 local
2024-09-26T06:23:57.751009000Z 	valueoffset:         	 6
2024-09-26T06:23:57.751037000Z 	offset:              	 6
2024-09-26T06:23:57.751068000Z 	inverterid:          	 CS35160072
2024-09-26T06:23:57.751097000Z 	mode:                	 proxy
2024-09-26T06:23:57.751127000Z 	grottip              	 default
2024-09-26T06:23:57.751161000Z 	grottport            	 5279
2024-09-26T06:23:57.751191000Z _MQTT:
2024-09-26T06:23:57.751223000Z 	nomqtt               	 False
2024-09-26T06:23:57.751252000Z 	mqttip:              	 192.168.1.243
2024-09-26T06:23:57.751302000Z 	mqttport:            	 1883
2024-09-26T06:23:57.751345000Z 	mqtttopic:           	 energy/growatt
2024-09-26T06:23:57.751388000Z 	mqttmtopic:          	 False
2024-09-26T06:23:57.751436000Z 	mqttmtopicname:      	 energy/meter
2024-09-26T06:23:57.751469000Z 	mqttinverterintopic: 	 False
2024-09-26T06:23:57.751507000Z 	mqtttretain:         	 False
2024-09-26T06:23:57.751540000Z 	mqtttauth:           	 True
2024-09-26T06:23:57.751575000Z 	mqttuser:            	 usr
2024-09-26T06:23:57.751615000Z 	mqttpsw:             	 **secret**
2024-09-26T06:23:57.751646000Z _Growatt server:
2024-09-26T06:23:57.751689000Z 	growattip:           	 8.209.71.240
2024-09-26T06:23:57.751737000Z 	growattport:         	 5279
2024-09-26T06:23:57.751777000Z _PVOutput:
2024-09-26T06:23:57.751808000Z 	pvoutput:            	 False
2024-09-26T06:23:57.751837000Z 	pvdisv1:             	 False
2024-09-26T06:23:57.751869000Z 	pvtemp:              	 False
2024-09-26T06:23:57.751903000Z 	pvurl:               	 https://pvoutput.org/service/r2/addstatus.jsp
2024-09-26T06:23:57.751941000Z 	pvapikey:            	 yourapikey
2024-09-26T06:23:57.751976000Z 	pvinverters:         	 1
2024-09-26T06:23:57.752014000Z 	pvsystemid:          	 systemid1
2024-09-26T06:23:57.752041000Z _Influxdb:
2024-09-26T06:23:57.752074000Z 	influx:             	 False
2024-09-26T06:23:57.752107000Z 	influx2:            	 False
2024-09-26T06:23:57.752140000Z 	database:           	 grottdb
2024-09-26T06:23:57.752174000Z 	ip:                 	 localhost
2024-09-26T06:23:57.752220000Z 	port:               	 8086
2024-09-26T06:23:57.752264000Z 	user:               	 grott
2024-09-26T06:23:57.752301000Z 	password:           	 **secret**
2024-09-26T06:23:57.752345000Z 	organization:       	 grottorg
2024-09-26T06:23:57.752387000Z 	bucket:             	 grottdb
2024-09-26T06:23:57.752419000Z 	token:              	 **secret**
2024-09-26T06:23:57.752448000Z _Extension:
2024-09-26T06:23:57.752509000Z 	extension:          	 False
2024-09-26T06:23:57.752555000Z 	extname:            	 grottext
2024-09-26T06:23:57.752588000Z 	extvar:             	 {'none': 'none'}
2024-09-26T06:23:57.752617000Z 
2024-09-26T06:23:57.752644000Z 
2024-09-26T06:23:57.752687000Z Grott proxy mode started
2024-09-26T06:23:57.752723000Z Hostname : docker2
2024-09-26T06:23:57.752765000Z IP :  192.168.1.247 , port :  5279 
2024-09-26T06:23:57.752811000Z 
2024-09-26T06:24:59.448557000Z 	 - ('192.168.1.30', 19300) has connected
2024-09-26T06:24:59.449277000Z Traceback (most recent call last):
2024-09-26T06:24:59.449362000Z   File "/app/grott.py", line 35, in <module>
2024-09-26T06:24:59.449417000Z     proxy.main(conf)
2024-09-26T06:24:59.449469000Z   File "/app/grottproxy.py", line 144, in main
2024-09-26T06:24:59.449538000Z 
2024-09-26T06:24:59.449593000Z 	 - Growatt packet received:
2024-09-26T06:24:59.449641000Z 		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('192.168.1.247', 60332), raddr=('8.209.71.240', 5279)>
2024-09-26T06:24:59.449676000Z     self.on_recv(conf)
2024-09-26T06:24:59.449711000Z   File "/app/grottproxy.py", line 236, in on_recv
2024-09-26T06:24:59.451693000Z     if len(data) > conf.minrecl :
2024-09-26T06:24:59.451820000Z        ^^^^^^^^^^^^^^^^^^^^^^^^
2024-09-26T06:24:59.451872000Z TypeError: '>' not supported between instances of 'int' and 'str'

johanmeijer, 2024-09-27:
How did you specify gminrecl? gminrecl=1 or ="1" ?

Supered1967, 2024-09-27:
Just 1. Its in the ENV setting in the container, using Portainer as front-end for the Container management

Supered1967, 2024-09-29:
Good morning Johan,

Here are the logs of this morning, with minrecl=1 and blockcmd=false

As you can see in the end, it looks like an endless loop every 5 seconds.

Hope the logs will get you somewhere.....
(Complete log file also attached)

Regards
Ed

2024-09-28T17:22:07.052369000Z 	 - Growatt packet received:
2024-09-28T17:22:07.052420000Z 		  <socket.socket fd=5, family=2, type=1, proto=0, laddr=('192.168.1.247', 5279), raddr=('192.168.1.30', 45313)>
2024-09-28T17:22:07.052454000Z 	 - Growatt original Data:
2024-09-28T17:22:07.052487000Z 		 \x00\x01\x00\x02\x00\x0c\x01\x16\x41\x48\x34\x35\x31\x36\x30\x36\x33\x36
2024-09-28T17:22:07.052523000Z 	 - Grott automatic protocol detection
2024-09-28T17:22:07.052555000Z 	 - Grott data record length 18
2024-09-28T17:22:07.052585000Z 	 - layout   :  T020116
2024-09-28T17:22:07.052618000Z 	 - no matching record layout found, try generic
2024-09-28T17:22:07.052647000Z 	 - Record layout used :  T020116
2024-09-28T17:22:07.052680000Z 	 - Growatt data decrypted V2
2024-09-28T17:22:07.052709000Z 	 - Grott Growatt data decrypted
2024-09-28T17:22:07.052740000Z 	 - Growatt plain data:
2024-09-28T17:22:07.052771000Z 		 00010002000c0116063a5b42504244714159
2024-09-28T17:22:07.052797000Z 	 - Grott data ack record or data record not defined no processing done
2024-09-28T17:37:09.775577000Z 	 - Grott connection error
2024-09-28T17:37:09.777010000Z 	 - peer has disconnected
2024-09-29T05:23:51.892846000Z 	 - ('192.168.1.30', 27397) has connected
2024-09-29T05:23:51.893474000Z 
2024-09-29T05:23:51.893536000Z 	 - Growatt packet received:
2024-09-29T05:23:51.893581000Z 		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('192.168.1.247', 47956), raddr=('8.209.71.240', 5279)>
2024-09-29T05:23:51.893620000Z 	 - Growatt original Data:
2024-09-29T05:23:51.893657000Z 		 \x00\x01\x00\x02\x00\x0c\x01\x16\x41\x48\x34\x35\x31\x36\x30\x36\x33\x36
2024-09-29T05:23:51.893699000Z 	 - Grott automatic protocol detection
2024-09-29T05:23:51.893737000Z 	 - Grott data record length 18
2024-09-29T05:23:51.893780000Z 	 - layout   :  T020116
2024-09-29T05:23:51.893817000Z 	 - no matching record layout found, try generic
2024-09-29T05:23:51.893856000Z 	 - Record layout used :  T020116
2024-09-29T05:23:51.893892000Z 	 - Growatt data decrypted V2
2024-09-29T05:23:51.893925000Z 	 - Grott Growatt data decrypted
2024-09-29T05:23:51.893961000Z 	 - Growatt plain data:
2024-09-29T05:23:51.893999000Z 		 00010002000c0116063a5b42504244714159
2024-09-29T05:23:51.894033000Z 	 - Grott data ack record or data record not defined no processing done
2024-09-29T05:23:52.045785000Z 
2024-09-29T05:23:52.045905000Z 	 - Growatt packet received:
2024-09-29T05:23:52.045967000Z 		  <socket.socket fd=5, family=2, type=1, proto=0, laddr=('192.168.1.247', 5279), raddr=('192.168.1.30', 27397)>
2024-09-29T05:23:52.046012000Z 	 - Growatt original Data:
2024-09-29T05:23:52.046063000Z 		 \x00\x01\x00\x02\x00\x10\x01\x19\x41\x48\x34\x35\x31\x36\x30\x36\x33\x36\x00
2024-09-29T05:23:52.046109000Z 		 \x04\x00\x15
2024-09-29T05:23:52.046165000Z 	 - Grott automatic protocol detection
2024-09-29T05:23:52.046219000Z 	 - Grott data record length 22
2024-09-29T05:23:52.046258000Z 	 - layout   :  T020119
2024-09-29T05:23:52.046288000Z 	 - no matching record layout found, try generic
2024-09-29T05:23:52.046322000Z 	 - Record layout used :  T020119
2024-09-29T05:23:52.046361000Z 	 - Growatt data decrypted V2
2024-09-29T05:23:52.046394000Z 	 - Grott Growatt data decrypted
2024-09-29T05:23:52.046426000Z 	 - Growatt plain data:
2024-09-29T05:23:52.046466000Z 		 0001000200100119063a5b4250424471415977657461
2024-09-29T05:23:52.046519000Z 	 - Grott data ack record or data record not defined no processing done
2024-09-29T05:23:52.100883000Z 
2024-09-29T05:23:52.100975000Z 	 - Growatt packet received:
2024-09-29T05:23:52.101271000Z 		  <socket.socket fd=5, family=2, type=1, proto=0, laddr=('192.168.1.247', 5279), raddr=('192.168.1.30', 27397)>
2024-09-29T05:23:52.101328000Z 	 - Growatt original Data:
2024-09-29T05:23:52.101363000Z 		 \x00\x01\x00\x02\x00\x0c\x01\x16\x41\x48\x34\x35\x31\x36\x30\x36\x33\x36
2024-09-29T05:23:52.101451000Z 	 - Grott automatic protocol detection
2024-09-29T05:23:52.101502000Z 	 - Grott data record length 18
2024-09-29T05:23:52.101545000Z 	 - layout   :  T020116
2024-09-29T05:23:52.101580000Z 	 - no matching record layout found, try generic
2024-09-29T05:23:52.101610000Z 	 - Record layout used :  T020116
[... 212 lines cut ...]
2024-09-29T05:24:20.892280000Z 		 \x1e\x07\x3a\x0a\x50\x12\x91\x13\x99\x58\x58\x58\x58\x58\x58\x58\x58\x58\x58
2024-09-29T05:24:20.892309000Z 		 \x00\x00\x90\x41\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x07\x35\x09\xdd\x12
2024-09-29T05:24:20.892337000Z 		 \x91\x13\x99\x07\x80\x0a\x28\x12\x69\x13\xc1\x00\x64\x02\x01\x00\x2d\x00\x59
2024-09-29T05:24:20.892385000Z 		 \x00\x00\x00\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x96\x00\x96\x00\x05\x00
2024-09-29T05:24:20.892427000Z 		 \x05\x00\x96\x00\x96\x00\x03\x00\x03\x20\x20\x20\x50\x56\x20\x49\x6e\x76\x65
2024-09-29T05:24:20.892470000Z 		 \x72\x74\x65\x72\x20\x20\x53\x42\x41\x41\x33\x34\x30\x30\x00\x00\x00\x00\x01
2024-09-29T05:24:20.892522000Z 		 \x2c\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
2024-09-29T05:24:20.892563000Z 		 \x00\x00\x00\x00\x00\x00\x00\x53\x00\x55\x00\x57\x00\x59
2024-09-29T05:24:20.892591000Z 	 - Grott automatic protocol detection
2024-09-29T05:24:20.892622000Z 	 - Grott data record length 223
[... cut ...]
2024-09-29T05:24:20.892920000Z 		 cee66f47e6f600664a0741045736f5a612d7447726f7762747447726f7761e274d1726a776474
2024-09-29T05:24:20.892993000Z 		 e247e46f7461775467523f21413d1a31171d0304065467212d3620474077426f776174756b726
2024-09-29T05:24:20.893026000Z 		 e7761747447726f7761747447726f7761747447726f776127741272387738
2024-09-29T05:24:20.893057000Z 	 - Grott data ack record or data record not defined no processing done

[_grott_with_min_rect_lebth_blockcmd_false_logs.txt](https://github.com/user-attachments/files/17178007/_grott_with_min_rect_lebth_blockcmd_false_logs.txt)

johanmeijer, 2024-10-03:
It is a strange problem. I have not seen this before. 

I see a lot of: 

     Grott - grottproxy - Invalid data record received, processing stopped for this record 

That is not normal. This means that the received record is not ok. The only thing is that grott does not log the wrong record so I do not know why it is wrong (my bad, failed to put that in the program). It can be a really crippled record, but it can also be a timing issue that causes that the network data buffer is not totally filled or filled with 2 records (I have seen this while programming grottserver and solved this there). 

It is weird that when the communication looks ok, the invalid data records are there also. 
The communication with growatt is normally established and the T020103 init record is being confirmed by Growatt and then the real data is being sent (T020104). 

If your inverter/datalogger does not get the confirmation on the init record, it will sent it over and over again (the loop). 
Maybe then in one of the invalid records there is a record that is needed for growatt to establish  the communication flow. 

I really do not understand why it works when grott is restarted. I must have to do with timing or so. 

In the older versions Grott did not checked the records and they were always sent to Growatt. Have you start using a newer Grott version lately? The record validation is introduced in version 2.7.5. 

Is it difficult for you to fall back to an older docker release (e.g. 2.7.2) and test if everything works. Maybe you can also create a log file with minrecl=1 with the older version. Probably the "invalid" records are printend then (while not detected as invalid records and send to growatt). 

Otherwise I have to change grott to see what is in the invalid record, hopefully that is the key to the problem. This will take some time.

johanmeijer, 2024-10-03:
By the way, thank you for sending the logs and point me to this problem. When solved it will improve the reliability of grott.

Supered1967, 2024-10-03:
Johan,

Will reply tomorrow in full.
What I do know in the past (don’t know which version) that on initial start of the inverter, Grott showed a device ID with random characters (even & and %), and they where send to HA as device XXXXXXXX.
Probably that was a decoding of an invalid record. I altered your script in that old version to always send the right device ID to HA.

I will try to get an older docker release and test what happens then.

thanks for your cooperation!

(By the way, the inverter is 8 years old)

regards
Ed

Supered1967, 2024-10-04:
> By the way, thank you for sending the logs and point me to this problem. When solved it will improve the reliability of grott.

You're welcome Johan. Always glad to help out!

Supered1967, 2024-10-04:
Johan,

See issue 514. This issue has connection with issue 600.
The log in issue 514 was recorded in grott 2.7.2. The issue was that grott was sending serial nr XXXXXXX due to "dataerror" when inverter starts..
Maybe that log will help,
I currently deployed an 2.7.2 with minrectl=1. Have to wait until the sun goes up ;-)
Keep you posted

Ed

Supered1967, 2024-10-04:
Johan,

This morning running Grott 2.7.2 with minrecl = 1.

I have 3 scenarios.

Scenario 1 : Grott is running before the inverter starts. This is logfile : _2.7.2.grott_with_min_rect_lebth_blockcmd_false_logs_inverter_start.txt

Scenario 2: Restarted Grott while inverter is running. This is logfile : _2.7.2.grott_with_min_rect_lebth_blockcmd_false_logs_restart_grott_while_inverter_on.txt

Scenario 3 : Reboot of docker while inverter is running. This logfile : _2.7.2.grott_with_min_rect_lebth_blockcmd_false_logs_docker_reboot_while_inverter_on.txt

Only Scenario 3 gives working system.

Hope the logs can help.

Ed[
[_2.7.2.grott_with_min_rect_lebth_blockcmd_false_logs_inverter_start.txt](https://github.com/user-attachments/files/17254658/_2.7.2.grott_with_min_rect_lebth_blockcmd_false_logs_inverter_start.txt)
[_2.7.2.grott_with_min_rect_lebth_blockcmd_false_logs_docker_reboot_while_inverter_on.txt](https://github.com/user-attachments/files/17254659/_2.7.2.grott_with_min_rect_lebth_blockcmd_false_logs_docker_reboot_while_inverter_on.txt)
[_2.7.2.grott_with_min_rect_lebth_blockcmd_false_logs_restart_grott_while_inverter_on.txt](https://github.com/user-attachments/files/17254661/_2.7.2.grott_with_min_rect_lebth_blockcmd_false_logs_restart_grott_while_inverter_on.txt)
](url)

johanmeijer, 2024-10-08:
Thanks. It is indeed what I thought two records in one buffer: 

	Growatt original Data:
	2024-10-04T05:33:13.233520000Z 		 \x00\x01\x00\x02\x00\x11\x01\x19\x41\x48\x34\x35\x31\x36\x30\x36\x33\x36\x00
	2024-10-04T05:33:13.233605000Z 		 \x05\x00\x01\x31\x00\x01\x00\x02\x00\x12\x01\x19\x41\x48\x34\x35\x31\x36\x30
	2024-10-04T05:33:13.233697000Z 		 \x36\x33\x36\x00\x06\x00\x02\x33\x32\x00\x01\x00\x02\x00\x1a\x01\x19\x41\x48
	2024-10-04T05:33:13.233757000Z 		 \x34\x35\x31\x36\x30\x36\x33\x36\x00\x08\x00\x0a\x41\x48\x34\x35\x31\x36\x30
	2024-10-04T05:33:13.233796000Z 		 \x36\x33\x36\x00\x01\x00\x02\x00\x11\x01\x19\x41\x48\x34\x35\x31\x36\x30\x36
	2024-10-04T05:33:13.233831000Z 		 \x33\x36\x00\x0a\x00\x01\x30

	Is record: 

	2024-10-04T05:33:13.233520000Z 		 \x00\x01\x00\x02\x00\x11\x01\x19\x41\x48\x34\x35\x31\x36\x30\x36\x33\x36\x00
	2024-10-04T05:33:13.233605000Z 		 \x05\x00\x01\x31\x00\x01\x00\x02\x00\x12\x01\x19\x41\x48\x34\x35\x31\x36\x30
	2024-10-04T05:33:13.233697000Z 		 \x36\x33\x36\x00\x06\x00\x02\x33\x32\x00\x01\x00\x02\x00\x1a\x01\x19\x41\x48
	2024-10-04T05:33:13.233757000Z 		 \x34\x35\x31\x36\x30\x36\x33\x36\x00\x08\x00\x0a\x41\x48\x34\x35\x31\x36\x30
	2024-10-04T05:33:13.233796000Z 		 \x36\x33\x36\

	And: 
	2024-10-04T05:33:13.233796000Z 		 \x00\x01\x00\x02\x00\x11\x01\x19\x41\x48\x34\x35\x31\x36\x30\x36
	2024-10-04T05:33:13.233831000Z 		 \x33\x36\x00\x0a\x00\x01\x30

Although I am not totally sure this the problem. The last log looks a little bit strange for me while is seems to have a totally other start cycle. 

Did it worked ok with 2.7.2 or does it also gives the same problems? 

Is it possible for you the makes changes in the python code (either Grott native or Grott Docker). Maybe changing timings can help. 

Can you change by example 

    delay = 0.0002 
in 
    delay = 0.0001 or delay = 0.00015

This will speed up Grott receive/send processing. 

If this not works I have to change Grott to split the records. As said I already did this for grottserver but have to change this grottproxy also (en create new images etc).

Supered1967, 2024-10-08:
Johan,

these logs are from an 2.7.2.

My system worked for months until a few weeks ago.
Maybe Growatt updated the firmware or whatever, but the Grott docker was working for months without any restart.

so I installed/deployed a new docker in proxmox and deployed the latest Grott beta, hoping that that would help.

whatever version (2.7.2) or beta, issue stays the same.

I will change the timing and let you know the results/logs.

thanks for your great support (and product)

Supered1967, 2024-10-09:
Johan,

Can it be the TCP stack on the docker LXC that is running in Proxmox?
Because reboot of the docker LXC is the only way to get the Grott running?

I will see in the meanwhile if I can create an other docker LXC (different brand).

Regards

johanmeijer, 2024-10-10:
It might be possible I have no experience with LXC. It has something to do with communication / timing. A multi layered network will always be a challenge I think. 

Proxmox is at my  todo list (but that one is to big, likely the winter is coming).  

Why do you use docker. Is it also possible to run Grott without docker on LXC?

Supered1967, 2024-10-10:
Johan, I used docker because it was easy for a Windows minded guy (not a Linux specialist)

I will see if I can install Grott on a Linux VM.
Perhaps some hints how to ? ;-)

regards
Ed

johanmeijer, 2024-10-13:
Hi @Supered1967 have you already test with changing the timing. 

To install grott on linux you need some basic Linux knowledge but that is not that difficult. 

- install git (depend on your linux repo e.g. sudo apt install git
- Install prereq python modules (see: https://github.com/johanmeijer/grott/wiki/@-First-time-installation)
- git clone -b 'Master' https://github.com/johanmeijer/grott  (install Master or other branche)
- cd grott (go to the grott directory)
- nano grott.ini (edit / modify the grott.ini, I use nano as editor might not be there VI always is but a little bit more complex)
- Run grott as a service: https://github.com/johanmeijer/grott/wiki/Grott-as-a-service-(Linux)

Hopefully this helps a bit.

Supered1967, 2024-10-13:
Johan,

I did indeed tested with different timings, but that didn’t solve the issue.

I installed a different docker on Proxmox (Debian VM in stead of Debian LXC) , but that also didn’t solve issue. But….. there is a difference:
Proxmox-Debian LXC-Docker-Grott : Have to restart the Debian LXC to get Grott working
Proxmox-Debian VM-Docket-Grott : Have to restart Grott container to get Grott working.

I now installed Grott on Synogy Docker (did that in the past also), have to see tomorrow morning the results.

Meanwhile I will install Grott in Debian : Proxmox-Debian VM-Grott following your guides .

Thanks again!

Supered1967, 2024-10-20:
Johan,

I did deploy some more versions on different platforms, but they all have the same (kind of) issue.

Proxmox-Debian LXC-Docker-Grott : Have to restart the Debian LXC to get Grott working
Proxmox-Debian VM-Docket-Grott : Have to restart Grott container to get Grott working.
Synology-Docker-Grott : Reboot of the docker doesn't help, only reboot of complete NAS
Proxmox-Debian VM-Grott (2.6.1h) : Have to reboot the Debian VM to get Grott working.

Also tried different delays, but that didn't help.

The next step I will try is to reboot the Shine Wifi-S when the inverter and Grott is running, lets see what that does.

Strange thing is still, that it all worked for months until a few weeks ago. Perhaps Growatt has deployed a new firmware to the dongle? I will see if I can find its firmware build.

Regards
Ed

Supered1967, 2024-10-22:
Johan,

I changed the Shine-Wifi-S to connect directly to server.growatt.com, and got the same kind of error.
This morning the dongle says "connected", but no data on server.growatt.com (device disconnected).
I rebooted the Shine-Wifi-S and directly see data on server.growatt.com.

I found out that the firmware on the Shine-Wifi-S is/was version 3.0 (I believe it was 2.5 a few months ago).

I tried to find a newer firmware, but that isn't available as downloadable file.
But, on the server.growatt.com -> scroll down -> all devices -> datalogger -> datalogger setting -> firmware update I was able to upgrade to latest 4.0 release. (after finding out that the password for upgrading is current YYYYMMDD)
Now I have to wait until tomorrow mornning when the sun goes up.

Lets hope the new firmware may be the solution, maybe the datalogger was faulty

Keep you posted

johanmeijer, 2024-10-22:
Ok. I am curious.

Supered1967, 2024-10-23:
Upgrading the firmware to 4.0.0 did not solve the issue.
To get the data I need to restart the datalogger.

From the growatt.com site I did now a “reset datalogger log”, Lets see of that helps tomorrow.
If not, I will reset the logger to default and deploy configuration.

Do you perhaps know if the logger has memory where it can store inverter messages? (Buffer)
Can it be that there is a (corrupt) message still pending in its memory that it sends everytime the datalogger starts?

johanmeijer, 2024-10-23:
Yes the datalogger has a memory to store "unsend" inverter data (e.g. if there are connection problems). I do not think it stores the real records. The records are created at the moment the datalogger tries sent (the data will be packed in a data record then). 

If I look at your logs. The invalid records are not data records but records needed for the communication (Ping, Configuration and announce records). These are sent before the real data is being sent. 

I do not know why but your config "stack" records. So there are multiple records in one buffer. Grott verifies the record length with the information of the first record but that does not match the length of the buffer (with multiple records). 
In the Grott next release I have code to split these records. I will also port that back to the 2.8.3 release. Maybe this will help. 

If you live in the neighbourhood of Amsterdam you can try my spare wifi-s datalogger to see if it is a datalogger issue. 

Can you also try to run it without a fix IP address for the growatt server (remove the ggrowattip) it will then use the Grott default and that is the server.growatt.com DNS entry. Maybe it has to pick an other IP address sometimes. (I am also seeking for the needle in the haystack).

johanmeijer, 2024-10-23:
I have changed the  2.8.3. Beta version to split stacked records. Can you try this one? ( I do not have created a docker container yet).

Supered1967, 2024-10-23:
Thanks Johan!

I Will try the latest beta tomorrow and see what that does :-)

But the strange thing remains is that when I don’t use Grott but program the datalogger directly to server.growatt.com there is also no data when the inverter has started.
The datalogger say “connected” but in my environment on server.growatt.con it says datalogger = offline.
When I restart the datalogger the connection comes on-line.
(Maybe a complete reset of the datalogger will help, or my datalogger is broken)

Thanks for your suggestion to lent me a spare datalogger, but I live in Drenthe…. So a bit far from Amsterdam.

perhaps I will create a Wireshark log on my outgoing connection and monitor the datalogger from the datalogger to Growatt (bypassing Grott to see what is happening)

we keep in touch, thanks again for your time

Supered1967, 2024-10-29:
Johan,

The new beta doesn’t solve the issue :-(

I tried everything now, even resetting the data logger to factory.
I noticed that the led on the data logger also fails.
So, I ordered a new data logger (was just 18€) en will install it tomorrow and see if that helps.

I guess (and hope) the data logger is the problem, not Grott.

Regards
Ed

Supered1967, 2024-11-01:
Good morning Johan,

2 days ago I installed a new ShineWifi-S.
The initial problem is gone, the system now works 100% with and without Grott, meaning that when the inverter starts in the morning data is transmitted and processed as it should be. 
There is a thing I see on the server.growatt.com if I look at my (now 2) data loggers.
The “old” (white) data logger is seen as a “ShineWifiBox” with firmware 4.0
The “new” (black) data logger is seen as a “ShineWifi-S” with firmware 1.7.7.7
I think that perhaps this might be the issue. For some reason my old WiFi-s is flashed as a WiFi box? (I guess that both are esp modules)

For me, the problem is solved and in the end had nothing to do with Grott.
But I want to thank you for your patience and support solving the issue.

case can be closed!

Regards
Ed

johanmeijer, 2024-11-04:
I am glad the problem is solved. It also learned me some thing that will improve the next versions of Grott. 

How does a ShinewfiBox looks like. I know the shinelanbox. I think I never saw a shinewifibox (can also not find it a the internet). 

I have a shinelanbox with firmware 7.0.0.7 (with shine rf stick), a shinewifi-s with firmeware 3.0.0.2 and a shinewifi-s with firmware 1.7.7.7 (not in use at this moment).

Supered1967, 2024-11-05:
Johan,

When looking at my growatt dashboard, the picture for the shinewifibox is the same as the wifi-s.
I attached a screenshot from the dashboard. I guess its an undocumented feature I have ;-)

I suspect that for reason Growatt flashed it with a wrong firmware........

Regards

[image]

Supered1967, 2024-11-09:
Issue closed as it was not a Grott issue but a faulty shine-s (for some reason flashed as a WiFibox 4.0.0). Ordered a new shinewifi-s and problem solved.

closed 2024-11-09 by Supered1967