johanmeijer/grott #602: MOD-3-10KTL3-XH support

RockyVod, 2024-10-17:
Hi,

I have inverter MOD-3-10KTL3-XH with SDM630 modbus v2 smart meter.
basically your addon is working, but I would like to add all data that I get from system.
How can I add custom template to forward all data?
I am using stable 2.8.3 egg addon for HA and muppet component.

Growatt original Data:
\x04\x14\x00\x06\x01\x21\x01\x38\x1f\x35\x2b\x42\x23\x33\x38\x75\x34\x56\x77
\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
\x29\x24\x31\x44\x30\x1f\x3a\x5f\x46\x36\x74\x74\x47\x72\x6f\x77\x61\x74\x74
\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x79\x7e\x7b\x4d\x5d\x5f\x76\x6e
\xd4\x64\x4c\x42\x3a\x26\x2c\x45\x44\x15\x40\x5c\x43\x33\x20\x44\x77\x20\x3c
\x77\x61\x76\x75\x47\x72\x6f\x77\x7d\xbb\x8b\xaa\x72\x31\x6b\xae\x7a\x1c\x47
\x50\x6f\x77\x60\xa9\x75\x3e\x72\x6f\x40\xf7\x74\x74\x47\x72\x6f\x77\x7c\x7f
\x74\x47\x55\x82\x77\x61\x74\x77\x49\x15\x6f\x76\x61\x6d\x75\x4f\x73\x5d\x77
\x81\x74\x77\x47\x43\x6f\x28\x61\x29\x74\x43\x72\x6f\x76\xe7\x74\x74\x47\x72
\x6f\x77\x61\x74\x74\x47\x72\x6f\x63\x87\x74\x74\x5a\xc2\x6f\x77\x61\x74\x74
\x45\x72\x6f\x77\x61\x74\x2a\x48\x63\x90\xab\x61\x74\x71\x3f\x7a\xf7\x77\x61
\x74\x10\x45\x4a\x6e\xa7\x61\x74\x74\x47\x77\x17\x77\x61\x74\x75\x4b\x87\x63
\x99\x63\x65\x74\x47\x72\x6f\x77\xa9\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x75\x46\x72\x6f\x6a\x2d\x74
\x74\x47\x72\x6c\xdb\x61\x74\x74\x47\x52\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
\x61\x74\x74\x47\x72\x6f\x77\x61\x65\x61\xd7\x61

Grott automatic protocol detection
Grott data record length 297
layout : T060138MIN
no matching record layout found, try generic
Record layout used : T060138MIN
Growatt data decrypted V2
Grott Growatt data decrypted
Growatt plain data:
04140006012101385847443542474c32463900000000000000000000000000000000000000004
65350304458483031570000000000000000000000000000000000000000180a0f0a2f30010fa0
100b3055514d31305232333452543030525300000201000000001ccfffed005e1ccf0e6800220
00001dd0179000037960000000000001d0b000027ed000000030e67000100190108013200e000
030031005f005d00040000018600000000000000000000000014e600001db0000000000002000
00000005e0f11ffdc00000578089800000064023801d0000000000578000000010cf50cee0211
0000000000c800000000000000000000000000000000000000000000010100001d4c000000000
3ac0000000020000000000000000000000000000000000011159013
Grott data ack record or data record not defined no processing done
Grott automatic protocol detection

Grott data record length 413
layout : T060138XMIN
no matching record layout found, try generic
Record layout used : T060138XMIN
Growatt data decrypted V2
Grott Growatt data decrypted
Growatt plain data:

FlexxFR, 2024-12-10:
I have a similar request for a MOD 8000TL3-XH
Using grott in a docker container and the HA muppet component to "translate" the MQTT message.

I tried to set invertor type to MOD but then it's worse. The default settings at least read most values correctly. So, that it good. I'm missing any battery related values (which might be explained as do not have one connected).

What do I need to do?
Is it still needed to do something special for a MOD 8000TL3-XH in the config? 
Do you need any logs, or can I do some "translation"

johanmeijer, 2024-12-10:
@RockyVod  this does not seems to be a data record. 
Grott expect a data record that ask for T060104MOD layout.

FlexxFR, 2024-12-10:
@johanmeijer ,  this is what I logged.
I tried with different settings, and without any changes to grott.ini (except for MQTT settings) I had most luck (basic values working). I tried to use invertor setting MOD (and mod and tl3 and ginvertor setting) but that was less successful.

What is the correct config for this type?

```
	 - Grott Send data to PVOutput disabled 
	 - Grott Send data to Influx disabled 
	 - Grott extension processing disabled 
	 - Growatt packet received:
		  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.112.2', 54742), raddr=('47.91.67.66', 5279)>
	 - Growatt original Data:
		 \x00\xee\x00\x06\x02\x41\x51\x04\x09\x33\x2c\x36\x22\x39\x4c\x75\x47\x5e\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x3e\x35\x2d\x45\x37\x0c\x4b\x5f\x4f\x22\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x75\x6a
		 \xcc\x78\x73\x72\x6f\x77\x61\x74\x74\x44\xf1\x6f\x77\x61\x74\x74\x47\x76\xed
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x73\x2e\x74\x74\x47\x72\x7c\xf2\x69\x96
		 \x74\x42\x72\x6f\x73\x10\x7c\xbe\x47\x76\x6f\x77\x62\xf0\x7c\xb4\x72\x6a\x77
		 \x61\x70\x0d\x48\x44\x60\x52\x6e\xfc\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x76\xbe\x90\x72\x6f\x77\x6b\x74\x74\x47\x1b\x6f\x77\x61\x29\x74
		 \x47\x72\x6c\x77\x61\x74\x43\x47\x72\x6f\x74\x61\x74\x74\x61\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x41\x72\x6f\x77\x3c\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x69\x77\x61\x74\x74\x64\x66\x6f\x77\x61\x74\x74\x47\x72\x6f\x70\xaa\x75
		 \x73\x46\xc9\x6f\x8f\x61\x74\x74\xaa\x7e\x6e\x7c\x8b\x3a\x54\x47\x72\x6f\x76
		 \x59\xf4\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x4e\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x7d\x78\x72\x7e\xde\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x05
		 \x77\x61\x74\x7d\x47\x72\x6f\x1d\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x53\x64\x47\x72\x78\x99\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x44\x52\x3a\x5f\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x90\xb2
	 - Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T065104X
	 - no matching record layout found, try generic
	 - Record layout used :  T06NNNNX
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00ee0006024151044e414341434d3832353100000000000000000000000000000000000000005
		 1424c31434b393038430000000000000000000000000000000000000000000000000000020bb8
		 0c340000000000000383000000000000048200000000000000000000000000000000000000000
		 000000000000000044f00000000138508e200050000047108ca00040000038408f30005000004
		 790f360f250f880000000000000000000000000002cad70000000a000000690000005d0000000
		 30000003700000003000000260000000000000000000000000000000000000000000000000000
		 00060000005d000000000000000000000006000000002314000000000000000007cb010701bb0
		 0f8000000ed0c010bea4e20000000013880000000000000000000000000000000000000000000
		 00003c0000000000000000000000000000000000090c350cb1000000000000000000000000000
		 0000000000000000000000000006a000000090000006a00000000000000000000000000000000
		 00000000000000000000000000000000000000002710000017ee0000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000032055280000000000000000000000000
		 00000000000e2dd
	 - Possible Inverter serial XXXXX
	 - Inverter serial not recognised - using inverter type default
	 - Growatt new layout processing
[... 27 lines cut ...]
		 -  totworktime          :  25.4
		 -  pvenergytoday        :  1.0
		 -  pvenergytotal        :  10.5
		 -  epvtotal             :  9.3
		 -  epv1today            :  0.3
		 -  epv1total            :  5.5
		 -  epv2today            :  0.3
		 -  epv2total            :  3.8
		 -  pvtemperature        :  26.3
		 -  pvipmtemperature     :  23.7
	 - MQTT jsonmsg: 
			 {"device": "XXXXX", "time": "2024-12-10T20:01:23", "buffered": "no",
			 "values": {"datalogserial": "XXXXX", "pvserial": "XXXXX",
			 "pvstatus": 0, "pvpowerin": 0, "pv1voltage": 899, "pv1current": 0,
			 "pv1watt": 0, "pv2voltage": 1154, "pv2current": 0, "pv2watt": 0,
			 "pvpowerout": 0, "pvfrequentie": 4997, "pvgridvoltage": 2274,
			 "pvgridcurrent": 5, "pvgridpower": 1137, "pvgridvoltage2": 2250,
			 "pvgridcurrent2": 4, "pvgridpower2": 900, "pvgridvoltage3": 2291,
			 "pvgridcurrent3": 5, "pvgridpower3": 1145, "totworktime": 182999,
			 "pvenergytoday": 10, "pvenergytotal": 105, "epvtotal": 93, "epv1today": 3,
			 "epv1total": 55, "epv2today": 3, "epv2total": 38, "pvtemperature": 263,
			 "pvipmtemperature": 237}}
	 - Grott MQTT topic used : energy/growatt
	 - MQTT message message sent
```

johanmeijer, 2024-12-10:
Ok I think we have two different challenges here: 

@RockyVod 
This "38" record seems to  be a record for BDC and BMS information (input registers 4000-4107). This is not documented in the documentation I have, but debugging the record I have to come to this conclusion.
I did not expected this data as a separate record. But it helps with understanding some new records type. 
Is it possible that this contains BDC/BMS information (and that your BDC BMC serial nummer is : FSP0DXH01W)? 

Do you also have the "T060104XMOD" record for me (original data with the /x in it)? I am very curious if this information is not already in that record as well it should be in the registers 3125~3249 (register 3165-3233 seems to be a copy of register 4000-4107). 
If the information is already in the data record in the "3xxx" register it is very easy to extend the MOD layout with this registers. 
(btw this will automatically be solved in the Grott 3.0.0 (alpha) version. This will detected the registers in the data record and show the content). 

If we need to process the "38" record it might be a bigger change in Grott. I have to do some more investigation. 
(I will add that then to the 3.x.x. version in the future). 

@FlexxFR I am going to test with your record and see which layout fits the best for your system.

johanmeijer, 2024-12-10:
@FlexxFR 

The mod record seems to be the best fit for you (specify invtype = mod). Which version of Grott do you run (plain java, docker, ha)? Your data processed with Grott 2.8.3: 

                 -  datalogserial        :  NACACM8251
                 -  pvserial             :  QBL1CK908C
                 -  group1start          :  3000
                 -  group1end            :  3124
                 -  pvstatus             :  0
                 -  pvpowerin            :  0.0
                 -  pv1voltage           :  89.9
                 -  pv1current           :  0.0
                 -  pv1watt              :  0.0
                 -  pv2voltage           :  115.4
                 -  pv2current           :  0.0
                 -  pv2watt              :  0.0
                 -  pv3voltage           :  0.0
                 -  pv3current           :  0.0
                 -  pv3watt              :  0.0
                 -  pv4voltage           :  0.0
                 -  pv4current           :  0.0
                 -  pv4watt              :  0.0
                 -  pvpowerout           :  0.0
                 -  qac                  :  110.3
                 -  pac                  :  0.0
                 -  pvfrequency          :  50.0
                 -  pvgridvoltage        :  227.4
                 -  pvgridcurrent        :  0.5
                 -  pvgridpower          :  113.7
                 -  pvgridvoltage2       :  225.0
                 -  pvgridcurrent2       :  0.4
                 -  pvgridpower2         :  90.0
                 -  pvgridvoltage3       :  229.1
                 -  pvgridcurrent3       :  0.5
                 -  pvgridpower3         :  114.5
                 -  vacrs                :  389.4
                 -  vacst                :  387.7
                 -  vactr                :  397.6
                 -  ptousertotal         :  0.0
                 -  ptogridtotal         :  0.0
                 -  ptoloadtotal         :  0.0
                 -  totworktime          :  25.4
                 -  eactoday             :  1.0
                 -  pvenergytoday        :  1.0
                 -  eactotal             :  10.5
                 -  pvenergytotal        :  10.5
                 -  epvtotal             :  9.3
                 -  epv1today            :  0.3
                 -  epv1total            :  5.5
                 -  epv2today            :  0.3
                 -  epv2total            :  3.8
                 -  epv3today            :  0.0
                 -  epv3total            :  0.0
                 -  etousertoday         :  0.0
                 -  etousertotal         :  0.0
                 -  etogridtoday         :  0.0
                 -  etogridtotal         :  0.0
                 -  eloadtoday           :  0.6
                 -  eloadtotal           :  9.3
                 -  epv4today            :  0.0
                 -  epv4total            :  0.0
                 -  epvtoday             :  0.6
                 -  reserved3085         :  0
                 -  deratingmode         :  0
                 -  iso                  :  8980
                 -  dcir                 :  0.0
                 -  dcis                 :  0.0
                 -  dcit                 :  0.0
                 -  gfci                 :  0
                 -  busvoltage           :  199.5
[... 113 lines cut ...]
                 -  bmswarn3             :  0
                 -  bmserror3            :  0
                 -  reserved3227         :  0
                 -  reserved3228         :  0
                 -  reserved3229         :  0
                 -  bmssinglevoltmax     :  0
                 -  bmssinglevoltmin     :  0
                 -  batloadvolt          :  0.0
                 -  reserved3233         :  0
                 -  debugdata1           :  0
                 -  debugdata2           :  0
                 -  debugdata3           :  0
                 -  debugdata4           :  0
                 -  debugdata5           :  0
                 -  debugdata6           :  800
                 -  debugdata7           :  21800
                 -  debugdata8           :  0
                 -  debugdata9           :  0
                 -  debugdata10          :  0
                 -  debugdata11          :  0
                 -  debugdata12          :  0
                 -  debugdata13          :  0
                 -  debugdata14          :  0
                 -  debugdata15          :  0
                 -  debugdata16          :  0

RockyVod, 2024-12-10:
> Ok I think we have two different challenges here:
>
> @RockyVod This "38" record seems to be a record for BDC and BMS information (input registers 4000-4107). This is not documented in the [...]

Hi,

It might be BMS since I also have the battery, but the FSP0DXH01W is serial number of inverter. 
I will try to get you more messages, but had to check how did I get those. It's been a while ;)

FlexxFR, 2024-12-11:
> @FlexxFR I do understand that the default Grott layout also fits for the Basic values in this case, but do not expect battery information etc.

Hi @johanmeijer , thanks for your support. As of now I no not have a battery yet but want to make sure I have it configured as good as it gets. I'm running grott in a docker, version 2.8.3. I have some doubts it this is really the latest & greatest as in the docker comments multiple 2.8.3 updates are mentioned.

Is used this config in portainer to deploy the contrainer.

```
version: '3.7'

services:
  grott:
    container_name: grott
    image: ledidobe/grott:2.8.3
    volumes:
      - grott_config:/app/
    ports:
      - 5279:5279
    restart: unless-stopped

volumes:
  grott_config:
    external: true
```

Don't now anymore how the grott.ini came to existence :), it was either generated or I downloaded it from github. It mentions version 2.7.0. Almost everything is #-ed out, except the changes I made for MQTT to HA.

Nevertheless, I tried to change the config to mod earlier but that gave weird results. I will try again in an hour when there is some "sun" so at least have some values.

Just to verify, this should be the correct config/syntax, right?

```
[Generic]

# Specify inverter id (not necessary in version >2.1.0 if compat = false!)
#inverterid = ABC1234567
# Specify the type of the inverter (default/sph/spf/max)
invtype = mod
```

FlexxFR, 2024-12-14:
@johanmeijer , I might found (at least part of) my issue.

1. Apparently, I did not have the latest 2.8.3 installed (as I used that specific tag and not master). So, I was working with a 10-month old version, before MOD support.
2. Portainer/docker issues (again my inexperience) Redeploy didn't work immediately. Not sure why, I believe some of the .py files remained on the external storage volume. Wipe all/rebuild solved that.
3. grott.ini: the version "spawned" again was older 2.4.0, I copied over the example from your git (2.7.0), changed intertor type & MQTT setting, and more is happening..

This now seems to be fine
```
2024-12-14T08:43:33.522435757Z 	 - Grott automatic protocol detection
2024-12-14T08:43:33.522438044Z 	 - Grott data record length 585
2024-12-14T08:43:33.522440329Z 	 - layout   :  T065104XMOD
2024-12-14T08:43:33.522442603Z 	 - no matching record layout found, try generic
2024-12-14T08:43:33.522491869Z 	 - Record layout used :  T06NNNNXMOD
2024-12-14T08:43:33.524306434Z 	 - Growatt data decrypted V2
2024-12-14T08:43:33.524321407Z 	 - Grott Growatt data decrypted

2024-12-14T08:43:33.525064159Z 	 - Growatt new layout processing
2024-12-14T08:43:33.525066450Z 		 - decrypt       :  True
2024-12-14T08:43:33.525068641Z 		 - offset        :  6
2024-12-14T08:43:33.525070874Z 		 - record layout :  T06NNNNXMOD
2024-12-14T08:43:33.525073144Z 
2024-12-14T08:43:33.525828773Z 	 - Grott data record date/time processing started

```

Values appearing in HA again, all seems to be fine