johanmeijer/grott #604: ShineWIFI-F Pro support on an SPF 6000 ES Plus

hristouzunovaz, 2024-10-22:
Hey guys, I'm wondering if Grott supports this datalogger `ShineWIFI-F Pro` on this inverter `SPF 6000 ES Plus`.
I can see the documentation says `spf support` but this exact inverter is not mentioned.
I'm only asking because I have installed and configured Grott (HA add-on) and it seems to be communicating with the datalogger:

> Hostname : 1ef934e0-grott
> IP :  172.30.33.4 , port :  5279 
>
>	 - ('192.168.2.248', 63588) has connected
>
>	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.4', 51494), raddr=('8.209.71.240', 7006)>
> - Grott - grottproxy - Invalid data record received, processing stopped for this record

The logs suggest to me that the logger and my Grott instance talk to each other, however, the messages are unexpected.
If this hardware setup is not supported, what can I do to help make it supported?

johanmeijer, 2024-10-22:
As far I can see from the issue/discussion Grott works with a SPF 5000 ES and SPF 3000TL. 

I am not sure it works with a ES Pro and/or datalogger ShiniWIFI-Pro (although that seems the datalogger specific  for the SPF. 

The challenge is that if have "forgotten" to program the additional verbose record logging when a record is invalid, so it is difficult to debug the record. 

I can update the 2.8.3. version with the necessary debugging info but that will not work directly for the HA addon I think we need @egguy for that. 

Are these the only records you see at this moment? Are there other records that shows Data less then minimum record length, data not processed? If that is true we have more capabilities to debug.

johanmeijer, 2024-10-22:
I added additional debug info showing full data record for invalid datarecord to grottrproxy.py in the beta 2.8.3 version. 

Maybe we can use this one to debug. Start with the -v  option to see the additional information. @egguy  is it possible to test this one with HA addon?

hristouzunovaz, 2024-10-23:
To answer your initial comment:
Yes, these are the only logs I can see. I am starting the server with `verbose : true`.
No, there are no other records.

Regarding your latter comment: how do I get to the beta version? I'm afraid I did not see a way to do this in HA. 
I did receive an update notification for Grott this morning and did apply it.

Is there anything more I can share from my setup to assist in debugging?

johanmeijer, 2024-10-23:
Maybe @egguy can help us to test the beta in HA? He is the owner of the HA addon.

hristouzunovaz, 2024-10-26:
Hi @egguy, what would be the best way to try and debug in this case. I have installed the beta branch from the HA add-on, and of course, I get the same result.

egguy, 2024-10-27:
Hi @hristouzunovaz , I have recently updated the 2.8 add-on based on the beta branch.

You can test it on the edge addon repository for the moment: https://github.com/egguy/ha-addons-edge/

hristouzunovaz, 2024-10-28:
Thank you guys for your support!

I did as instructed by @egguy and did install the beta channel on edge. Do let me know if this is the correct thing.
Once I brought the server up I got the following log:

```
Hostname : 3bd86b2d-grott-beta
IP :  172.30.33.4 , port :  5279 

	 - ('192.168.2.248', 59367) has connected
Traceback (most recent call last):
  File "/app/grott.py", line 35, in <module>
    proxy.main(conf)
    ~~~~~~~~~~^^^^^^
  File "/app/grottproxy.py", line 150, in main
    self.on_close()
    ~~~~~~~~~~~~~^^
TypeError: Proxy.on_close() missing 1 required positional argument: 'conf'
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service base-addon-log-level: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service base-addon-log-level successfully stopped
s6-rc: info: service base-addon-banner: stopping
s6-rc: info: service base-addon-banner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

I believe the server crashes because it spills the same logs after a while, i.e. it reboots and starts from the begging.

johanmeijer, 2024-10-29:
I have uploaded some bug fixes to the Grott Beta, hopefully @egguy can generate a new HA plugin. 

Sorry for the inconvenience.

egguy, 2024-10-30:
Hi, thanks for your work @johanmeijer. 

The update has been merged as part of https://github.com/egguy/addon-grott-beta/pull/89 . The new add-on should be published soon on the edge repo @hristouzunovaz

hristouzunovaz, 2024-10-31:
Hi guys, this is a sit rep:
These are the logs I get when I start the server:

```
Hostname : 3bd86b2d-grott-beta
IP :  172.30.33.4 , port :  5279 

	 - ('192.168.2.248', 55962) has connected
	 - ('192.168.2.248', 55584) has connected
	 - ('8.209.71.240', 7006) has disconnected
	 - ('8.209.71.240', 7006) has disconnected
	 - ('192.168.2.248', 55585) has connected
```
I was expecting more logs out of this. This is the config that Grott reports:
```
Grott settings:

_Generic:
	version:             	 2.8.3
	verbose:             	 True
	trace:               	 False
	config file:         	 /config/grott/grott.ini
	minrecl:             	 100
	decrypt:             	 True
	compat:              	 True
	invtype:             	 default
	invtypemap:          	 {'HAQ0E6808E': 'SPF', 'HAQ0E6U03N': 'SPF'}
	include_all:         	 True
	blockcmd:            	 False
	noipf:               	 False
	time:                	 auto
	sendbuf:             	 True
	timezone:            	 local
	valueoffset:         	 6
	offset:              	 6
	inverterid:          	 automatic
	mode:                	 proxy
	grottip              	 default
	grottport            	 5279
_MQTT:
	nomqtt               	 False
	mqttip:              	 172.30.33.0
	mqttport:            	 1883
	mqtttopic:           	 energy/growatt
	mqttmtopic:          	 False
	mqttmtopicname:      	 energy/meter
	mqttinverterintopic: 	 False
	mqtttretain:         	 False
	mqtttauth:           	 True
	mqttuser:            	 addons
	mqttpsw:             	 **secret**
_Growatt server:
	growattip:           	 8.209.71.240
	growattport:         	 7006
_PVOutput:
	pvoutput:            	 False
	pvdisv1:             	 False
	pvtemp:              	 False
	pvurl:               	 https://pvoutput.org/service/r2/addstatus.jsp
	pvapikey:            	 yourapikey
	pvinverters:         	 1
	pvsystemid:          	 systemid1
_Influxdb:
	influx:             	 False
	influx2:            	 False
	database:           	 grottdb
	ip:                 	 localhost
	port:               	 8086
	user:               	 grott
	password:           	 **secret**
	organization:       	 grottorg
	bucket:             	 grottdb
	token:              	 **secret**
_Extension:
	extension:          	 True
	extname:            	 grottext.ha
	extvar:             	 {'ha_mqtt_host': 'core-mosquitto', 'ha_mqtt_port': '1883', 'ha_mqtt_user': 'addons', 'ha_mqtt_password': 'Oop0phaeg3theo2Jesighohphing8ahdooNahthe0Oodauthah6koo4uwoo7cohh', 'ha_mqtt_retain': False}
```
Just to confirm, this is what the add-on tells me:
```
since v0.1.7

⬆️ Update grott digest to 3bfeec8
⬆️ Update Add-on base image to v15.0.1
⬆️ Update grott digest to 4b526a8
```

Please advise further!

johanmeijer, 2024-11-04:
Hmm this are totally strange messages for me. I do not thing Grott uses web_protocol.py. 
Can this be a Home assistent errror?

egguy, 2024-11-05:
This looks like a homeassistant log. Where did you get this log?

From the look of the provided changelog from the addon, you are on the commit: https://github.com/johanmeijer/grott/commit/3bfeec866e43e73303d818dfe50e242717e3145c version, with the verbose mode enabled.

hristouzunovaz, 2024-11-05:
yeah, disregard the logs from HA, I was looking too hard into them.
The question remains, though, what is wrong with Grott. Looks like the communication between Grott and Growatt servers is not happening.
I'm not really experienced with backend stuff, but if I fork the repos and add my logs, what would be the easiest way to look into the packets and read the messages?

PedroKTFC, 2024-11-10:
> I have uploaded some bug fixes to the Grott Beta, hopefully @egguy can generate a new HA plugin.
> 
Can you point me to where the changes are please? I'm looking at recent changes to the 2.8.3(beta) branch but I'm not sure if that's the right place.

ferdsyou, 2025-01-21:
Any update on this? I have exactly the same device and have problems trying to connect to Grott, thanks

vasil-github, 2025-05-14:
Unfortunately I was also unable to run SPF6000ES with the new Shine F pro adapters and GROTT :-) I have been running SPF5000ES ( of them) with the Shine F adapters with no problems from years.
Interesting thing is that I observed that the new Shine F Pro adapters reset to stock server every 24h.
And also unfortunately the new 6000ES inverters do not work with the old Shine F adapters....
I have tested the beta releases and all suggestions but no result.

johanmeijer, 2025-05-14:
Ok. Maybe the Shine F-pro uses an other protocol (or encryption type?).  
Can you send the grott log (with minrecl=1 specified in the grott.ini)?

vasil-github, 2025-05-15:
When I have the opportunity I will do that. 
Interesting is that I noted when setting up the wifi dongle F pro is that in its stock settings server is set as mqtt.growatt.com.
In the old dongle F series (not pro) stock server is server.growatt.com

hristouzunovaz, 2025-05-15:
I was also not able to make it work so I decided to throw some hardware onto the inverter.
I have bought [Elfin EW11A](https://www.aliexpress.com/item/4001152539658.html?src=google&albch=search&acnt=479-062-3723&isdl=y&aff_short_key=UneMJZVf&albcp=266121556&albag=7593673036&slnk=&trgt=dsa-42862830006&plac=&crea=426823514858&albad=426823514858&netw=g&device=c&mtctp=&memo1=&albbt=Google_7_search&aff_platform=google&albagn=888888&isSmbActive=false&isSmbAutoCall=false&needSmbHouyi=false&gad_source=1&gad_campaignid=266121556&gbraid=0AAAAAD2chG71VtzcUcoNIBr3qkqHSzPsh&gclid=Cj0KCQjwoZbBBhDCARIsAOqMEZWZc12iHrnujLQoBF_KFmhQh5-ZqCzeMkp00XypsNsDiIyDD1Jzxh8aAlHkEALw_wcB&gatewayAdapt=4itemAdapt) RS485 to wifi adapter and created a modbus integration in HA.
Works like a charm. 
Let me know if any of you guys is interested in this setup.

vasil-github, 2025-05-15:
I'm definitely interested. 
Please share. I have to connect 3 6000es inverters in remote location. By remote I mean internet access (but 1h drive away), but the mqtt and HA servers are at home (with real static IPs so reachable directly)

johanmeijer, 2025-05-21:
@vasil-github I think the F_pro uses a different protocol (based on MQTT) that is also used in other newer growatt like NEO 800M-X . 

Maybe you can look at https://github.com/robertzaage/GroBro as described in #552. 

If this is the new way Growatt is going to communicate I will see if I can addopt this grott in the future.