johanmeijer/grott #605: MID 20KTL3-XL

zjonesz, 2024-10-28:
Hi :D

I'm using Docker ( ledidobe/grott:beta )

[Config] - Version 2.8.3
verbose = True
#minrecl = 100
mode = proxy
time = auto 
sendbuf = True 
compat = False
offset = 6
valueoffset = 6
invtype = tl3
include_all = True
decrypt = True

I only get data from 3 of the 4 mppt's

I'm missing data from the last one.

If i change the inverter type from tl3 to min i get data from all 4, but the "pvenergytotal" is showing "655372"

Can somebody help? :)

johanmeijer, 2024-10-28:
I think you are the first one with tl3 with  4 mppt's. The max grott now exposes is 3. I have to change the layout for 4 mppt's. Give me some time to do this and create a new docker container. 

If you are able to add (copy) a file in your docker container I can create a temporary layout file.

zjonesz, 2024-10-29:
Ahh, Thank you :D

I can add the file to the container if you manage to create one.

johanmeijer, 2024-10-29:
If you can add the TL3 layout file from the example directory: 
  
https://github.com/johanmeijer/grott/blob/Master/examples/Record%20Layout/t06NNNNXTL3.json
   
to the container /app directory (all grott files should be in here), and restart the container it should show all 4 mppt info.

zjonesz, 2024-10-30:
The example only had 3 mppt, but i added number 4 from the example of "T06NNNNXMIN.json" it looks like everything is working as i should now.

Thanks for the help 😊

johanmeijer, 2024-10-30:
Oj sorry. I pointed you to the master example directory. I changed the file in the beta branche example.