johanmeijer/grott #607: Growatt  Inverter MIN 2500TL-XE

LeoVe, 2024-11-16:
Hi,

Thanks for the great work !
Hope anyone can help me to get it to work.

I am working on connecting a Growatt Inverter type MIN 2500TL-XE to OpenHAB.

The Grott application is installed on a Synology NAS (Docker).
OpenHAB is installed on a Raspberry Pi.

Lines from the Grott log:
- Inverter serial not recognised - using inverter type default
- Growatt new layout processing
   - offset        :  6
   - record layout :  T06NNNNX

The found and decrypted data looks good.

**Q1:** Is the record layout OK or should a better matching one be chosen?

The grottext needed for the OpenHAB binding logs the following:
  - Grott extension processing started :  grottext
  - Grott import extension failed: grottext

The url defined in Grott ini is tested with a browser and via the linux wget in the Grott docker machine, both return:
  Growatt Binding Servlet
  Status: Bridge Online

**Q2:** What could be the problem with the extension?

</div>

LeoVe, 2024-11-18:
Hi,

Found a grottext.py in the grott repo in grott/examples/Extensions, is this the right extension to use with the OpenHab Grott binding?

I copied the file to the grott docker and it is processed by Grott,  returncode 0!

This looks good, but, in OpenHab only the Grott Bridge is Active, the inverter thing has Communication Error.

The thing type is sph, is this correct?
What should be done?

Thanks for your feedback.

LeoVe, 2024-11-18:
Had to fix the url in grottext.py

from:
   url = f"http://{conf.extvar['ip']}:{conf.extvar['port']}"
to:
   url = f"http://{conf.extvar['ip']}:{conf.extvar['port']}/growatt"

Now it runs.

Maybe a suggestion to add the correct grottext.py to the Grott docker.

Thanks !

johanmeijer, 2024-11-24:
I will have a look at it. Maybe I should create a special OpenHab extension?

LeoVe, 2024-12-02:
Hi Johan,

Great work man, thanks for it !

In fact I used ledidobe/grott:beta image and that one did not contain the grottext.py script.
I found one and used it but in that one the before-mentioned error had to be corrected.
I think it will be OK if the corrected grottext.py will be included in de grott image. As far as I understand a special one for OpenHab is nice but does seem necessary.

Thanks

johanmeijer, 2024-12-08:
The grottext.py is provide as an example I did not realized that it can be used in "real life" I will add it to the next docker version (with the modification).