johanmeijer/grott #619: No data received to grott

Stealthbird97, 2025-01-08:
Hello,

So I have tried configuring Grott for my newly installed SPA1~3k TL BL. 

I have configured the shinelink box via the datalogger control section on "server.growatt.com". Additionally I have confirmed the shinelink box is showing the correct IP.
[image]

This is what I get in after running grott and leaving it for a while. There are no connections at all. The only one which appears is when I attempted to visit the grott server in my web browser.

[grottlog.txt](https://github.com/user-attachments/files/18351150/grottlog.txt)

I note that despite the change of server ip in the shinelink datalogger, the growatt servers still get data.

Any ideas?

FlexxFR, 2025-01-11:
As growatt cloud is still receiving message's, it's either:
1. Still the old network settings are used (data is not going through grott)
2. Data is going through grott but not used, only proxied. 

As with 2 you would expect logs, I assume it's 1, double chekc shinelan config, save, maybe even restart / uplug and reconfirm the settings again, My changed settingsL
- relovddomain = off
- Server IP = grott IP
- Data transfer interval = 1 (optional)

I've read sotry of Grotwatt updates that resets these IP's or somehome there are sometime reset but for me it;s working for a while now.

Stealthbird97, 2025-01-11:
Hey @FlexxFR, thanks for taking the time to reply.

So, very strangely, I changed the IP on the datalogger to my personal computer last night while trying to get it working on a different computer. It wasn't working when I left it and went to bed. I wake up this morning, and find that the in the shine phone app the system is showing offline with the last data being a few hours after I went to bed. 

So I boot up my PC, go to the datalogger ip in my browser and it's still running with the IP of my PC. 
Quickly start up grott, and low and behold, I get a data packet. 
Swap the IP over to the actual server I want to run grott on, and it also works.

Very strange, but it appears to be working now.