johanmeijer/grott #620: Grott MQTT messages crashes Home Assistant

FlexxFR, 2025-01-11:
Happened more than a couple of times and I traced it back to grott mqtt messages.

What happens:
Works for days and than suddenly HA starts acting slowly, CPU load rises, eventually restart automatically
Following in the supervisor log:

```
2025-01-10 18:54:36.855 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'mqtt_user_grott'
2025-01-10 18:54:46.573 ERROR (MainThread) [supervisor.homeassistant.api] Timeout on call http://172.30.32.1:8123/api/core/state.
2025-01-10 18:55:07.575 ERROR (MainThread) [supervisor.homeassistant.api] Timeout on call http://172.30.32.1:8123/api/core/state.
2025-01-10 18:55:07.575 INFO (MainThread) [supervisor.auth] Home Assistant not running, checking cache
2025-01-10 18:55:09.570 ERROR (MainThread) [supervisor.homeassistant.api] Timeout on call http://172.30.32.1:8123/api/core/state.
2025-01-10 18:55:09.570 ERROR (MainThread) [supervisor.misc.tasks] Watchdog missed 2 Home Assistant Core API responses in a row. Restarting Home Assistant Core!
2025-01-10 18:55:09.576 INFO (SyncWorker_2) [supervisor.docker.manager] Restarting homeassistant
```

Checking MQTT logs:
```

2025-01-10 18:51:54: New connection from 192.168.2.210:33471 on port 1883.
2025-01-10 18:51:54: Bad socket read/write on client automatic: Unknown error.
time="2025-01-10T18:51:59+01:00" level=error msg="POST error: Post \"http://127.0.0.1:80/authentication\": context deadline exceeded (Client.Timeout exceeded while awaiting headers)"
time="2025-01-10T18:51:59+01:00" level=error msg="Post \"http://127.0.0.1:80/authentication\": context deadline exceeded (Client.Timeout exceeded while awaiting headers)"
2025-01-10 18:52:00: New connection from 192.168.2.210:34991 on port 1883.
time="2025-01-10T18:52:05+01:00" level=error msg="POST error: Post \"http://127.0.0.1:80/authentication\": context deadline exceeded (Client.Timeout exceeded while awaiting headers)"
time="2025-01-10T18:52:05+01:00" level=error msg="Post \"http://127.0.0.1:80/authentication\": context deadline exceeded (Client.Timeout exceeded while awaiting headers)"
2025-01-10 18:52:00: Bad socket read/write on client automatic: Unknown error.
2025-01-10 18:52:07: New connection from 192.168.2.210:54191 on port 1883.
```

I'm trying to catch this in a grott logfile and network packet captures but so far only normal looking stuff.

@johanmeijer, any idea. Could this be cause by malformed/incomplete mqtt packets from grott?

johanmeijer, 2025-01-18:
@flexxfr. Have you change the update interval? I can imagine (just thought about it) that having a short update interval can push MQTT handling. Grott is now sending every message in a seperate MQTT dialogue, starting a new connection every time. 

With a short update interval it might be a better idea to create a connection with the MQTT server once and leave this "open". 

I have to see how to implement this. Especially the error handling (e.g. what is a connection is lost) and that Grott 3.0 has a separate thread per connection (inverter/datalogger) needs some additional attention. 

If you changed the update interval maybe you can test if changing this would help?

FlexxFR, 2025-01-19:
I did notice the same behaviour compared to other MQTT sources, grott is creating (nd closing) a new connection every time.
Not sure which update interval you mean, I assume the interval in the ShineLink gateway. Yes, changed that to 1 minute. Worked fine for weeks btw.

However, I'm not sure if this would  be an issue. MQTT cpu load/memory is really low. At the same time, my problems were solved when I disabled grott and another (even more consuming MQTT integration), but after I re-enabled them both (also grott) the problem did NOT re-appear?? In the end, I'm not sure anymore if grott or MQTT in general was the cause of these strange problems.

Network connectivity issues between grott and HA (with mqtt broker add-on) are unlikely as the run on the same proxmox server, they communicate directly.

I did some packet captures (from the grott proxmox instance) but I could not catch any suspicious ones. In one trace I did notice multiple MQTT connects, about 10 sec appart, without a MQTT publish. I'm not sure but this could be normal retry behaviour (because my system was is a non-responding/slow state and having "the issue"). However, at the same time this might contribute to overload MQTT even more. (sending more and more retries, like a DoS). This  trace was not during an actual crash and the system recovered itself quickly. But what would happen if the second interval started? Would the first missed one retry forever, and the second also create it's own retry queue? (and-so-on). Just guessing.

In short, the issue "solved" itself. Grott was also working fine for weeks before the issues started. And after this investigation it might not be the root cause but just the first to alarm these issues. Maybe its a catalyst to make the other issues worse, I don;t know. 

Maybe an open connection could be useful to implement, maybe a retry limiter? 
For now, no pressing issues. I'll keep an eye open. Thanks

johanmeijer, 2025-01-20:
I will put an open MQTT connection on the wish list, but have to look at some other challenges first ;)