johanmeijer/grott #623: Very sporadic data from my inverter.

danielvandenberg95, 2025-01-15:
I think I'll need some help properly getting the data out of my inverter. I only seem to be receiving data every so often, and can't quite get a handle on what is wrong. This already was the case with the app, where I would get data roughly every hour and a half apart (which messed with my HomeAssistant dashboard quite a bit). After moving to grott, the issue unfortunately was not magically resolved.

I've attached some logs from my docker container: 
[_solarpanels_grott.1.ifbt63wlyh35y8u4funr9jzx4_logs.txt](https://github.com/user-attachments/files/18420115/_solarpanels_grott.1.ifbt63wlyh35y8u4funr9jzx4_logs.txt)

This morning I switched to alpha,, my docker stack now looks like this:
```

services:
  grott:
    image: ledidobe/grott
    ports:
      - "5279:5279"
    environment:
      gmqttip: "192.168.0.67"
      ggrowattip: "192.168.0.67"
      ggrowattport: "5781"
    command: >
      /bin/sh -c "
      apt update &&
      apt install -y git &&
      cd / && rm -rf app && git clone https://github.com/johanmeijer/grott.git /app && cd app &&
      git checkout Alpha &&
      cp examples/Home*Assistent/grott_ha.py ./ &&
      echo [extension] > grott.ini &&
      echo extension = True >> grott.ini &&
      echo extname = grott_ha >> grott.ini &&
      echo extvar = {\\\"ha_mqtt_host\\\": \\\"192.168.0.67\\\", \\\"ha_mqtt_port\\\": \\\"1883\\\"} >> grott.ini &&
      python3 -m pip install psutil &&
      export LOGLEVEL=DEBUGV && python3 grott.py -l debugv"
    deploy:
      placement:
        constraints:
          - node.hostname==truenas-docker
  grottserver:
   image: ledidobe/grottserver
   ports:
    - "5781:5781"
    - "5782:5782"
   command: >
      /bin/sh -c "
      apt update &&
      apt install -y git &&
      cd / && rm -rf app && git clone https://github.com/johanmeijer/grott.git /app && cd app &&
      git checkout Alpha &&
      python3 -m pip install psutil &&
      export LOGLEVEL=DEBUGV && python3 grottserver.py -l debugv"
   deploy:
    placement:
     constraints:
     - node.hostname==truenas-docker
```

I am getting a lot of "pings"
```
	\x00\x05\x00\x06\x00\x20\x01\x16\x0c\x25\x24\x46\x22\x37\x22\x76\x26\x59\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x
	47\x72\x6f\x77\x61\x74\x74\x47\x72\x7b\xb9 
2025-01-15 06:33:41,891 - grottproxy - 	DEBUG: Data less then minimum record length, data not processed
2025-01-15 06:33:41,890 - grottproxy - 	DEBUG: Data record validation started
2025-01-15 06:33:41,890 - grottproxy - 	DEBUG:  - <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.168', 5279), raddr=('10.0.0.5', 1029)>
2025-01-15 06:33:41,890 - grottproxy - 	DEBUG: Growatt packet received:
	47\x72\x6f\x77\x61\x74\x74\x47\x72\x1b\x82 
	\x00\x07\x00\x06\x00\x20\x01\x16\x0c\x25\x24\x46\x22\x37\x22\x76\x26\x59\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x
2025-01-15 06:33:41,889 - grottproxy - 	DEBUGV: Received buffer:
```

[image]

(order got messed up by docker....)

I'll be digging into this deeper, but maybe someone here could point me in the right direction?

[image]
[image]
[image]
[image]

dberrocal-git, 2025-01-15:
Hello,

I have a similar problem, a user advises to update both dataloger and inverter.
https://github.com/johanmeijer/grott/discussions/621

I still don't have the installer code to register the device, I'm waiting for it, so I haven't been able to test it.

johanmeijer, 2025-01-20:
Please see also my comment (thoughts) on #621. 

_Is it possible that when you shorten the update interval to much the datalogger (RF stick/ShineWifi) can not keep up with it and will not function anymore? It is designed the send a datarecord every 5 minutes. With Grottserver it is maybe possible to go above (or below) the capabilities of the datalogger._

J4CE2, 2025-01-22:
Did it just start this week? Last 2-3 days I've noticed some issues and I'm sure it is server outages with growatt, as it randomly will come back after a few hours.

dberrocal-git, 2025-01-24:
@danielvandenberg95 Check #621 
 
The Shinelink X RF Box and RF Stick fimmware update solved the problem for me.

danielvandenberg95, 2025-01-26:
> Please see also my comment (thoughts) on [#621](https://github.com/johanmeijer/grott/discussions/621).
>
> _Is it possible that when you shorten the update interval to much the datalogger (RF [...]

I have it set to 5 minutes, so not really no. 

> Did it just start this week? Last 2-3 days I've noticed some issues and I'm sure it is server outages with growatt, as it randomly will come back after a few hours.

Nope, has been going on for a couple of years. I just never took the time to dive into it yet.

> [@danielvandenberg95](https://github.com/danielvandenberg95) Check [#621](https://github.com/johanmeijer/grott/discussions/621)
> 
> The Shinelink X RF Box and RF Stick fimmware update solved the problem for me.

That's awesome, I asked you some questions there regarding how you got the firmware updates!