johanmeijer/grott #626: MID 30KTL3-XH support

ReviloEgros, 2025-01-30:
I would love to have support for the MID 30KTL3-XH, or any hint/help on how to get it running properly. 
Here is the output from grott, i am using tl3 as inverter type because its in his name.

`Jan 30 23:35:43 nuc8i3beh grott[2189]:          - Growatt packet received:
Jan 30 23:35:43 nuc8i3beh grott[2189]:                   <socket.socket fd=4, family=2, type=1, proto=0, laddr=('192.168.188.200', 36356), raddr=('47.254.130.145', 5279)>
Jan 30 23:35:43 nuc8i3beh grott[2189]:          - Growatt original Data:
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x00\x2c\x00\x06\x03\x3f\x01\x04\x00\x22\x28\x47\x22\x38\x20\x74\x45\x58\x77
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x3d\x3c\x2c\x44\x30\x70\x3c\x5f\x46\x55\x74\x74\x47\x72\x6f\x77\x61\x74\x74
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x74\x6a
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \xcc\x78\x73\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x7c\xf0\x69\xbd
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x74\x47\x72\x6f\x77\x61\x7c\xb7\x47\x72\x6f\x77\x61\x74\x7c\x8a\x72\x6f\x77
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x61\x74\x74\x48\x59\x60\x40\x6e\x36\x74\x47\x66\x94\x77\x61\x74\x74\x47\x72
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x7b\x8c\x61\xad\x99\x27\x72\x6f\x77\x3e\x74\x75\x56\x60\x6f\x76\x71\x34\x74
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x47\x72\x55\x77\x61\xd6\x9a\x47\x72\x6f\x51\x61\x74\x19\x15\x72\x6f\x77\x61
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x74\x74\x47\x72\x6f\x77\x61\x15\x74\x47\x55\x72\x77\x61\x74\x6e\x47\x72\xa3
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \xb8\x61\x74\x74\xe0\x72\x6f\x1d\xef\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x72\x0f\x77\x61\x74\x74\x50\x88\x6f\x77\x61\x74\x74\x47\x72\x6f\x73\xe3\x75
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x43\x46\xe6\x6e\x40\x61\x74\x74\x8a\x74\x2b\x71\x20\x3a\x54\x47\x72\x6f\x73
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \xf2\x94\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x6f\x77\x61\x74\x74\x47\x4e\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x47\x72\x6f\x77\x61\x17\x78\x72\x7e\xde\x77\x61\x74\x74\x47\x72\x6f\x77\x61
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x46\x66\xbe
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x77\x61\x74\x3d\x47\x72\x28\x9e\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x74\x47\x72\x6f\x77\x61\x53\x64\x47\x72\x78\xfd\x61\x74\x74\x47\x72\x6f\x77
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x61\x74\x74\x47\x72\x6f\x77\x61\x71\x60\x47\x78\x6f\x76\xe3\x96\x74\x47\x72
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x6f\x73\x61\x51\x70\x05\xd9\x6f\x77\x61\x74\x74\x47\x7e\xdd\x7a\x4f\x74\x74
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x72\x6f\x77\x61\x74\x74\x44\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x61\x2c\x2c\x1f\x2a\x6f\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x39\x2c\x2c\x1f\x2a
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x37\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x39\x2c\x74
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x74\x74\x52\xe7\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x61\xe1\x47\x72\x6f\x77\x61\x74\x74\x47
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  \x6f\xcc\x19
Jan 30 23:35:43 nuc8i3beh grott[2189]:          - Grott automatic protocol detection
Jan 30 23:35:43 nuc8i3beh grott[2189]:          - Grott data record length 839
Jan 30 23:35:43 nuc8i3beh grott[2189]:          - layout   :  T060104XTL3
Jan 30 23:35:43 nuc8i3beh grott[2189]:          - no matching record layout found, try generic
Jan 30 23:35:43 nuc8i3beh grott[2189]:          - Record layout used :  T06NNNNXTL3
Jan 30 23:35:43 nuc8i3beh grott[2189]:          - Growatt data decrypted V2
Jan 30 23:35:43 nuc8i3beh grott[2189]:          - Grott Growatt data decrypted
Jan 30 23:35:43 nuc8i3beh grott[2189]:          - Growatt plain data:
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  002c0006033f010447504730434c5433373700000000000000000000000000000000000000005
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  24b4d3044374e3031340000000000000000000000000000000000000000000000000000030bb8
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  0c340000000000000000000000000000000000000000000000000000000000000000000000000
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  000000000000000000000000000138708c900000000000008c300000000000008cd0000000000
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  000f2b0f370f42000014fb00000000000014fb00d9ed600000005f00011112000110400000003
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  a0000a2ee0000002600006d520000000000000000000000610000271d0000001a0000cccf0000
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  00a700006a8e0000000000000000000000600000000017fa00000000000000000482013701940
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  137000000cd064406414e200000000493e0000000000000000000000000000000000000000000
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  00003c0000000000000000000000000000000000630c350cb1000000000000000000000000000
Jan 30 23:35:43 nuc8i3beh grott[2189]:                  000000000000000000000000114d100000049000047e900000000000000000000000000000000
[... cut ...]
Jan 30 23:35:43 nuc8i3beh grott[2189]:          - Grott MQTT topic used : energy/growatt
Jan 30 23:35:43 nuc8i3beh grott[2189]:          - Grott MQTT message retain enabled
Jan 30 23:35:43 nuc8i3beh grott[2189]:          - MQTT message message sent
Jan 30 23:35:43 nuc8i3beh grott[2189]:          - Grott Send data to PVOutput disabled
Jan 30 23:35:43 nuc8i3beh grott[2189]:          - Grott Send data to Influx disabled
Jan 30 23:35:43 nuc8i3beh grott[2189]:          - Grott extension processing disabled
`

johanmeijer, 2025-01-31:
This should be the right invtype for mod (tl3). What is the problem?

You can also try it with grott v3 (alpha branche) and then invtype = auto. This will perform an auto detection and hopefully that will match better.

johanmeijer, 2025-01-31:
O sorry I see he wrong values. I will test it and seevwhat is wrong. Please if possible,  try grott v3. I think that should show the right values.