johanmeijer/grott #630: Imcorrect temperature when below 0deg C

ridax67, 2025-02-16:
Inverter MOD 10KTL3-XH(BP)

Using pvtemp tweak to send inverter temp1 with grott.  When this temperature is below zero like -4.1 HA (through mqtt) reports a high value, like 6549.5 and pvoutput site report no data.  As 65495+41 = 65536 my conclusion is that the growatt logger sends this temperature value as a SHORT, while grott reads it as a USHORT.

[image]

[image]

[image]

[image]

johanmeijer, 2025-02-16:
Grott uses the Inside Inverter temperature to sent PVOutput. The change that this is negative seems very low to me. 
How do you modify this temperature so Grott send the outside temperature? 

I use nodered to sent outside temperature to PVOutput.

johanmeijer, 2025-02-16:
Be aware Grott does not sent any temperature information to PVOutput by default. You have to enable this by pvtemp = True in the grott.ini.

ridax67, 2025-02-16:
I know grott doesn't send temp as standard to pvoutput which is why I specified that I used the pvtemp tweak.  As you can see from the pvoutput it does send temperatures when it is positive, but my guess is it sends a very high temp when negative and pvoutput is filtering abnormal values.

mqtt receives values like 6549.5 degrees.

As you can see from the Growatt app shinephone, it IS measuring minus temperatures and as the inverter is placed outside in winter climate it is not strange at all that it is reporting minus temp when waking up in the morning.

As the reported incorrect temperature fits perfectly with 65536/16bit pattern, it is quite clear the temp values should be interpreted as 16 bit signed ints and then divided by 10.

ridax67, 2025-02-16:
So if you get the values in 4 digit hex values into a ushort you should just test if it is > 32767 and if that's the case subtract 65536 from it and then divide by 10, if you can't read it into the correct data type directly.

johanmeijer, 2025-02-17:
Maybe stupid question but is the pvtemp tweak?

ridax67, 2025-02-17:
I mean using pvtemp = True in config to send the inverter temp as the outside temperature to pvoutput.

I am sure you are right about numx.  I didn't check what datatype it was, it seemed to be for 32 bit values, as it was the length I saw in the tables, but I didn't investigate the decoding of it.

I have tested my mod and the inverter definitely sends negative temperatures if measured and this explains why my cooling fan started in the winter... :-)

dberrocal-git, 2025-04-18:
Hi, exactly the same thing happens with **bdc1_ibat** and **bms_batterycurr**.

When the battery is discharging, grott uses UNSIGNED SHORT, but it should be SHORT, as the current can be negative (discharging) or positive (charging).

Alternatively, in MQTT I have added logic to subtract USHRT_MAX (65535) if the value exceeds 60000.

But the right thing to do would be to change the record layout to **snum**.

johanmeijer, 2025-04-23:
@dberrocal-git which version of Grott and which layout (invtype) do you use? bms_batterycurr should be able to be negative, 

dbc1_ibat is indeed the old value (num i.s.o numx). In version 3.0 (alpha, everything should work fine). Be aware in 3.0.0 some naming changes has been implemented. 

@ridax does numx not work for pvtemperature (is alos changed to numx in 3.0.0) ? Question: when is the inside temp of the inverter negative?

ridax67, 2025-04-23:
I never tried using numx as I had already solved it by implementing my own snum implementation.

I don't understand why you feel negative temperatures inside inverter temperatures are s

dberrocal-git, 2025-04-30:
@johanmeijer Grott 3.0 alpha, 
[T06NNNNXMIN.json](https://github.com/johanmeijer/grott/blob/Alpha/examples/Record%20Layout/T06NNNNXMIN.json)

It seems that the T06NNNNXMIN does not have the new numx data type in the alpha branch.

johanmeijer, 2025-04-30:
No that is the same layout definition file (but you do not need it). 

In the Alpha branche the layout processing has been changed. If you specify invtype=auto then it will use the new processing method and determine which registers are used in the data file and use the new layout definitions for that (as descriped in the Growatt documentation). 

But it still is possible I have not changed al definitions with numx if that fits better (but we can change that).