johanmeijer/grott #634: SDM630 Modbus Meter Data Not Recognized by Grott

alvarolopes01, 2025-02-28:
The SDM630 Modbus meter data is not being correctly recognized by Grott. The meter is connected to a Growatt MOD 4000TL3-XH inverter, which communicates with the Growatt Shine LAN box. The communication between the meter and the inverter is confirmed to be working because I can see the meter data in the ShinePhone app, and Grott is successfully forwarding data to the Growatt servers. If Grott is stopped, the ShinePhone app does not receive any data.

Despite this, Grott does not seem to properly parse or assign values to the expected variables for the SDM630 meter. There is a package coming from the inverter that is correctly decrypted, and another package that I suspect contains the meter data, but Grott does not assign any values to the corresponding variables. I have tested Grott both running directly in Home Assistant via the add-on and running on a separate PC, and both setups exhibit the same behavior.

Below are my Grott configuration, the log of a correctly decrypted package, and the log of the suspected meter data package. I would be grateful if someone can help me with this :)

[config.txt](https://github.com/user-attachments/files/19033872/config.txt)
[Undecrypted packages.txt](https://github.com/user-attachments/files/19033874/Undecrypted.packages.txt)
[Decrypted package.txt](https://github.com/user-attachments/files/19033873/Decrypted.package.txt)

johanmeijer, 2025-03-02:
Your right. Your smart-meter is not detected by Grott while it has a different device number then I defined in the code. I will make Grott detection for smart meters independent from device numbers.  

For now I added T065120.json layout file in example directory (https://github.com/johanmeijer/grott/tree/Beta-(2.8.x)/examples/Record%20Layout) 

I you copy this one in the Grott home directory and stop/start Grott it should work.

alvarolopes01, 2025-03-02:
Thanks for the fast response! It works great! Just a note for those who have a 3 phases system (my case): don't forget to edit the JSON in the fields related to the 2nd and 3rd phases excluding the ("incl" : "no") at the end. Great repository!

closed 2025-03-02 by alvarolopes01

johanmeijer, 2025-03-02:
Great that it works!

You can also specify includeall = True in the grott.ini. to show everything.