johanmeijer/grott #635: Data seems to still be mirrored to server.growatt.com, despite running grott server.

danielvandenberg95, 2025-03-02:
As the title says, I'm running grottserver (and grotto, both in docker). It's working great, and I'm loving this project.

However, somehow I'm still seeing my data in server.growatt.com. I was expecting to not have any connections to external servers any more. The image with grottserver shows no connections to the cloud.

What did I misunderstand?

I'm using the following docker config:

```yaml

services:
  grott:
    image: ledidobe/grott
    ports:
      - "5279:5279"
    environment:
      gmqttip: "192.168.0.67"
      ggrowattip: "192.168.0.67"
      ggrowattport: "5781"
    command: >
      /bin/sh -c "
      apt update &&
      apt install -y git &&
      cd / && rm -rf app && git clone https://github.com/johanmeijer/grott.git /app && cd app &&
      git checkout Alpha &&
      cp examples/Home*Assistent/grott_ha.py ./ &&
      echo [extension] > grott.ini &&
      echo extension = True >> grott.ini &&
      echo extname = grott_ha >> grott.ini &&
      echo extvar = {\\\"ha_mqtt_host\\\": \\\"192.168.0.67\\\", \\\"ha_mqtt_port\\\": \\\"1883\\\"} >> grott.ini &&
      python3 -m pip install psutil &&
      export LOGLEVEL=DEBUGV && python3 grott.py -l debugv"
    deploy:
      placement:
        constraints:
          - node.hostname==truenas-docker
  grottserver:
   image: ledidobe/grottserver
   ports:
    - "5781:5781"
    - "5782:5782"
   command: >
      /bin/sh -c "
      apt update &&
      apt install -y git &&
      cd / && rm -rf app && git clone https://github.com/danielvandenberg95/grott.git /app && cd app &&
      git checkout Alpha &&
      python3 -m pip install psutil &&
      export LOGLEVEL=DEBUGV && python3 grottserver.py -l debugv"
   deploy:
    placement:
     constraints:
     - node.hostname==truenas-docker
```

johanmeijer, 2025-03-02:
Do you see that grott is communicating with your grottserver (e.g. in the grottserver docker log)? 

At the first glance I do not see something wrong.

danielvandenberg95, 2025-03-02:
Sorry for wasting your time, user error here! The ShinePhone app is so terrible that I thought I saw data, while actually I was looking at 10 days ago... Very glad to have a self-hosted option!

closed 2025-03-02 by danielvandenberg95