johanmeijer/grott #636: T06221b Record Layout not processed

BarmanMaz, 2025-03-03:
Hello,
Im running the add-on Grott Beta branch (2.8) Ver: 0.1.7

I have full dataa from the inverters ( the record layeout used is T06NNNNXSPH) so all good, my problem is that i can't have the data from the SDM630 meter so not able to have Import and Export values.

Can you plase give me an help?

The following growatt data probably belong to the SDM630 meter 

Growatt original Data:
		 \x00\x01\x00\x06\x01\x1e\x22\x1b\x03\x2b\x2b\x47\x25\x36\x40\x77\x41\x3c\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x34\x78\x77\x77\x4d\x5f\x72\x77\x95\x42\x4d\x73\x44\x57\x59\x57\x40\x58
		 \x75\x43\x5b\x43\x55\x5a\x45\x76\x5e\x42\x45\x52\x42\x5a\x71\x5e\x5c\x44\x53
		 \x5a\x45\x6b\x5f\x5e\x46\x56\x5a\x40\x6b\x41\x5f\x40\x4f\x43\x58\x75\x45\x58
		 \x59\x52\x58\x46\x76\x46\x41\x4e\x4d\x47\x4c\x75\x5c\x57\x5b\x55\x46\x46\x69
		 \x42\x43\x45\x55\x44\x5a\x77\x5e\x42\x47\x4f\x42\x58\x77\x5c\x58\x5b\x4c\x44
		 \x5a\x73\x5e\x5d\x45\x59\x5a\x4d\x6b\x40\x5d\x41\x4f\x4d\x58\x75\x40\x58\x59
		 \x52\x58\x46\x69\x44\x43\x45\x4f\x4d\x58\x75\x5c\x5e\x5b\x51\x5a\x44\x6b\x45
		 \x56\x4e\x4f\x4d\x58\x70\x4b\x56\x59\x58\x58\x59\x77\x5c\x5f\x5b\x55\x4d\x5a
		 \x7e\x5e\x5a\x45\x55\x42\x45\x69\x41\x43\x46\x56\x44\x44\x70\x5c\x5d\x5b\x50
		 \x44\x44\x73\x45\x41\x4e\x4d\x45\x45\x74\x4b\x59\x59\x50\x58\x43\x75\x45\x5f
		 \x44\x4f\x45\x58\x7f\x4a\x5f\x41\x4f\x4c\x58\x71\x46\x5c\x40\x4f\x44\x58\x71
		 \x40\x5f\x47\x4f\x46\x58\x77\x5c\x5f\x5b\x51\x5a\x44\x6b\x42\x41\x47\x4d\x4d
		 \x44\x7e\x43\x5d\x59\x56\x41\x58\x74\x4b\x5b\x59\x57\x58\x47\x7e\x46\x41\x47
		 \x4d\x47\x4d\x73\x5c\x57\x5b\xca\x45
	 - Grott automatic protocol detection
	 - Grott data record length 294
	 - layout   :  T06221b
	 - no matching record layout found, try generic
	 - Record layout used :  T06221b
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00010006011e221b4459443044423430335300000000000000000000000000000000000000000
		 0431903030a2d1d00f436393436382e36342c32313434342e31312c2d3233362e362c3333322e
		 312c2d3131372e342c3330372e372c3237372e332c3231342e392c3338322e382c3432322e302
		 c3234302e302c2d302e362c302e372c2d302e342c3232382e392c3232362e392c3232372e332c
		 322e362c322e392c322e312c302e302c3739392e392c3739392e392c2d302e302c34392e392c3
		 5323436312e332c31373030372e322c31303034372e392c31313339362e312c37323730332e31
		 2c383830362e382c363433372e302c363230302e322c302e302c302e302c302e302c393039313
		 22e37352c3339342e362c3339342e302c3339342e382cab31
	 - Grott data ack record or data record not defined no processing done

Happy to supply other datas if necessary

Many Thanks

johanmeijer, 2025-03-03:
If you copy the t06221b.json lay-out file from the examples directory ( https://github.com/johanmeijer/grott/tree/master/examples  ) into the grott main directory and stop/start grott it should work.

BarmanMaz, 2025-03-04:
Hello,
I mange to make it working thanks a lot.

Can you please give me another advise, I have changed to smart meter going to  CHINT DTU666, I wanted to get rid of the " clamps".
I connected it to the shinemaster, all fine, all datas are sent to growattwebsite, but i can't read it into GROTT.
Following the datas:
- Growatt original Data:
                 \x00\x01\x00\x06\x00\xb4\x21\x1b\x03\x2b\x2b\x47\x25\x36\x40\x77\x41\x3c\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x30\x78\x77\x70\x52\x7d\x43\x77\xeb\x47\x4d\x71\x5c\x58\x5b\x52\x4d\x41
                 \x69\x4a\x43\x44\x58\x43\x5a\x76\x5e\x5d\x45\x58\x5a\x4c\x6b\x40\x5d\x4f\x4f
                 \x47\x58\x75\x40\x57\x59\x59\x58\x45\x69\x47\x43\x46\x4f\x42\x58\x76\x5c\x58
                 \x5b\x50\x45\x5a\x7f\x5e\x42\x41\x53\x5a\x4d\x6b\x40\x56\x44\x4f\x4c\x58\x6a
                 \x40\x5e\x4e\x4f\x44\x58\x6a\x40\x58\x4f\x4f\x41\x58\x6a\x43\x5c\x40\x4f\x45
                 \x58\x76\x43\x57\x59\x58\x58\x59\x75\x44\x5f\x59\x52\x58\x44\x69\x42\x5f\x4e
                 \x4d\x59\x44\x69\x43\x58\x41\x4d\x44\x5a\x70\x47\x56\x5b\x4c\x44\x5a\x72\x41
                 \x5c\x5b\x55\x4d\x5a\x77\x5e\x5f\x59\x51\x58\x44\x69\x42\x43\x31\x4a
         - Grott automatic protocol detection
         - Grott data record length 188
         - layout   :  T06211b
         - no matching record layout found, try generic
         - Record layout used :  T06211b
         - Growatt data decrypted V2
         - Grott Growatt data decrypted
         - Growatt plain data:
                 0001000600b4211b4459443044423430335300000000000000000000000000000000000000000
                 047190304150f2c008a3339362e372c3339352e382c3339372e312c3232392e382c3232382e33
                 2c3232382e382c312e352c312e362c312e372c31312e382c2d36322e392c3239332e382c2d323
                 1392e302c2d3237382e352c2d3133372e312c3131382e392c2d3236302e332c302e3030392c2d
                 302e3137362c302e3735392c2d302e3533332c34392e302c302e302c302e302c462b
         - Grott data ack record or data record not defined no processing done.
As you told me t06211b is for SDM630, I have put in the main grott directory your latest CHINT record type but I still have the same result....always use T06211b....what I'm doing wrong,
Thanks in adavance

johanmeijer, 2025-03-04:
We have a channels here. The chint gives the same record type but a different layout. I am not sure how to detect the difference if I want to enable automatic detection for this. 

For now you can use the : tchntthree.json from the layout example library.

BarmanMaz, 2025-03-05:
Thanks A lot..now its running

paralichko, 2025-03-12:
Hello,

how can i force which layout to be used ? I am using grott, as a HA add-on.

johanmeijer, 2025-03-19:
You have to specify invtype = parameter (for a container that is specify the ginvtype= environmental varaiable). I am not sure how to specify this in HA. 

Maybe @egguy can help with this.