johanmeijer/grott #637: Grott keeps disconnecting from growatt servers

molekuul, 2025-03-15:
For a gew days now,  grott keeps disconnecting from the growatt servers, I have to restart grott container regularly

johanmeijer, 2025-03-19:
Which version of Grott do you use? Where are you located? I might be a Growatt server IP definition change. 

If you use the wrong Growatt IP address Growatt try to change this. 

It might also be a faulty datalogger (that seems to happens more often).

molekuul, 2025-03-20:
Hi Johan, 
I'm using grott 2.8.3 
I live in the Netherlands. What I can see is that the loadbalancer for server.growatt.com is switching between IP-addresses
In the grot.conf I have the name: server.growatt.com configured, instead of the IP-address connected to the loadbalancer.

% dig  server.growatt.com

; <<>> DiG 9.10.6 <<>> server.growatt.com
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 30061
;; flags: qr rd ra; QUERY: 1, ANSWER: 3, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;server.growatt.com.		IN	A

;; ANSWER SECTION:
server.growatt.com.	60	IN	CNAME	nlb-nbtf7vlspietfgd3ro.eu-central-1.nlb.aliyuncs.com.
nlb-nbtf7vlspietfgd3ro.eu-central-1.nlb.aliyuncs.com. 60 IN A 47.254.130.145
nlb-nbtf7vlspietfgd3ro.eu-central-1.nlb.aliyuncs.com. 60 IN A 8.209.71.240

;; Query time: 77 msec
;; SERVER: 192.168.2.10#53(192.168.2.10)
;; WHEN: Thu Mar 20 07:37:48 CET 2025
;; MSG SIZE  rcvd: 142

As you can see growatt has 2 IP-addresses configured for the same LB name. I think they use Round-Robin DNS, or some zone redundant Loadbalancer. 
I saw that when grott is running for some time, it stops resolving correctly and when the Loadbalancers is down where DNS points to in grott, it stops working. 

It now looks it is stable again.

Kind regards,
Joris Weijters

johanmeijer, 2025-03-21:
Ok. I live in the Netherlands to and did not noticed one problems but I am running Grott V3 in server/passthrough mode and that is very different.  

If the problems come back I will start the same Grott version as you use and see if I have the same problems.

molekuul, 2025-04-02:
Hi Johan,
today it happened again, Grott stopped sending data to the growatt servers at approx 05:05 CET.
The IP was 47.254.130.145 here is the logging when growatt didn't recieve data
`
	 - Growatt packet received:
		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.240.2', 37222), raddr=('47.254.130.145', 5279)>
	 - Growatt original Data:
		 \x04\x0a\x00\x06\x03\x3f\x01\x04\x1f\x35\x2b\x41\x22\x3f\x26\x7e\x28\x5d\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x3f\x2e\x2d\x44\x37\x0f\x4a\x5f\x4f\x2c\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x78\x70\x75\x57\x64\x6f\x74\x6a
		 \xcc\x78\x73\x72\x6e\x77\x61\x74\xfa\x42\xe5\x6f\x77\x61\x74\x74\x47\x77\xfc
		 \x77\x60\x74\x74\x47\xfc\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x6e\xf7\x72\x6f\x65\x4e\x74\x74\x5b\xa5\x7c\xf0\x68\x72
		 \x74\x4a\x72\x6f\x7c\xda\x7c\x81\x47\x7f\x6f\x77\x6a\xd0\x7c\x85\x72\x7e\x77
		 \x61\x7a\x97\x48\x0d\x60\x27\x6e\x27\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x75\xc7\x62\xcd\xae\x92\x72\x6f\x77\x45\x74\x74\x8e\x9f\x6f\x77\xcc\xd9\x74
		 \x47\x72\x6f\x77\x61\x25\x8c\x47\x72\x6f\x77\x61\x74\x2f\xf2\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x57\x74\x47\x20\x85\x77\x61\x74\x76\x47\x72\x4a
		 \x67\x61\x74\x74\x62\x72\x6f\xa6\x38\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\xb8\x88\x6f\x4a\x61\x7b\x74\x00\x72\x6f\x6f\x18\x75
		 \x6f\x46\xe2\x6e\x7c\x61\x74\x75\x0a\x7e\x58\x7b\x5c\x3a\x54\x47\x7c\x6f\x77
		 \xa2\x24\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x4e\x6f\x77\x61\x74\x74\x46\x72\x6f\x77\x61\x74\x74
		 \x5d\xc2\x6f\x77\x61\x55\x78\x72\x7e\xde\x77\x61\x74\x56\x47\x72\x2e\x78\x61
		 \x74\x74\x59\x72\x6f\x3c\x5c\x74\x74\x47\x6f\x6f\x77\x79\xd0\x74\x47\xcd\xca
		 \x77\x61\x74\x6b\x47\x72\xf5\xe7\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x53\x64\x47\x72\x77\x0e\x61\x74\x74\x47\x70\x6e\x77
		 \x61\x74\x74\x15\xc7\x6f\x6a\x61\x37\x6c\x3c\x79\x9e\x77\x71\x74\x64\x46\x73
		 \x6f\x77\x61\x74\x6d\x53\x72\x6f\x77\x61\x74\x74\x06\x7d\x6f\x77\x2a\x49\x74
		 \x47\x72\x6c\x7b\xeb\x74\x71\x47\x4b\x6f\x77\x61\x96\x74\x87\x72\x6c\x77\x71
		 \x74\x74\x47\x72\x6f\x76\x61\x74\x74\x77\x72\x5f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x67\xbd\x72\x6f\x64\x9b\x74\x74\x47\x72\x6f\x75\x61\x74\x74\x47
		 \x72\x2c\x25\x0d\x8a\xbe\x46\x0a\x66\xb3\x68\xb0\x74\x47\x72\x0f\x2f\xa1\x3d
		 \xb4\x47\x72\x6f\x77\x48\xcd\x74\x47\x72\x6f\x7b\x81\x78\xbc\x15\x10\x6f\x77
		 \x61\x76\x74\x47\x72\x4f\x77\x61\x75\x75\x46\x86\x6f\x76\x3d\x9e\x74\x47\x73
		 \x6f\x77\x61\x74\x76\x47\x72\x6f\x77\x61\x74\x74\x47\x7e\xdd\x7a\x4f\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6c\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x16\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x44\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x2c\x2c\x1f\x2a\x6f\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x39\x2c\x2c\x1f\x2a
		 \x37\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x39\x2c\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\xef
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x5e\x4e\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x6d\xbe\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\xa2\xf0
	 - Grott automatic protocol detection
	 - Grott data record length 839
	 - layout   :  T060104XMIN
	 - no matching record layout found, try generic
	 - Record layout used :  T06NNNNXMIN
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 040a0006033f010458474436434b52395a3200000000000000000000000000000000000000005
		 0594c3043483830384d0000000000000000000000000000000000000000190401101600030bb8
		 0c3400010000008e0597000000000000059300010000008e00000000000000000000000000000
		 00000001ab00000122f00001cd713870906000d00000bbb08f5000d00000ba408c2001100000e
		 e30f7f0f500f53000000000000000000001ab003b9dad5000000240000c9ed0000adad0000000
		 0000051f80000000000005bb5000000000000000000000023000052ea00000002000025100000
		 00250000d15900000000000000000000000000000000fffa003d000f004700001879011b01900
		 10b0000014d0c370c3d4e20000e0000c350000000000000000000000000000000000000000000
		 00003c0000000000010000000000001ab0000000210c350cb1000000220000410f0000001e000
		 04b3d0000001d000018a40000bfa50000001f00009a9000000000000000000000000000000000
[... 219 lines cut ...]
	 - Grott Send data to Influx disabled
	 - Grott extension processing disabled

	 - Growatt packet received:
		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.240.2', 5279), raddr=('192.168.240.1', 40286)>
	 - Grott - grottproxy - Invalid data record received, processing stopped for this record

	 - Growatt packet received:
[... cut ...]
		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.240.2', 5279), raddr=('192.168.240.1', 40286)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.240.2', 37222), raddr=('47.254.130.145', 5279)>
	`

When I restarted my grott container, I saw the IP address to be of the grott server: 8.209.71.240
and the growatt servers are recieving data again.

kind regards,

Joris Weijters

johanmeijer, 2025-04-04:
Hi Joris,

I noticed that some times Grott servers are unavailable or unstable. Grott is setting up a session and will keep that session until the Inverter goes offline. It should also create a new connection when the connection has been closed by the endpoint. 

I seems that unstable or intermittent connections are not handled very well at this moment. If a connection is not working but is not totally gone also (at least Grott does not is aware of that) this is not detected, so there is no reason for Grott to create a new session. 

In Grott V3 I also noticed some strange behavior with connection cleanup. I am now trying to debug that. I will also review  the session health checking that is already there to see if we can improve this. 

For now stopping/starting Grott seems to be the best solution.