johanmeijer/grott #640: APX battery info with MIN4200TL-XH via grott into HA

berndh1979, 2025-04-07:
Great project. Use it already for a period. However, last week I upgraded my Solar system with two APX batteries. I use a MIN4200TL-XH inverter and a DataLogger ShineLAN-X. In grott I do get the inverter values, but I'm missing the APX battery records. 

Running the lastest (beta) version via docker. I do see the records coming in, at least the standard inverter values.

Tried tethe ginvtype=min in the docker environments, but no luck there. I do see all the relevant records on the server.growatt.com, so the communication is setup properly. 

Goal is to get all these records/values into MQTT and show them in HA.

Any other suggestions? Missing some configs?

    image: ledidobe/grott:latest
    environment:
      - ginvtype=min
      - gblockcmd=False
      - gnomqtt=False
      - gmqttip=zzzzz
      - gmqttport=1883
      - gmqttauth=True
      - gmqttuser=yyyyy
      - gmqttpassword=yyyyy
      - gmqtttopic=energy/growatt
      - gpvoutput=False
      - ginflux=True
      - ginflux2=False
      - gifdbname=grottdb
      - gifip=zzzzzz
      - gifport=8086
      - TZ=Europe/Amsterdam
      - gtime=server

closed 2025-06-08 by johanmeijer

johanmeijer, 2025-06-15:
Did you looked at this one? 

https://github.com/muppet3000/homeassistant-grott

johanmeijer, 2025-06-15:
Ok. It hard to keep thins in sync especially it  is managed by different persons. I am sorry I am not an active HA users. I have for some initial testing now both the Full integration and the direct MQ with Muppet3000 sometimes active and it does not look bad, but I have not looked in detail at the completeness. 

Maybe this something where AI can help us. I will see if ChatGPT or other AI tool can define the HA sensor information from the Grott record layout information. It will be an experiment from my side and not ready tomorrow I am afraid.