johanmeijer/grott #641: Failing records to decrypt, any suggestions how to get them decrypted

berndh1979, 2025-04-08:
In the logs I see that two rather big records are not decrypted. I can't make up what they are. Perhaps some one can tell. The data is going to the growatt server, so I expect that this is either APX battery, inverter, or SmartMeter related. I have a ShineLAN-X datalogger, a MIN4200TL-XH inverter, APX BMS with 1x battery and a ChiNT Smartmeter. 

The rest of the records seem to be decrypted perfectly for my inverter and smartmeter. 

	 - Growatt packet received:
		  <socket.socket fd=5, family=2, type=1, proto=0, laddr=('172.18.0.2', 42542), raddr=('47.254.130.145', 5279)>
	 - Growatt original Data:
		 \x00\xf7\x00\x06\x01\x21\x01\x38\x00\x22\x28\x47\x22\x3e\x30\x75\x43\x25\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x2e\x32\x2b\x44\x36\x00\x33\x5f\x47\x2f\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x76\x6e
		 \xd4\x64\x4c\x42\x27\x32\x30\x47\x44\x1d\x20\x5e\x2d\x57\x20\x44\x77\x36\x21
		 \x7c\x76\x76\x75\x47\x72\x6d\xc8\x73\xd5\x74\x47\x72\x0b\x65\xc0\x66\xd5\x47
		 \x72\x6f\x77\x60\xf1\x74\xb1\x72\x6f\x77\x53\x74\x74\x47\x72\x6f\x77\x60\x6f
		 \x74\x47\x73\xfb\x77\x61\x74\x76\x47\x72\x6f\x78\x61\x7c\x75\x49\x73\x7a\x76
		 \x60\x74\x77\x47\x73\x6f\x13\x61\x10\x74\x46\x72\x6f\x77\x03\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x4c\x7b\x6f\x77\x61\x75\x74
		 \x43\x72\x6f\x77\x61\x74\x10\x57\x22\x6f\x77\x61\x74\x74\x47\x70\x49\x77\x61
		 \x74\x10\x45\x4a\x6e\xa7\x61\x74\x74\x47\x73\x31\x77\x61\x74\x74\x4a\x43\x62
		 \x64\x63\x6c\x7d\x87\x72\x6f\x77\x53\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x76\x46\x7b\xab\x77\x61\x74
		 \x74\x47\x72\x6c\x9f\x61\x74\x74\x47\x52\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x65\x61\x2f\xe2
	 - Grott automatic protocol detection
	 - Grott data record length 297
	 - layout   :  T060138MIN
	 - no matching record layout found, try generic
	 - Record layout used :  T060138MIN
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00f700060121013847504730434a4432314a00000000000000000000000000000000000000004
		 1454a3042474130304e0000000000000000000000000000000000000000000000000000010fa0
		 100b3048455133305a52315a36543030444e0b170201000002bf12a10000006412a112a100000
		 000018500f600000032000000000000011b00000194000000020000000f0008010e0115010100
		 03000100640064000100000062000000000000000000000000000000000b09000000010004000
		 0000000641050000000000000022600000064023801d000000000015e000000000d310d130218
		 09c00000003200000000000000000000000000000000000000000000020109c40000000000000
		 3e80000000020000000000000000000000000000000000011156890
	 - Grott data ack record or data record not defined no processing done

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('172.18.0.2', 5279), raddr=('10.0.0.6', 1032)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=5, family=2, type=1, proto=0, laddr=('172.18.0.2', 42542), raddr=('47.254.130.145', 5279)>
	 - Growatt original Data:
		 \x00\xf8\x00\x06\x00\x99\x01\x38\x00\x22\x28\x47\x22\x3e\x30\x75\x43\x25\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x2e\x32\x2b\x44\x36\x00\x33\x5f\x47\x2f\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x76\x72
		 \xac\x67\xb8\x72\x6d\x13\x05\x74\x10\x45\x6b\x90\x88\x9e\x8f\x74\x47\x73\x54
		 \x7a\x53\x79\x67\x46\x67\x6e\x76\x61\x74\x76\x61\x75\x6e\x74\xbc\x55\x17\x47
		 \x72\x6d\xc8\x61\x76\x74\x47\x73\x37\x77\x61\x74\x74\x47\x72\x99\x6f\x62\x9c
		 \x64\x0d\x72\x69\x77\x63\x7f\x74\x47\x72\x6f\x77\x6e\x23\x74\x47\x72\x6f\x43
		 \x74\xf6\xae\x46\xe0\x7f\x67\x44\x14
	 - Grott automatic protocol detection
	 - Grott data record length 161
	 - layout   :  T060138MIN
	 - no matching record layout found, try generic
	 - Record layout used :  T060138MIN
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00f800060099013847504730434a4432314a00000000000000000000000000000000000000004
		 1454a3042474130304e00000000000000000000000000000000000000000000000000000113d8
		 13ff0002646400640219fffffffb0000013b0d320d130115010100000226070103dd216300000
		 2bf000200000158000000000000f61803e8104a000600020b00000000000f5700000000341582
		 da019210102560
	 - Grott data ack record or data record not defined no processing done

johanmeijer, 2025-04-10:
Type 38 record I have not decoded yet while until now we did not missed this information (probably some kind of announcement record, but no sure). Do you miss information? 
First thing I see is that there is some information on devices: 8GPG0CJD21J (datalogger), AEJ0BGA00N (Inverter?) and HEQ30ZR1Z6 /T00DN (BMS?)

Good news is I have a bought a MIN TL-XH type inverter myself so the coming weeks we will see if I miss data and will work on this. 

The config will be a little bit different (ARK type batteries, smart meter I do not know yet, I will get and install everything soon).

berndh1979, 2025-04-11:
Hi thanks for the reply. Well, I don't miss data (for now). But of course it is always interesting to decode the information you don't know. :) FOMO

The HEQ.... is indeed the BMS module. All the status parameters of the APX battery combination is not in HA. So I expect that these records have to do with the battery pack.

Is there anything that I can be of help for decoding this? Is the hex coding based on ASCII or anything else?

berndh1979, 2025-04-14:
I gues the records are indeed related to the battery. 

I do miss the 'Power' values for Charge and Discharge. I do have the Energy values, but not the actual power going in and out of the battery from and towards the inverter. 

On the growatt server I see the values, but perhaps they are calculated.

johanmeijer, 2025-04-14:
I have installed my min tl-xh with ark hv batteries today.

Let's see if we can find it.

I expect the  38 records to be a announcement/ init record. I know there is
more info in the data record that grott not shows at this moment. Otherwise
It should be sent more often I think.

I will change grott to show that.

johanmeijer, 2025-04-16:
Which version of grott do you use? 

For version 3.0.0 and invtype = auto I think the Cgar info is displayed in pchr and discharge in pdischr. 

For version 2.8.3. the information is in bdc1_pdischr and  bdc1_pchr. 

The totals are displayed in edischrtoday, edischrtotal, echrtoday, echrtotal   (both 2,8,3 and 3.0.0)

I see these are filled for my ARK_HV system. I am not sure if this is also true for APX.

johanmeijer, 2025-04-16:
My charge info today (from Grafana). Be aware I am now playing around with the setting to determine how to control the configuration. 

[image]

berndh1979, 2025-04-17:
Great, after some more research I see that the values you mention are indeed decoded but only when I use invtype=MIN. When I change this to invtype=AUTO it simply does not decode the inverter records at all. I get only my smart meter data decoded.

But, it seems that HA Grott integration does not list the additional Inverter values. It just selects a subset of the inverter records. Reconfiguring HA Grott integration does not help either. 

I'm using grott version 2.8.3 with invtype=MIN.

berndh1979, 2025-04-17:
I went through the HA-grott integration code and I see the sensors not being listed in the sensors enumeration. Hence this integration does not insert the sensors into HA. 

E.g. bdc1_pdischr  is not listed as a sensor in the HA integration.

I use the 1.1.0 version from HACS, perhaps there is a newer version around on github. Will do some investigation.

So far it looks to be an issue with the integration but not with grott itself. 

Should I upgrade to grott v3.0.0 or stay at v2.8.3?? Any benefit?

johanmeijer, 2025-04-20:
Ok. I have the same issues with HA (at least in the Docker version, I will also look in the standalone RPI HA version). 

I do see this one: 

[image]

Strange value for the battery voltage I think, but ok (should be divided by 100, not by 10?). 

[image]

I am able to add sensors in the HA docker image, but that is not how to do this I think. Maybe @muppet3000 can help us here?