johanmeijer/grott #644: Please explain the different branches

stefan-nu, 2025-04-14:
Hi Johan,

first of all, thank you very much for this great piece of software. It is of great help to me.

I tried to run the master branch but realised that it did not correctly extract data from my inverter MID 25KTL3-XH. So I switched to branch Alpha, which seems to work better for my inverter. But I realised that it sends the wrong value to PVoutput. I would like to fix this bug. But I don't know which branch is the one to work on. As far as I found there are at least four active branches master, Alpha, Beta-(2.8.x), ba06665 (support for Smart meter). Could you please explain, what the different branches are ment for and which one you consider appropriate for adding new features?

Thank you
Stefan

johanmeijer, 2025-04-16:
@stefan-nu I will try to explain. 

The master branch is 2.8.3 and is the stable branche. 

The Beta is a copy of the master branche where I will change things (at this moment only bug fixes now real enhancements any more). 

The Alpha branche is the new GrottV3 that I am developing and is still under construction (but runs at my home without any major problems for weeks). Main features: 

- Grott and Grottserver are integrated, mode = server will start Grott in a combination of Grott and Grottserver. This means standard no data will be sent to or received from Growatt (see the wiki or discussions  for more information  on Grottserver). 
Be aware you can not change configuration via Growatt (you need to do this via the API). You can retrieve and change configuration of the datalogger / inverter via the Grott API. This is to protect your Inverter from changes from the outside. 
- The serverpassthrough = True parameter (in the new [Server] section of the grott.ini) will enable sending data to Growatt (not receiving) so you can still use the Growatt reporting on the website or in the phone app. 
- invtype = auto is introduced.  This will retrieve from the data record which registers are included and will process them automatically (Is not working for SPF at this moment). 

Grott v3 is compatable with V2 so you can run it in proxy mode (without grottserver) and still use the old invtype = layout definitions. 
All other Grott functions are still available. 
Grottserver can also be used to run standalone (without a grott.ini file or running grott with a grott.ini file with mode = serversa).  

What is wrong with the pvoutput value? This is the value (pvpowerout) that provides the value of power that the inverter sends to the grid. That can be less a generated while it is without the inverter loss of power and the power that is used to charge the batteries. . I do understand (now having an installation with a battery) this might not be the value you want to sent to pvoutput if you only want to see the PV generated power.

I can not change it while it is not a real bug, probably a lot of users are familiar with this value and using it.   What I want to do in a next version (probably v3.2 is to make the values that are being sent to pvoutput configurable.

If you want to change it in the short term you can create a new layout file (use t06NNNNXTL3.json from the examples/layout directory,  change it and and copy this one in the grott main directory). Most easy thing is to rename pvpowerout nd add a new pvpowerout with the same definitions as the one you want to use (you can have multiple variables pointing to the same value in the data record, see als: https://github.com/johanmeijer/grott/wiki/Grott-advanced-(customize-behaviour) ).   
Be aware if you use this new layout definition file you have to specify invtype = TL3!

stefan-nu, 2025-04-16:
Thank you very much for your explanation. I will look into branch Alpha then.

The issue about pvoutput is exactly as you said. The value currently includes the battery output power, which in my case is discharged at night. I want the pvoutput data to show only the actual PV output power, which obviously is zero at night.

Thanks for your hint about how to fix it. In fact, I am currently looking into your code.

johanmeijer, 2025-04-16:
You do not have to change to code to use another layout, you can use the layout file to change it (but you loose the auto functionality of V3 then)

johanmeijer, 2025-04-20:
Sorry I did not realized you closed it, I thought I did it by accident.

closed 2025-04-20 by johanmeijer