johanmeijer/grott #645: Configuration credentials new Growatt website

johanmeijer, 2025-04-16:
Growatt has renewed its website and I think it is looking good, but......

I am not able to change the datalogger/inverter configuration any more. At the configuration screen you now have to enter userid/password/stat grid colecter s/n and Plant Id. I have not managed to find the right combination for this. 

Is there somebody who knows which credentials to use here? 

The new datalogger screen: 

[image]

(I know I can change things with grottserver API but sometimes I want to use the Growatt website as well).

tabascoz, 2025-04-17:
Hi @johanmeijer , have you gotten any clue? What about trying to contact GW support? If they don't disclose the batt, would be challenging for new installations to use Grott, but still possible i think (by main in the middle etc. ) 

I Also noticed your fw version 7.0.0.7 may you confirm if this fw is fully compatible with grott? any issues?

johanmeijer, 2025-04-17:
I have now idea what this mean. 7.0.0 for the shinelink box works fine for me but with the RF-S stick. I were not able to get it working with a RF-X (but maybe the combination of shinelink-s box with RF-X is not the right one). 

I am using a shinewifi-x now.

I still have no clue which user/password to use in this screen. So if I want to configure things I have to run in grottserver mode and use the API I think.

dberrocal-git, 2025-04-18:
Hi, I haven't seen that they have changed the website, I use server.growatt.com, and openapi.growatt.com, I can still enter the password growatt + year + month + day to set things up.

From the ShinePhone app I can also change the settings with the same password form.

What url do you use?

johanmeijer, 2025-04-20:
I also use https://server.growatt.com/login (from the Netherlands). 

I can read and set the inverter settings (different screen layout) with the standard password. If I want to read or set the datalogger I get the screen above. 

The first screen when I  login tto growatt: 

[image]

johanmeijer, 2025-04-20:
That was interesting: after changing the plant information (setting I use my own energy and export the surplus)  I now have to old site back and can change everything again with the standard password,

johanmeijer, 2025-04-20:
That is a pitty in the other site I was able to export all measured values in a excel sheet and was able to run diagnostics for my inverter. 

That is not possible any more and I do not now how to go back ;( I want to be able to switch.

johanmeijer, 2025-04-23:
I need sometimes the communiction with growatt to debug/very the communication flow (used in grottserver)>

At this moment I also use the mobile app to test wit my new configuration (MIN  TL-XH inverter with ARK HV batteries)