johanmeijer/grott #646: Unit Tests

stefan-nu, 2025-04-18:
Currently, the project has no unit tests. 

When adding new features, there is always a risk that existing features get broken or the behaviour changes. Unit tests would help to avoid such pitfalls.

Is there interest to add unit tests to the project?

stefan-nu, 2025-04-20:
That sounds great. I guess what you describe are system tests. That would be really useful too. 

My intention was rather to add unit tests. Thus, small tests that check if single code parts work as expected. If you are interested in it, I can write you an example and if you feel this is useful, you can use it.

johanmeijer, 2025-04-20:
Yes please do. I started Grott also to improve my python programming skills and learn more about modern programming methods.

stefan-nu, 2025-04-20:
I've added a pull request with a commit that contains a simple unit test. I used branch Alpha, as this should probably not go into the master until it was properly tested and more tests have been added.

johanmeijer, 2025-04-23:
I am going to look at it, thanks. 

Internally I already working on 3.1 (not published on Github, still doing my own old fashion version management, sorry). I will see if this fit on that release too.

stefan-nu, 2025-04-24:
I am very sure that it will fit to your new version 3.1 too. 

This is the idea of unit tests. It will check if a changed version does still behave as the old one. If not, then either the new version contains a bug, which was luckily detected by the unit test and can be changed before the bug gets into the world. Or the new version intentionally works differently and in this case, the unit test has to be updated, which again is good to be detected.

closed 2025-05-03 by stefan-nu