johanmeijer/grott #650: Grott Received Data but Not decrypted

eismarkt, 2025-04-28:
I have 2 inverter One Min 4600 with shine lan Stick. That work an Grott Received the data and decrypted ever Minute.
The second is an SPH 10000 and the Problem is sometimes work Grott every minutes to decrypt Data and the most of the time works it only every 1-3 hour. 
I see every Minute the SPH send data to Grott but Grott don’t decrypt it

	 - Growatt data decrypted V2
	 - Matched inverter serial to inverter type sph
	 - Grott MQTT topic used : energy/growatt
	 - Growatt packet received:
		  <socket.socket fd=15, family=2, type=1, proto=0, laddr=('172.30.33.9', 5279), raddr=('192.168.3.100', 1034)>
	 - Growatt packet received:
		  <socket.socket fd=15, family=2, type=1, proto=0, laddr=('172.30.33.9', 5279), raddr=('192.168.3.100', 1034)>
	 - Growatt packet received:
		  <socket.socket fd=14, family=2, type=1, proto=0, laddr=('172.30.33.9', 58004), raddr=('8.209.71.240', 5279)>
	 - Growatt packet received:
		  <socket.socket fd=15, family=2, type=1, proto=0, laddr=('172.30.33.9', 5279), raddr=('192.168.3.100', 1034)>
	 - Growatt packet received:
		  <socket.socket fd=14, family=2, type=1, proto=0, laddr=('172.30.33.9', 58004), raddr=('8.209.71.240', 5279)>
	 - Growatt data decrypted V2
	 - Growatt packet received:
		  <socket.socket fd=15, family=2, type=1, proto=0, laddr=('172.30.33.9', 5279), raddr=('192.168.3.100', 1034)>
	 - Growatt packet received:
		  <socket.socket fd=14, family=2, type=1, proto=0, laddr=('172.30.33.9', 58004), raddr=('8.209.71.240', 5279)>
	 - Growatt packet received:
		  <socket.socket fd=16, family=2, type=1, proto=0, laddr=('172.30.33.9', 44540), raddr=('8.209.71.240', 5279)>
	 - Growatt packet received:
		  <socket.socket fd=17, family=2, type=1, proto=0, laddr=('172.30.33.9', 5279), raddr=('192.168.3.100', 1035)>
	 - Growatt packet received:
		  <socket.socket fd=17, family=2, type=1, proto=0, laddr=('172.30.33.9', 5279), raddr=('192.168.3.100', 1035)>
	 - Growatt packet received:
		  <socket.socket fd=16, family=2, type=1, proto=0, laddr=('172.30.33.9', 44540), raddr=('8.209.71.240', 5279)>
	 - Growatt packet received:
		  <socket.socket fd=16, family=2, type=1, proto=0, laddr=('172.30.33.9', 44540), raddr=('8.209.71.240', 5279)>
	 - Growatt packet received:
		  <socket.socket fd=16, family=2, type=1, proto=0, laddr=('172.30.33.9', 44540), raddr=('8.209.71.240', 5279)>
	 - Growatt packet received:
		  <socket.socket fd=16, family=2, type=1, proto=0, laddr=('172.30.33.9', 44540), raddr=('8.209.71.240', 5279)>
	 - Growatt packet received:
		  <socket.socket fd=16, family=2, type=1, proto=0, laddr=('172.30.33.9', 44540), raddr=('8.209.71.240', 5279)>
	 - Growatt packet received:
		  <socket.socket fd=16, family=2, type=1, proto=0, laddr=('172.30.33.9', 44540), raddr=('8.209.71.240', 5279)>
	 - Growatt packet received:
		  <socket.socket fd=16, family=2, type=1, proto=0, laddr=('172.30.33.9', 44540), raddr=('8.209.71.240', 5279)>
	 - Growatt packet received:
		  <socket.socket fd=16, family=2, type=1, proto=0, laddr=('172.30.33.9', 44540), raddr=('8.209.71.240', 5279)>
	 - Growatt packet received:
		  <socket.socket fd=16, family=2, type=1, proto=0, laddr=('172.30.33.9', 44540), raddr=('8.209.71.240', 5279)>
	 - Growatt packet received:
		  <socket.socket

johanmeijer, 2025-04-30:
Do you use a shinelink-box with a RF stick? 

With my new inverter I tried to use an RF stick with my "old" shinelink-s. The behavior looks the same what you describing occasionally sending out a data record. 

I didn't really feel to find out what was wrong (after a couple of attempts to upgrade the firmware of the shinelink) and bought a shinewifi-x.

eismarkt, 2025-04-30:
hi, I used a Shine LAN box until 2 weeks ago. but suddenly this no longer connects with grot. Then I got a LAN stick for each inverter. And so the min4600 runs smoothly and sends data every minute. The SPH only occasionally connects and sends data about every 20-30 minutes. Sa's behavior appeared suddenly.

johanmeijer, 2025-05-01:
Strange do you know if there has been a firmware update for the Lan Stick? Did you tried to reset it (I suppose you did).

johanmeijer, 2025-05-01:
Do you use only Grott or also Grottserver?

eismarkt, 2025-05-01:
I have the HA Addon installed and Grott .
I don't know, but the two LAN sticks have the same software on them. One works properly and the other doesn't. What surprises me is that the Shine LAN box doesn't connect at all anymore. I also tried a second one, and it doesn't connect either...

johanmeijer, 2025-05-01:
This is the network config for my shinelink. Be aware the RF-X stick with my MIN inverter does not work correctly, but the shinelink connects to grott (and Growatt). 

[image]

eismarkt, 2025-05-01:
I have  the Same with different IP s.
But One Night stop the data Connection to the addon and then never start it again with the shine lan box. With the Shine lan Stick and the same network config it start but correctly only with one of them.. the second one send the data not regularly.